nordicsemi/Android-DFU-Library #5: NPE on openInputStream()

eliotstock, 2015-07-02:
You are not checking for a null Cursor instance on this line:

https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1881

In my case I'm using a hex file which is bundled in with the Android app under res/raw. The input stream URI for it looks like this:

android.resource://cc.biketracker.android/2131099651

That URI has been built like this:

Uri firmwareHex = Uri.parse("android.resource://" + getPackageName() + "/"
                + R.raw.bike_s110);

You are querying for "_display_name". Does that always exist?

eliotstock, 2015-07-02:
In fact, the NPE is in the finally block, where you assume the Cursor is non-null.

philips77, 2015-07-02:
Hi,
Would it work if you moved the hex file to assets? I'm querying for the name to check the file extension. If you want to take it from res/raw I guess it would be better to just add a third option for passing the data. Not you may use URI or filePath, the resId+type (bin/hex/zip) is missing.
But updating this may take some time.

eliotstock, 2015-07-02:
Not sure about using the assets directory. res/raw at least gives you build-time checking that the file exists. I think this is a valid use case.

philips77, 2015-07-22:
Hi,
I've just push some changes to the DFU Library. Now you may specify the resource id for the firmware file(s). Please, use the new DfuServiceInitiator class, like shown in the nRF Toolbox -> DfuActivity class.

closed 2015-07-22 by philips77

eliotstock, 2015-07-22:
Looks marvellous! Thank you very much indeed.