nordicsemi/Android-DFU-Library #7: Proposal/Improvement: checking version dependent requirements

boaks, 2015-07-17:
Trying the DFU with SDK-8.1 and this DFU android library, I stuck with “general error numbers” report by the device. Finally I found the comments (DfuBaseService, line 1570, above comment Send DFU Init Packet) that clarifies the usage of the init packet.

Though the DFU library knowns the DFU version at this point, wouldn’t it be an improvement to report the missing “init packet” as error (“since DF0 v0.6 the init packet is required!”)?

And, maybe you can also extend the Makefiles to call the “nrf dfu” to create the zip (e.g. “examples\ble_peripheral\ble_app_hrs\pca10028\s110_with_dfu”)

philips77, 2015-07-17:
Hi, you're right. I'll add another error number indicating that init packet is required.

Regarding your last request (Makefile), please write it on DevZone or create an issue on our website.
Please, find more info here: https://www.nordicsemi.com/eng/Support-Community

philips77, 2015-07-22:
The service should report with the new error number now. Check out the latest changes. I've also added the DfuServiceInitiator to make it easy to configure and start the service, and DfuServiceListenerHelper + 2 listeners for events handling.

closed 2015-07-22 by philips77