nordicsemi/Android-DFU-Library #9: dsiconnect while sending data

wim-vogt, 2015-09-17:
I/DfuBaseService﹕ mBytesSent：29140
I/DfuBaseService﹕ mImageSizeInBytes：51732
D/BluetoothGatt﹕ writeCharacteristic() - uuid: 31552658-1212-efde-1523-785feabcd123
D/BluetoothGatt﹕ onCharacteristicWrite() - Device=F1:11:11:11:11:11 UUID=00001532-1212-efde-1523-785feabcd123 Status=0
I/DfuBaseService﹕ mBytesSent：29160
I/DfuBaseService﹕ mImageSizeInBytes：51732
D/BluetoothGatt﹕ writeCharacteristic() - uuid: 31552658-1212-efde-1523-785feabcd123
D/BluetoothGatt﹕ onCharacteristicWrite() - Device=F1:11:11:11:11:11 UUID=00001532-1212-efde-1523-785feabcd123 Status=0
I/DfuBaseService﹕ mBytesSent：29180
I/DfuBaseService﹕ mImageSizeInBytes：51732
D/BluetoothGatt﹕ writeCharacteristic() - uuid: 31552658-1212-efde-1523-785feabcd123
D/BluetoothGatt﹕ onCharacteristicWrite() - Device=F1:11:11:11:11:11 UUID=00001532-1212-efde-1523-785feabcd123 Status=0
I/DfuBaseService﹕ mBytesSent：29200
I/DfuBaseService﹕ mImageSizeInBytes：51732
D/BluetoothGatt﹕ onClientConnectionState() - status=0 clientIf=5 device=F1:11:11:11:11:11
I/DfuBaseService﹕ Connection state change: 0 newState: 0
I/DfuBaseService﹕ Disconnected from GATT server
E/DfuBaseService﹕ Disconnected while sending data
E/DfuBaseService﹕ Uploading Firmware Image failed: device disconnected (connection state: 0)
I/DfuBaseService﹕ Cleaning up...
D/BluetoothGatt﹕ close()
D/BluetoothGatt﹕ unregisterApp() - mClientIf=5

philips77, 2015-09-18:
Hi, could you please give some more details about your issue?
Does it happen every time? What devices are you using? Is is your custom DFU bootloader or a sample from SDK? I can't say the reason of terminating the connection after just this log.
Could you send the hci logs from the Android?
Also, if you have problem with DFU, I recommend you creating a case on our website. Our support may help you better than we in the mobile apps team.

wim-vogt, 2015-09-18:
Thank you for your answer , this is a problem found today on the hardware , when dfu more than one minute , the hardware will automatically reset , I am very sorry

philips77, 2015-09-18:
OK, no problem. Hope it works now.

closed 2015-09-18 by philips77