nordicsemi/Android-DFU-Library #10: Connection state change error: 8 problem

cnbuff410, 2015-11-28:
Hi:

I am facing a problem during the OTA. So the error is 

E/DfuBaseService: Connection state change error: 8 newState: 0
E/DfuBaseService: Unable to write Op Code (error 32776)

It generally doesn't happen for the first time user using it, but happens every single time afterwards.

My firmware update process is

1. Find device
2. Send command to put device into DFUTarg mode
3. Wait for disconnection signal, once it comes, I refresh the Gatt cache and close Gatt, then rescan for "DFUTarg"
4. Once I find it, I create DfuServiceInitiator and start the process

Besides this error, the onError method in DfuProgressListenerAdapter will be called and the error type is 1, error is 8.

One thing to add, during my test I would run into the same error sometimes(5-6 out of 10) by using nRF Toolbox, but not always.

This is blocking all our Android users from getting firmware update. It could be something wrong on my side and I'm still debugging it, but any hint or suggestions would be very helpful at this time.

philips77, 2015-11-28:
Hi,
As you may read here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/error/GattError.java, in the first method, the error 8 is a timeout. Is your device advertising? Does it advertise with the address you want to connect? In non-buttonless DFU, since SDK 8 or 7, the DFU targ is advertising with address +1 in such case.

And another question, why don't you use the service to switch from app mode to DFU targ mode? Do you have your own, custom implementation?

From which sdk is your DFU implementation?

cnbuff410, 2015-11-28:
Hi Philips

We don't have the dfu service, we enter the boot loader through a system reset. The dfu service is implemented in the bootloader. The flow is basically like sending a command in master control panel, once it's disconnected, check if it's in "DFUTarg" mode now, if so, use Toolbox to do the OTA.

We use SDK 7.2

Once I set zip and call DfuServiceInitiator.start(), it actually connects to the DFUTarg I believe, and got error 8 when trying to setCharacteristicNotification. Here is the log from BluetoothGatt:

D/Stryd: Start DfuServiceInitiator
D/BluetoothGatt: connect() - device: FF:7C:7E:1E:FD:C0, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=1a17a109-8589-4640-a900-d3d0a021513b
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=FF:7C:7E:1E:FD:C0
D/BluetoothGatt: discoverServices() - device: FF:7C:7E:1E:FD:C0
D/BluetoothGatt: onSearchComplete() = Device=FF:7C:7E:1E:FD:C0 Status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=FF:7C:7E:1E:FD:C0
E/DfuBaseService: Connection state change error: 8 newState: 0
E/DfuBaseService: Unable to write Op Code (error 32776)
D/BluetoothGatt: refresh() - device: FF:7C:7E:1E:FD:C0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=5

cnbuff410, 2016-02-05:
On certain combinations of Android OS versions + phone models, the DFU update fails. The failure happens in this sequence, which we see from the BLE Sniffer:

1 Android requests a BLE connection, which is outside the set connection interval range of the DFU target.
2 Android performs a service discovery, and in tandem attempts to update the connection interval. Up to 6 connection intervals updates procedures are initiated by Android. Near the end of this process, Android is already initiating the DFU process by enabling the DFU characteristic, writing the image size and DFU type, and writing the DFU start packet.
3 Next, correct operation would then have the DFU target send a notification with opcode 16, ReqOpCode 1 and RespValue 1. This does not happen. Instead, the DFU target fails to send any BLE packets at all, and after 4 seconds, the connection supervisory timeout triggers, and Android ends the connection while flagging the error, BLE_CONN_TIMEOUT.

Here's what we've learned why this is happening, from debugging the bootloader during this process:
1 The DFU target, when processing the START_PACKET, throws the error BLE_DFU_RESP_VAL_OPER_FAILED.
2 Drilling down into the data handling, the above error code originates in the flash pstorage module as a NRD_ERROR_TIMEOUT.

Here's what we've tried, which solves the issue:
1 Increase the connection interval range from 15-30ms to 30-50ms. iOS is negotiating a connection interval of 30 ms, where Android normally selects 16.25 ms.

Here's what we want to try, so as not to require modifying the bootloader:

1 Remove the service discovery when connecting to DfuTarg. For comparison, the iOS DFU library does not do service discovery, and we have not seen any issues with it.
2 Insert a delay between the BLE connection and the DFU initialization. We hope this will decouple the high traffic of connection parameter negotiation (as I said, for some reason Android does this over and over, up to 6 times).

cnbuff410, 2016-02-05:
For now, one quick thing we will test is to comment out the "discoverServices" in the DfuBaseService.java and see if it fixes the issue or not. Please let us know if there is any other features in the library which rely on it and would prevent us from removing it.

Thanks

philips77, 2016-02-05:
Hi, in my opinion until you call gatt.discoverServices() the gatt.getServices() will return an empty list. As of the bug with too short connection interval - indeed we see that DFU bootloader needs more time to flash data than is given when connection interval is short and master sends 6 packets per connection interval. I'd say it's a DFU bug (or pstorage, etc). A workaround is to set the notification receipt number to 4. Then DFU will send a notification every 4 packets and phone will never try to send more than that. It makes the process a bit longer, thou.

cnbuff410, 2016-02-05:
When you say "to set the notification receipt number to 4", do you mean to change the "Number of packets" in the DFU Settings from "1" to "4"?

If so, we already tried it and it doesn't work.

cnbuff410, 2016-02-06:
We made it work here, and what we did is to manually add some delay(1s in the example code) between the service discovery and DFU enabling.

The code is 

		@Override
		public void onServicesDiscovered(final BluetoothGatt gatt, final int status) {
			if (status == BluetoothGatt.GATT_SUCCESS) {
				logi("Services discovered");
				mConnectionState = STATE_CONNECTED_AND_READY;
			} else {
				loge("Service discovery error: " + status);
				mError = ERROR_CONNECTION_MASK | status;
			}

			**mHandler.postDelayed(new Runnable() {
				@Override
				public void run() {
					// Notify waiting thread
					synchronized (mLock) {
						mLock.notifyAll();
					}
				}
			}, 1000);**
		}

Do you think this would cause any other issues? Or this is a legit workaround and you can merge it into the next release?

cnbuff410, 2016-02-06:
Honestly any delay would work, so if the handler code is not ideal, we can also change it to be

`			synchronized (this) {
				try {
					wait(1000);
				} catch (InterruptedException e) {
					e.printStackTrace();
				}
			}
			synchronized (mLock) {
				mLock.notifyAll();
			}
`

philips77, 2016-02-06:
Delays are a very good solutions. I'll add this one, and the one from existing pull request to the new version. Could you please create a pull request with changes that work for you? Thanks!

philips77, 2016-02-06:
But without e.printStackTrace()

cnbuff410, 2016-02-07:
PR submitted

philips77, 2016-02-08:
Hi, I've accepted the PR. It is now available in version 0.6.2 (available from jcenter, or as source code here). Could you try if it works better?
Thanks for contributing!

ps. Regarding the too-fast uploading (15ms conn interval, 6 packets in each) I will experiment with http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestConnectionPriority%28int%29 if the interval can't be changed to 30ms on a device.

closed 2016-02-08 by philips77

philips77, 2016-02-09:
The updated lib is available in the new nRF Master Control Panel 4.0.4.