nordicsemi/Android-DFU-Library #13: Proguard configuration missing

natangr, 2016-04-13:
I'm using proguard and minifyEnabled in my project in the release build and I'm not able to get the device in DFU mode. The debug build has minifyEnabled set to false and works as expected.

My question is, what should I add to the proguard-rules for this library?

natangr, 2016-04-13:
I found the solution, I'm leaving the answer here in case someone else have this issue. All I had to do is add the following to my proguard-rules.pro:
`-keep class no.nordicsemi.android.dfu.** { *; }`

philips77, 2016-04-13:
Thanks! I'll write it to the README file and close the issue.

philips77, 2016-04-14:
I've added information about proguard to Android-DFU-Library project's readme file: https://github.com/NordicSemiconductor/Android-DFU-Library/commit/269cb53a3aed6c876f76ec0786c8a31d555bf1f5
Thanks again, closing.

closed 2016-04-14 by philips77