nordicsemi/Android-DFU-Library #15: GATT Error 133 retry policy

michaldrabik, 2016-06-28:
Hello there. It's not that much of an issue but maybe an improvement idea. 

Shouldn't there be some kind of auto-retry policy while connecting to a device and receiving GATT 133 error?

Right now, when update process is initialised, we try to connect to a device and if we receive GATT 133 error, the update error broadcast is being send and connection is terminated. During my work with BLE devices I encounter GATT 133 error a lot of time (mostly on lower-end devices) so I came up with some auto-retry policies that try reconnecting 2-3 times before throwing an actual 133 error.

That might be a nice improvement for DfuBaseService or maybe there are some drawbacks that I'm not aware of or retries are already there and I can't see them?

Best Regards

philips77, 2016-06-28:
Hello Michał,
Very often the 133 error while connecting means that the device is not reachable (disconnected, advertising with a different address, etc), but I know that sometimes it just happens even if everything is OK. So far I assumed that it may be up to the final app to retry starting the service in such case, but as you know it's not that rare and perhaps the service should retry at least 2 times.
There will be an update of DFU Library in July and I will look into it deeply. I want to solve the Invalid_state error as well (perhaps by trying to add resuming the old connection, hopefully it may work).
Let the issue hang here and remind me about it.

michaldrabik, 2016-06-28:
@philips77 That's great. Until then I guess I'll just modify the service to suit my needs (yay open-source). I might come up with some pull requests If I find some time and find them good enough.

philips77, 2016-06-30:
Hi, the issue should now be "solved". I have added a second attempt in case of status > 0. Also, another second attempt if Start DFU returns Invalid State error (this happens if previous upload failed and DFU target stored part of the firmware. Sending Start DFU again causes this state. Service could resume upload from the last location, but it's safer to start from scratch, as we don't know if the same fw is to be uploaded).
Version on jcenter: 0.6.3. Could you give it a try? Dzięki!

michaldrabik, 2016-06-30:
Looks good. I've already made some changes by myself but I guess I'll just swap to 0.6.3 as it now does the same thing and it also gives State Error support (although I've never encountered this one).

One thing that might be worth considering is giving developers an option to choose amount of retries via Intent Extra (DfuBaseInitiator `setConnectionRetriesAmount` method or smthing like that).

Thanks a lot, I guess we can close now. Pzdr

closed 2016-06-30 by michaldrabik