nordicsemi/Android-DFU-Library #16: How close the notifications?

Kevin-Stark, 2016-08-08:
HI,I used your dependency of Lib.but it always showed the notification,I want to close it,how should I do?

philips77, 2016-08-08:
If you use the DFUServiceInitiator you can use the setDisableNotification(boolean) method to disable it.

Kevin-Stark, 2016-08-08:
thanks a lot. @philips77

closed 2016-08-08 by Kevin-Stark

devangichhatbar, 2018-11-01:
@philips77  setDisableNotification(boolean) method doesn't work for me. Any fresh solution for Oreo+ devices?

philips77, 2018-11-01:
To show, or hide the notification? On Oreo+, to show, call https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L765

devangichhatbar, 2018-11-01:
What is for hiding notification? 

.setDisableNotification(true)
.setForeground(false) 

I already use that thing but still receiving an error just like: 

```
AndroidRuntime: android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1768)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:106)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:164)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:6494)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
10-31 08:46:29.421 27745 27745 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

philips77, 2018-11-01:
Yes, by setting those 2 flags in the Initiator, you should not have any notifications. Could you debug where are they created? Where the startForeground is being called?