nordicsemi/Android-DFU-Library #18: Can't use with external storage zip file

AchrafAmil, 2016-08-19:
Hello!
I'm always getting the following error when passing an external storage Uri (of a freshly downloaded .zip file).
I've tried many Uris constructed with:
file:///storage/emulated/0/dfu3.zip
/storage/emulated/0/dfu3.zip
Environment.getExternalStorageDirectory().getAbsolutePath() + File.separator+ "dfu"+ serverVersion +".zip"
content://com.android.externalstorage.documents/document/primary:dfu3.zip

only this worked (after importing the .zip into project raw res folder):
android.resource://my.package.name/" + R.raw.dfu

`
08-20 15:39:38.534 14730-16014/co.supertino.airmote E/DfuBaseService: An exception occurred while calculating file size
                                                                      no.nordicsemi.android.dfu.internal.exception.HexFileValidationException: Not a HEX file
                                                                          at no.nordicsemi.android.dfu.internal.HexInputStream.checkComma(HexInputStream.java:324)
                                                                          at no.nordicsemi.android.dfu.internal.HexInputStream.calculateBinSize(HexInputStream.java:97)
                                                                          at no.nordicsemi.android.dfu.internal.HexInputStream.<init>(HexInputStream.java:83)
                                                                          at no.nordicsemi.android.dfu.internal.ArchiveInputStream.parseZip(ArchiveInputStream.java:273)
                                                                          at no.nordicsemi.android.dfu.internal.ArchiveInputStream.<init>(ArchiveInputStream.java:113)
                                                                          at no.nordicsemi.android.dfu.DfuBaseService.openInputStream(DfuBaseService.java:2091)
                                                                          at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1196)
                                                                          at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:66)
                                                                          at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                          at android.os.Looper.loop(Looper.java:148)
                                                                          at android.os.HandlerThread.run(HandlerThread.java:61)`

philips77, 2016-08-19:
Hi, what do you have in the zip?

AchrafAmil, 2016-08-19:
Hi,
I asked for a new zip file and the problem got resolved. Please close the issue. Thanks for reactivity :+1:

closed 2016-08-19 by philips77