nordicsemi/Android-DFU-Library #19: DfuService has leaked IntentReceiver

Alkisum, 2016-09-05:
Using the Android DFU Library 1.0.0, I receive the following error when the DFU successfully finishes:

```
09-05 13:19:43.407 29695-29695/no.nordicsemi.android.nrftoolbox E/ActivityThread: Service no.nordicsemi.android.nrftoolbox.dfu.DfuService has leaked IntentReceiver no.nordicsemi.android.dfu.BaseDfuImpl$1@96850a2 that was originally registered here. Are you missing a call to unregisterReceiver()?
android.app.IntentReceiverLeaked: Service no.nordicsemi.android.nrftoolbox.dfu.DfuService has leaked IntentReceiver no.nordicsemi.android.dfu.BaseDfuImpl$1@96850a2 that was originally registered here. Are you missing a call to unregisterReceiver()?
at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:918)
at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:719)
at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1172)
at android.app.ContextImpl.registerReceiver(ContextImpl.java:1152)
at android.app.ContextImpl.registerReceiver(ContextImpl.java:1146)
at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:554)
at no.nordicsemi.android.dfu.BaseDfuImpl.<init>(BaseDfuImpl.java:218)
at no.nordicsemi.android.dfu.BaseCustomDfuImpl.<init>(BaseCustomDfuImpl.java:225)
at no.nordicsemi.android.dfu.LegacyDfuImpl.<init>(LegacyDfuImpl.java:127)
at no.nordicsemi.android.dfu.DfuServiceProvider.getDfuImpl(DfuServiceProvider.java:37)
at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:963)
at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:66)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:148)
at android.os.HandlerThread.run(HandlerThread.java:61)
```

This error occurs with the nRF Toolbox and also with my own application that uses the Android DFU Library. The tests have been ran with a Nexus 5X on Android 6.0.1.

Should something be done with the DFU library? Or with the apps using it?

philips77, 2016-09-05:
It should be fixed in the library itself. I will look into it. Thanks for the bug report!

philips77, 2016-09-05:
This broadcast receiver is unregistered in https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L747. Could you check why it's not called in your app?

Alkisum, 2016-09-06:
The `onDestroy()` method in DfuBaseService is actually called and the BaseDfuImpl broadcast receiver unregistered.

However, `DfuServiceProvider.getDfuImpl(...)` is called several times: once before Sending Start DFU command and once after when reconnecting to the DFU target. This method returns every time a new instance of BaseDfuImpl and I guess the old instances of BaseDfuImpl are still registered to the broadcast receiver.

Does it make sense?

philips77, 2016-09-06:
Hmm.. Reconnecting is done by starting the service again and, as I thought, with killing the previous instance. But looks like you are right and the onDestroy will not be called if I start the service again before the old one dies, or will be called but will call onDestroy on the new instance of BaseDfuImpl instead of the old one, which could be even worst. 
I did some testing yesterday, but without buttonless DFU switching. If you are using the legacy DFU only you can still use the 0.6.3 version for now.
I just started a short vacations and I'll be able to test it next Wednesday.

Alkisum, 2016-09-06:
Thanks for checking this out. I will use the 0.6.3 version for now.

philips77, 2016-09-15:
Hi, I've just released 1.0.1 with this bug fixed. Closing issue.

closed 2016-09-15 by philips77