nordicsemi/Android-DFU-Library #20: I can't OAD success!

TimAimee, 2016-09-21:
the callback message:OnError=4096,errorType=0,Message:DFU DEVICE DISCONNECTED!
I try many time,every time is fail.

philips77, 2016-09-21:
What device are you testing on? 
I had similar problem on Nexus 6 with N preview release (there is still no final Android 7.0 for this phone). I don't know why it happens, the system sends acl_disconnected broadcast. If you have this phone and version, I think you need to wait for the final release, as it looks like Android bug. I will look into this broadcast. If not, please provide more details.

TimAimee, 2016-09-21:
i come from china,and i meet this problem on a lots phone,  eg. vivo phone，乐视 phone，360 phone, they are  cutom OS base on android.

philips77, 2016-09-21:
Could you do DFU using nRF Connect and send me the log if it fails?

TimAimee, 2016-09-21:
ok.please wait a moment.

TimAimee, 2016-09-21:
09-21 15:59:19.533 2051-7154/com.veepoo.testoad D/BluetoothAdapter: isEnabled
09-21 15:59:19.534 2051-7154/com.veepoo.testoad I/DfuBaseService: Connecting to the device...
09-21 15:59:19.536 2051-7154/com.veepoo.testoad D/BluetoothDevice: mAddress: F9:C2:C4:DB:A2:96
09-21 15:59:19.537 2051-7154/com.veepoo.testoad D/BluetoothGatt: connect() - device: F9:C2:C4:DB:A2:96, auto: false
09-21 15:59:19.537 2051-7154/com.veepoo.testoad D/BluetoothGatt: registerApp()
09-21 15:59:19.538 2051-7154/com.veepoo.testoad D/BluetoothGatt: registerApp() - UUID=e51d5f5a-156b-442d-81df-166497a6835c
09-21 15:59:19.545 2051-2068/com.veepoo.testoad D/BluetoothGatt: onClientRegistered() - status=0 clientIf=1
09-21 15:59:19.545 2051-2068/com.veepoo.testoad D/BluetoothDevice: mAddress: F9:C2:C4:DB:A2:96
09-21 15:59:39.577 2051-2069/com.veepoo.testoad D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=1 device=F9:C2:C4:DB:A2:96
09-21 15:59:39.584 2051-2069/com.veepoo.testoad D/BluetoothDevice: mAddress: F9:C2:C4:DB:A2:96
09-21 15:59:39.584 2051-2069/com.veepoo.testoad I/DfuBaseService: Disconnected from GATT server
09-21 15:59:39.584 2051-7154/com.veepoo.testoad E/DfuBaseService: Device got disconnected before service discovery finished
09-21 15:59:39.587 2051-7154/com.veepoo.testoad D/BluetoothDevice: mAddress: F9:C2:C4:DB:A2:96
09-21 15:59:39.587 2051-7154/com.veepoo.testoad D/BluetoothGatt: refresh() - device: F9:C2:C4:DB:A2:96
09-21 15:59:39.587 2051-7154/com.veepoo.testoad D/BluetoothDevice: mAddress: F9:C2:C4:DB:A2:96
09-21 15:59:39.589 2051-7154/com.veepoo.testoad I/DfuBaseService: Refreshing result: true
09-21 15:59:39.589 2051-7154/com.veepoo.testoad I/DfuBaseService: Cleaning up...
09-21 15:59:39.589 2051-7154/com.veepoo.testoad D/BluetoothGatt: close()
09-21 15:59:39.590 2051-7154/com.veepoo.testoad D/BluetoothGatt: unregisterApp() - mClientIf=1
09-21 15:59:40.193 2051-2051/com.veepoo.testoad I/timaimee-MainActivity: ╔════════════════════════════════════════════════════════════════════════════════════════
09-21 15:59:40.194 2051-2051/com.veepoo.testoad I/timaimee-MainActivity: ║ onDeviceDisconnected
09-21 15:59:40.195 2051-2051/com.veepoo.testoad I/timaimee-MainActivity: ╚════════════════════════════════════════════════════════════════════════════════════════
09-21 15:59:40.195 2051-2051/com.veepoo.testoad I/timaimee-MainActivity: ╔════════════════════════════════════════════════════════════════════════════════════════
09-21 15:59:40.196 2051-2051/com.veepoo.testoad I/timaimee-MainActivity: ║ onError=4096,errorType=0,message=DFU DEVICE DISCONNECTED
09-21 15:59:40.196 2051-2051/com.veepoo.testoad I/timaimee-MainActivity: ╚════════════════════════════════════════════════════════════════════════════════════════

TimAimee, 2016-09-21:
this is my demo code       
        
        Boolean isBinder = false;
        final DfuServiceInitiator starter = new DfuServiceInitiator(macAddress)
                .setDeviceName(name)
                .setKeepBond(isBinder);
        mFilePath = oad_path;
        starter.setZip(mFileStreamUri, mFilePath);
        starter.start(this, DfuService.class);

philips77, 2016-09-21:
I was talking about the log from nRF Connect app. You can find it on Play store, or Android-nRF-Connect project on Github. But from the log you gave it looks like it disconnects near service discovery. Do you use library version 1.0.1?

TimAimee, 2016-09-21:
yes! i use the library version 1.0.1.

TimAimee, 2016-09-21:
it seems like the phone can't find the service.some time i close bluetooth and then open it ,the dfu work success,but most time is fail

TimAimee, 2016-09-21:
should i use nRF Connect app? laster year i had  see the nRF Connect app  source code ,it also the use the DFU library ,and the code is similar the version library  1.0.1 code.

TimAimee, 2016-09-21:
i have use the nRF Connect for Mobile.when device into the DFU model,the phone is also can not find the service.

TimAimee, 2016-09-21:
so.... what should i do? help me.

TimAimee, 2016-09-21:
in the other phone.
     09-21 16:57:13.585 19495-19964/com.veepoo.hband I/DfuBaseService: Connecting to the device...
09-21 16:57:19.090 19495-19506/com.veepoo.hband E/DfuBaseService: Connection state change error: 133 newState: 2
09-21 16:57:19.090 19495-19964/com.veepoo.hband E/DfuBaseService: An error occurred while connecting to the device:133
09-21 16:57:19.105 19495-19964/com.veepoo.hband I/DfuBaseService: Disconnecting from the device...
09-21 16:57:19.115 19495-19964/com.veepoo.hband I/DfuBaseService: Refreshing result: true
09-21 16:57:19.115 19495-19964/com.veepoo.hband I/DfuBaseService: Cleaning up...
09-21 16:57:19.115 19495-19964/com.veepoo.hband I/DfuBaseService: Restarting the service
09-21 16:57:21.150 19495-19964/com.veepoo.hband I/DfuBaseService: Connecting to the device...
09-21 16:57:26.285 19495-19684/com.veepoo.hband E/DfuBaseService: Connection state change error: 133 newState: 2
09-21 16:57:26.285 19495-19964/com.veepoo.hband E/DfuBaseService: An error occurred while connecting to the device:133
09-21 16:57:26.305 19495-19964/com.veepoo.hband I/DfuBaseService: Disconnecting from the device...
09-21 16:57:26.315 19495-19964/com.veepoo.hband I/DfuBaseService: Refreshing result: true
09-21 16:57:26.315 19495-19964/com.veepoo.hband I/DfuBaseService: Cleaning up...

philips77, 2016-09-21:
Yes, the nRF Connect is also using the same library, v1.0.1.
I was just interested in the log from this app (not from LogCat, but the one available in the app after you are connected (menu->Show log)).
However, from the log you provided, I see you get 133 error after ~5 seconds from calling connecting to the device. This is timeout error, are you sure your device is advertising?

TimAimee, 2016-09-21:
they are two phone ,use the same app,first one is can't find server,the laster one is can't connect device(yes i sure the devcie is adversiting,on other phone can scaned "dufLang").,

TimAimee, 2016-09-21:
OK ,I get the nRF Connect log from the first phone. 
    nRF Master Control Panel, 2016-09-21
DfuLang (E6:81:02:E6:05:0F)
V	17:33:35.211	Connecting to E6:81:02:E6:05:0F...
D	17:33:35.756	gatt = device.connectGatt(autoConnect = false)
D	17:33:56.533	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	17:33:56.562	Disconnected
D	17:33:56.581	gatt.refresh()
D	17:33:56.613	gatt.close()

philips77, 2016-09-21:
The device that can't find the service might have cached previous services. To verify it and fix, please use the nRF Connect on this device, connect to your device, verify the service is listed. If no, go to menu->Refresh device services and check again. This will clear the cache and do service discovery again.
Also, make sure that if your device changes attribute table, it also changes the device address. 

For example: some implementations allow to switch to the DFU mode using a button. However, the phone is not aware of this change, so had the device didn't change the address, the phone would just take service from the cache (old ones). Therefore you need to force refreshing services. (in theory, if a device have the Service Changed characteristic in Generic Attribute service, and is not bonded, the phone shouldn't keep cache at all. It's a bug in Android.

philips77, 2016-09-21:
Are you sure you try to connect to the same device address?
Previously your device had address F9:C2:C4:DB:A2:96, but the DfuLang is E6:81:02:E6:05:0F. Or maybe you are just testing on several devices. But please, check it you try to connect to the same address as you can scan on another phone.

TimAimee, 2016-09-21:
i am testing  on several devices,i am sure the same device.

TimAimee, 2016-09-21:
what i do is  make the Oad  can Success  in the Multiple phone .i have tested the thirty phones.but we have four phone  most time is fail.

TimAimee, 2016-09-21:
what i do is make the Oad can Success in the Multiple phone .i have tested the thirty phones.but we have four phone most time is fail.another phone is always success.

philips77, 2016-09-21:
Hmm... I see 2 solutions then. First is stupid and simple - did you try to restart them and try again?
Second can take a long time - could you write the names of the phones on which the DFU fails? We will try to buy some and test is here in Norway. Hopefully something can be done. If not (e.g. the BLE on that phones is bad) they can be listed as not supported.

TimAimee, 2016-09-21:
all right,thanks,i will try again!the one of the  bad phone's name is vivo x5l.

TimAimee, 2016-09-28:
i want tell you a phenomenon，I user my APP can't OAD,then i user the nRF to connect the device which name is "dfuLang",it also hard to connect device and find the service, if  nRF connect device and find the service, then user my app can OAD success, so i think  the problem is what you have said the phone have cached previous services.so can you teach me whick time is for  refreshing cache? before/after connect or before disconnenct?

TimAimee, 2016-10-18:
I have fixed the problem. i just add the scan before the connect!

closed 2016-10-18 by TimAimee

philips77, 2016-10-18:
Hi, yes, scanning before connecting is important. If you create a BluetoothDevice object using new operator ```new BluetoothDevice(address)```, Android will assume that it's a public address (reason: how in real world someone would know a private address that changes?). If your device uses a private one, connection will fail despite that all 48 bits of the address are the same, as the public/private bit in Connect Req will not match. 
However, if a device was scanned at least once since boot, the type of the address is saved by the system and then, even if you create the BluetoothDevice with new, the Connect is using the correct address type.
All of this is because the BLE address is in fact 49 bit long.  In adv packets and connect req there is 49th bit specifying is the address public or private. You can't provide this info in android API.