nordicsemi/Android-DFU-Library #21: DFU disconnecting on Nougat, LG Nexus 5X

Iljo, 2016-09-23:
Version of DFU library is 1.0.3. OS build number on Nexus 5x is `NRD90R`, which is last official update.

Short after beggining of DFU process, when sending firmware to device, device disconnects from phone with status `8`. According to Androids BLE api, it's a status for error `A remote device connection is congested.` 

Comit before adef02a6ec49a65d826dda0b8ccdf0e7917eadbf, it's working fine.
So far this is happening only with Nougat on LG Nexus 5X. On the other hand, Nexus 6P, running Nougat, is working quite fine.

Here is part of stack trace, when error occures:
```
E/DfuActivity: Extras: DFU_PROGRESS_CHANGED; 14;
D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=DD:46:F1:C7:CE:F2
E/DfuBaseService: Connection state change error: 8 newState: 0
E/DfuImpl: Disconnected while sending data
E/DfuBaseService: Uploading Firmware Image failed: device disconnected
I/DfuBaseService: Cleaning up...
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=5
```

Currently we are fixing it by setting PRN in Androids `SharedPreference`, before every DFU, like so:
```java
PreferenceManager.getDefaultSharedPreferences(this)
        .edit()
        .putBoolean(DfuSettingsConstants.SETTINGS_PACKET_RECEIPT_NOTIFICATION_ENABLED, true)
        .putString(DfuSettingsConstants.SETTINGS_NUMBER_OF_PACKETS, String.valueOf(12))
        .commit();
```

If not reverting commit adef02a6ec49a65d826dda0b8ccdf0e7917eadbf, could PRN setting be an extra for an intent that's starting DfuBaseService, instead of value in preferences?

philips77, 2016-09-23:
Hi, 
Thanks for the report. I will test it on Nexus 5X in a second and come back to you. I did some testing on different phones and Android version, but not on Nexus 5X, of course... However, from what I saw in AOSP the buffer overflow has been fixed in Android 6.0, so PRNs should not be required. But let me check.
Regarding the PRN value - yes, it should be an extra. However, no to break backwards compatibility, I will have to check the ```SharedPreferences``` value as well when the extra is not set.

Iljo, 2016-09-23:
OK, thanks.

philips77, 2016-09-26:
Hi, I've tried DFU on Nexus 5X (mine build version is NRD90S) and had no problems. I managed to perform 2 tests: sending only an app and sending SD+BL+App. Both finished with success.

[image]

[image]

As you see from the graph PRNs were disabled. During SD+BL upload first it fills the whole buffer, then the buffer is partially-emptied before continuing.
I'm pretty much sure it is NOT the update to NRD90S which causes it to work on my side. Could you send my some hex files and zip so that I could try with your configuration?

philips77, 2016-09-26:
Hi, I've just published version 1.0.4 of the DFU Library.
Changes:
- one bug fixed in ArchiveInputStream
- Setting PRN state and value, and Force DFU flag from DfuServiceInitiator

Also, I did a lot of testing on Nexus 5X and I couldn't get error 8... it was working 100% times using different firmwares, in Legacy and Secure DFU. If it doen't work for you you should be able to set PRNs now with the initiator, as you requested.

KentaHarada, 2016-09-28:
Same problem occurred  on my Nexus 5X(Android ver 7.0 NRD90M), Nexus 5(ver6.0.1), Huawei Y6(ver 5.1).

Iljo, 2016-09-28:
@philips77 Unfortunately I can't send you firmware, because it's proprietary software. :(
I can send you screencast of DFU process using `nRF Connect` application, that fails when not using PRN. I'm uploading only application.

KentaHarada, 2016-09-30:
@philips77 
I've tried with following conditions on my Nexus 5X(ver 7.0 NRD90M) and other android devices.

DFU finished successfully with case 3 condition. I think that BLE buffer over flow
problem has not been fixed on marshmallow. How do you think ? 
If the bug fixed already, I will use case 3 condition on my Android Application.
Is that valid workaround? 

Case 1 
   PRN : Disabled
   PRN Value : -
   Result : Failed with error 8 short after beginning of DFU process.

Case 2 
   PRN : Enabled
   PRN Value : 1
   Result : DFU finished successfully, but my device didn't boot up.

Case 3
   PRN : Enabled
   PRN Value : 2
   Result : DFU finished successfully, but my device(nRF51) boot up normally.

philips77, 2016-09-30:
How about leaving PRN value at default 12? It's significantly faster. Also, it's really weird that the device hasn't reboot. Which SD and SDK are you using?

KentaHarada, 2016-09-30:
@philips77 
Thank you for your reply.
I tried with PRN default value 12. But failed with error 8.

And this is my device information.
nRF51822
SDK6.1.0
SoftDevice s110 7.3.0

KentaHarada, 2016-10-04:
@philips77 
I've tested DFU on several Android devices such as KitKat, Lollipop, Marshmallow and Nougat with following conditions. And  DFU finished successfully on all devices.

DFU Library : ver 1.0.4
PRN : Enabled
PRN Value : 2
Device : nRF51822, SDK6.1.0, SoftDevice s110 7.3.0

So, I plan to release my application with DFU library ver 1.0.4.
If you have any update or concern  about this issue, could you leave a comment ?

GHDreamer, 2016-10-20:
when i  develop the  DFU,a toast show me" DFU FILE NOT FOUND"  DfuServiceInitiator.setZip(A,B),A is the fileStream path or the file absolute path? please help.

philips77, 2016-10-20:
@GHDreamer please open another issue for this.

philips77, 2016-10-20:
I'm closing this issue.

closed 2016-10-20 by philips77