nordicsemi/Android-DFU-Library #22: OTA update filed

Kevin-Leung, 2016-09-26:
HI,my phone is Nexus 5X,I found recently when the progressbar is running about 30%-50% percent,It will stop to update the rom.but log is ok,I can't figure out what's going on.Could you please to check my problem.Other phones like HUAWEI honor and China native phone are no problems.
here is my usage:

>         final String path = StorageUtil.getSDPath() + StorageUtil.SDRootName + romName;
        final DfuServiceInitiator starter = new DfuServiceInitiator(
                device.getAddress()).
                setDisableNotification(true).// close  notifications
                setDeviceName(device.getName()).
                setKeepBond(true);
        starter.setZip(null, path);
        starter.start(context, DfuService.class);

here is my log:

> 09-26 13:51:44.703 10000-25142/com.opal.app I/DfuImpl: Reading DFU version number...
09-26 13:51:44.763 10000-25142/com.opal.app I/DfuImpl: Version number read: 0.8
09-26 13:51:44.764 10000-25142/com.opal.app I/DfuImpl: Enabling notifications...
09-26 13:51:44.764 10000-25142/com.opal.app D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
09-26 13:51:45.850 10000-25142/com.opal.app I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
09-26 13:51:45.910 10000-25142/com.opal.app I/DfuImpl: Sending image size array to DFU Packet (0b, 0b, 49804b)
09-26 13:51:51.898 10000-25142/com.opal.app I/DfuImpl: Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
09-26 13:51:51.963 10000-25142/com.opal.app I/DfuImpl: Sending 49804 bytes of init packet
09-26 13:51:51.963 10000-25142/com.opal.app I/DfuImpl: Sending init packet (Value = 03-00-03-00-11-00-00-01-01-00-64-00-02-00-00-00-8C-C2-00-00)
09-26 13:51:51.969 10000-25142/com.opal.app I/DfuImpl: Sending init packet (Value = 69-24-C9-DC-F1-A8-7E-3B-61-89-EC-26-DB-27-9F-BF-C9-07-17-70)
09-26 13:51:51.974 10000-25142/com.opal.app I/DfuImpl: Sending init packet (Value = 79-A1-B7-46-19-71-CB-1E-E5-F0-60-20-78-3D-71-96-63-D2-D4-FB)
09-26 13:51:51.980 10000-25142/com.opal.app I/DfuImpl: Sending init packet (Value = F4-D7-C3-F6-DF-57-DF-1F-26-CB-2B-20-C4-B8-38-D9-29-78-8B-1A)
09-26 13:51:51.983 10000-25142/com.opal.app I/DfuImpl: Sending init packet (Value = C3-70-B4-85-F8-0D-21-9D-FE-F9-97-35-C9-6F-4A-66-31-D8-9E-F8)
09-26 13:51:51.987 10000-25142/com.opal.app I/DfuImpl: Sending init packet (Value = 47-68-8B-62-FF-F7-42-41-71-0C-00-28-F3-F6-0E-E8)
09-26 13:51:51.992 10000-25142/com.opal.app I/DfuImpl: Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
09-26 13:52:19.754 10000-25142/com.opal.app I/DfuImpl: Sending Receive Firmware Image request (Op Code = 3)
09-26 13:52:19.815 10000-25142/com.opal.app I/DfuImpl: Uploading firmware...

philips77, 2016-09-26:
Hi, I think it may be related with PRNs. A bug in Android causing the outgoing buffer to overflow has been fixed in Android 6, so I disabled the PRNs by default on 6+, but it looks like Nexus 5X is not working properly. There is another issue regarding this here.
I'll debug it today and release a new version.
For now, as a workaround, you may set the SharedPreferences value related to it to "12" (String), and anther value to enable it. Check BaseCustomDfuImpl constructor for details.

Kevin-Leung, 2016-09-26:
OK.Thanks for your attention about this issue.

philips77, 2016-09-26:
Hi, Ive just tested it on Nexus 5X with Android 7.0 (NRD90S), security patch from 6 Sep 2016, and it managed to send the fw correctly.
I was sending the HRM_DFU sample from SDK 8 using Legacy DFU (the ble_app_hrm_dfu_s110_v8_0_0_sdk_v9_0.zip file from sdcard/Nordic Semiconductor/Board/pca10028/..).
The application bin is 20.85 KB.

philips77, 2016-09-26:
I've just release version 1.0.4. It shouldn't change the behaviour you experience (I couldn't replicate it on my Nexus 5X). However, it adds new methods to [DfuServiceInitiator](https://github.com/NordicSemiconductor/Android-DFU-Library/commit/c620ae46c8a2fdbc73ddfcb483a0ec6e379a2081). Using [setPacketReceiptNotificationsEnabled(boolean)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L104) method you force enabling notifications. This should help, I suppose.

Kevin-Leung, 2016-09-27:
thanks,So what I need is add the method  of setPacketReceiptNotificationsEnabled(boolean) to block the PRNs? I will test my application today to check the performance.

Kevin-Leung, 2016-09-27:
It's seems ok, It's not happened again.Thanks

philips77, 2016-09-27:
Great, closing issue.

closed 2016-09-27 by philips77