nordicsemi/Android-DFU-Library #24: i cant download dfu lib

navidrct, 2016-10-24:
hi
in this step

```
    Clone the project, or just the DFULibrary folder (using sparse-checkout) to a temporary location.
    Copy the DFULibrary folder to your projects root, for example to AndroidstudioProjects.
    Add the dfu module to your project:
        Add '..:DFULibrary:dfu' to the settings.gradle file: include ':app', '..:DFULibrary:dfu'
        Open Project Structure -> Modules -> app -> Dependencies tab and add dfu module dependency. You may also edit the build.gradle file in your app module manually by adding the following dependency: compile project(':..:DFULibrary:dfu')

```
i did not find DFULibrary folder to copy it to my project.where can i download it?
there is not direct download in [here](https://bintray.com/nordic/android/dfu-library/_latestVersion)

Tychus, 2016-10-24:
add this to your repositories in build.gradle (project) 
```
        maven {
            url  "http://dl.bintray.com/nordic/android"
        }
```
then this to your build gradle (app) 
`compile 'no.nordicsemi.android:dfu:1.0.4'`
that should get you the latest version.

navidrct, 2016-10-24:
I did it and now i can use their function . but where is the library . its not showing up in my project

navidrct, 2016-10-24:
is there any documentation or tutorial about it for me to use it?
im new to android programing.

Tychus, 2016-10-24:
Documentation is included in the repository. using it is pretty straight forward pretty much all you have to do is in the Usage section here 
https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation

navidrct, 2016-10-24:
thanks a lot

closed 2016-10-24 by navidrct