nordicsemi/Android-DFU-Library #25: correct if im wrong pls

navidrct, 2016-10-24:
hi
i don't know if i allow to ask such a q in here .  if i should not please tell me . thanks .

i use sdk v6 and it can be updated with hex file so i put .hex file in asset folder.
```
final DfuServiceInitiator starter = new DfuServiceInitiator(mdevice.getAddress())
                    .setDeviceName(mdevice.getName())
                    .setKeepBond(true);
```
with above code i init the service initiator adn then 

`starter.setZip("file:///android_asset/bleapp.hex");`

in code above i set the file in asset folder

but in code below it show me a null exception
`starter.start(this, DfuService.class);`

my DfuService class is copy of documentation code.

philips77, 2016-10-24:
Hi, it is possible that the lib does not support reading files from assets folder. If you want your hex file be built-in in the app, place it in the res/raw folder and then use setZip(R.raw.my_file) instead.
Also, could you paste the exception stack trace here?

navidrct, 2016-10-24:
i change it to raw folder and it get me this exception
```
10-24 13:38:13.239 1638-1638/com.asidesigner.smartlamp W/System.err: java.lang.NullPointerException
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.content.ContextWrapper.getPackageName(ContextWrapper.java:135)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.content.ComponentName.<init>(ComponentName.java:77)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.content.Intent.<init>(Intent.java:4064)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at no.nordicsemi.android.dfu.DfuServiceInitiator.start(DfuServiceInitiator.java:322)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at com.asidesigner.smartlamp.BLE.BLE.DFU_Update(BLE.java:459)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at com.asidesigner.smartlamp.MainActivity.update_hardware(MainActivity.java:268)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invokeNative(Native Method)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invoke(Method.java:515)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.support.v7.app.AppCompatViewInflater$DeclaredOnClickListener.onClick(AppCompatViewInflater.java:288)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.view.View.performClick(View.java:4485)
10-24 13:38:13.244 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.view.View$PerformClick.run(View.java:18827)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.os.Handler.handleCallback(Handler.java:808)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:103)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.os.Looper.loop(Looper.java:193)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5322)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invokeNative(Native Method)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invoke(Method.java:515)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:829)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:645)
10-24 13:38:13.245 1638-1638/com.asidesigner.smartlamp W/System.err:     at dalvik.system.NativeStart.main(Native Method)
```

philips77, 2016-10-24:
Did you add your DFUService class to Android manner?

philips77, 2016-10-24:
Manifest*

navidrct, 2016-10-24:
no i didnt add it but now that i add it its the same exception .
```
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err: java.lang.NullPointerException
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.content.ContextWrapper.getPackageName(ContextWrapper.java:135)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.content.ComponentName.<init>(ComponentName.java:77)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.content.Intent.<init>(Intent.java:4064)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at no.nordicsemi.android.dfu.DfuServiceInitiator.start(DfuServiceInitiator.java:322)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at com.asidesigner.smartlamp.BLE.BLE.DFU_Update(BLE.java:459)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at com.asidesigner.smartlamp.MainActivity.update_hardware(MainActivity.java:268)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invokeNative(Native Method)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invoke(Method.java:515)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.support.v7.app.AppCompatViewInflater$DeclaredOnClickListener.onClick(AppCompatViewInflater.java:288)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.view.View.performClick(View.java:4485)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.view.View$PerformClick.run(View.java:18827)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.os.Handler.handleCallback(Handler.java:808)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:103)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.os.Looper.loop(Looper.java:193)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5322)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invokeNative(Native Method)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at java.lang.reflect.Method.invoke(Method.java:515)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:829)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:645)
10-24 14:04:43.933 18980-18980/com.asidesigner.smartlamp W/System.err:     at dalvik.system.NativeStart.main(Native Method)
```
i clean and recompile the project to be sure

philips77, 2016-10-24:
Then it's not DFU lib issue.
Please search you problem in the Internet, for example here: http://stackoverflow.com/questions/16745806/java-lang-nullpointerexception-at-android-content-contextwrapper-getpackagename

closed 2016-10-24 by philips77

Tychus, 2016-10-24:
use the context to start the service. and if you are using a hex file setZip is not the method to use.