nordicsemi/Android-DFU-Library #29: 19 errors: toast show"Upload failed: DFU SERVICE NOT FOUND"

GHDreamer, 2016-11-01:
when i press the update button,this is the logcat info:

11-01 13:53:16.611 20786-20786/xxxxos E/ContentValues: mFileStreamUri=: file:///storage/emulated/0/firmware.zipmFilePath=/storage/emulated/0/firmware.zip
11-01 13:53:18.731 20786-22447/xxxxos D/BluetoothGatt: connect() - device: DA:7B:4D:51:CB:4C, auto: false
11-01 13:53:18.731 20786-22447/xxxxos D/BluetoothGatt: registerApp()
11-01 13:53:18.731 20786-22447/xxxxos D/BluetoothGatt: registerApp() - UUID=f3542e4f-e512-4aa0-9dd7-aebc94646e55
11-01 13:53:18.781 20786-21076/xxxxos D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
11-01 13:53:18.791 20786-20799/xxxxos D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DA:7B:4D:51:CB:4C
11-01 13:53:20.391 20786-20799/xxxxos D/BluetoothGatt: discoverServices() - device: DA:7B:4D:51:CB:4C
11-01 13:53:20.421 20786-20799/xxxxos D/BluetoothGatt: onSearchComplete() = Device=DA:7B:4D:51:CB:4C Status=0
11-01 13:53:20.591 20786-22447/xxxxos D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
11-01 13:53:20.891 20786-22447/xxxxos D/BluetoothGatt: cancelOpen() - device: DA:7B:4D:51:CB:4C
11-01 13:53:20.901 20786-20799/xxxxos D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DA:7B:4D:51:CB:4C
11-01 13:53:20.901 20786-22447/xxxxos D/BluetoothGatt: close()
11-01 13:53:20.901 20786-22447/xxxxos D/BluetoothGatt: unregisterApp() - mClientIf=6
11-01 13:53:22.991 20786-22447/xxxxos D/BluetoothGatt: connect() - device: DA:7B:4D:51:CB:4C, auto: false
11-01 13:53:22.991 20786-22447/xxxxos D/BluetoothGatt: registerApp()
11-01 13:53:22.991 20786-22447/xxxxos D/BluetoothGatt: registerApp() - UUID=f3b56e3d-39ec-4232-a554-f01cda9335cb
11-01 13:53:23.041 20786-20801/xxxxos D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
11-01 13:53:23.051 20786-20801/xxxxos D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DA:7B:4D:51:CB:4C
11-01 13:53:24.651 20786-20801/xxxxos D/BluetoothGatt: discoverServices() - device: DA:7B:4D:51:CB:4C
11-01 13:53:24.681 20786-20801/xxxxos D/BluetoothGatt: onSearchComplete() = Device=DA:7B:4D:51:CB:4C Status=0
11-01 13:53:24.891 20786-20786/xxxxos E/ContentValues: onDeviceConnected: DA:7B:4D:51:CB:4C
11-01 13:53:26.051 20786-22447/xxxxos D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
11-01 13:53:28.201 20786-21076/xxxxos D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=6 device=DA:7B:4D:51:CB:4C
11-01 13:53:28.201 20786-21076/xxxxos E/DfuBaseService: Connection state change error: 19 newState: 0
11-01 13:53:28.201 20786-22447/xxxxos D/BluetoothGatt: close()
11-01 13:53:28.201 20786-22447/xxxxos D/BluetoothGatt: unregisterApp() - mClientIf=6
11-01 13:53:28.661 20786-20786/xxxxos D/MultiPhoneWindow: MinimizeAnimator::removeWindow
11-01 13:53:28.671 20786-20786/xxxxos D/MultiPhoneWindow: MinimizeAnimator::removeWindow
11-01 13:53:30.361 20786-22447/xxxxos D/BluetoothGatt: connect() - device: DA:7B:4D:51:CB:4C, auto: false
11-01 13:53:30.361 20786-22447/xxxxos D/BluetoothGatt: registerApp()
11-01 13:53:30.361 20786-22447/xxxxos D/BluetoothGatt: registerApp() - UUID=c534be4e-f0f0-463b-9f86-33b1d775e489
11-01 13:53:30.421 20786-20799/xxxxos D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
11-01 13:53:30.421 20786-20801/xxxxos D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DA:7B:4D:51:CB:4C
11-01 13:53:32.031 20786-20801/xxxxos D/BluetoothGatt: discoverServices() - device: DA:7B:4D:51:CB:4C
11-01 13:53:32.031 20786-20799/xxxxos D/BluetoothGatt: onSearchComplete() = Device=DA:7B:4D:51:CB:4C Status=0
11-01 13:53:32.061 20786-22447/xxxxos D/BluetoothGatt: cancelOpen() - device: DA:7B:4D:51:CB:4C
11-01 13:53:32.071 20786-20799/xxxxos D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DA:7B:4D:51:CB:4C
11-01 13:53:32.091 20786-22447/xxxxos D/BluetoothGatt: close()
11-01 13:53:32.091 20786-22447/xxxxos D/BluetoothGatt: unregisterApp() - mClientIf=6
11-01 13:53:32.871 20786-20786/xxxxos D/SRIB_DCS: log_dcs ThreadedRenderer::initialize entered!
11-01 13:53:32.871 20786-21374/xxxxos D/mali_winsys: new_window_surface returns 0x3000, [1071x154]-format:1
11-01 13:53:32.911 20786-21374/xxxxos V/RenderScript: 0xeec11000 Launching thread(s), CPUs 8

1,I doubt why I have call the DfuServiceInitiator,start() method, it still show me "Upload failed:DFU SERVICE NOT　FOUND"，Is it necessary to keep the dfuservice alive longly or How to initial the dfuservice? Do you have any ideas?

2,I want to  show my thought,Anybody can  show me  ,is there a problem or  mistake I have made?
I copy the  DFU package(firmware.zip) to the res/raw ,i call a method to copy  the .zip to the phone root directory, when i get the fileStreamUri and  mFilePath base on the .zip file. Finaly, i copy these codes:

  DfuServiceInitiator starter = new DfuServiceInitiator(DEVICE_ADDRESS)
                .setDeviceName(DEVICE_NAME)
                .setKeepBond(true);

        // Init packet is required by Bootloader/DFU from SDK 7.0+ if HEX or BIN file is given above.
        // In case of a ZIP file, the init packet (a DAT file) must be included inside the ZIP file.
        if (mFileType == TYPE_AUTO) {
            starter.setZip(mFileStreamUri, mFilePath);
            Log.e(TAG, "mFileStreamUri=: " + mFileStreamUri + "mFilePath=" + mFilePath);
        } else {
            starter.setBinOrHex(mFileType, mFileStreamUri, mFilePath).setInitFile(mInitFileStreamUri, mInitFilePath);
            Log.e(TAG, "mFileStreamUri=: " + mFileStreamUri + "mFilePath=" + mFilePath + " mInitFileStreamUri=" + mInitFileStreamUri + " mInitFilePath=" + mInitFilePath);
        }
        starter.start(DFUActivity.this, DfuService.class);

This is  the parameters of  starter.setZip();
fileStreamUri=: file:///storage/emulated/0/firmware.zip
mFilePath=/storage/emulated/0/firmware.zip

2,when  i search in the  "storage/emulated/0/",there is no firmware.zip ,but there is  one   in the  root dirctory.is it  I  got  the  wrong path? Next is  the  code to  copy the firmware.zip  and  get  the  uri and  path ,is there some mistakes?please  help...

   File   f = new File(Environment.getExternalStorageDirectory(), "firmware.zip");
        copyRawResource(R.raw.firmware, f);
        Log.e(TAG, "url=" + f.getAbsolutePath());
        mFileTypeTmp = TYPE_AUTO;

        mFileStreamUri = Uri.fromFile(f);
        mFilePath = f.getAbsolutePath();

philips77, 2016-11-02:
Hi, 
1. The DFU service that hasn't been found is the one on the peripheral device. In order to perform DFU make sure there is such service on your device. Make sure the cache is refreshed if you flashed a new application on your device. It's not about Android DFU Service.
2. There is a method in there initiator to give zip from resources. Just use setZip(R.raw.firmware) and don't copy it. You'd have to request Storage permission etc. or use your app private folder.
3.  Error 19 which you got is GATT CONN TERMINATE PEER USER.

GHDreamer, 2016-11-03:
@philips77  Thanks you very much for your reply.
          Now，I can update peripheral device successfully,but another problem appeared. After the DFU process,when i call  the "write order " method to get the physical datas,the response code always be -1and response bytes array is null.  I use the[ BluetoothKi](https://github.com/dingjikerbo/BluetoothKit)t to read/write order to the  peripheral device. If i disable the DFU method, read/write can finish successfully.Could you tell me what should i do when i  finish the DFU?   I doubt if i forget call some methods  to set the mode of  peripheral device to  normal(can read/write) or anything elses.Is there any  suggestions about  that problem?

philips77, 2017-07-28:
Hi, sorry for the delay. Did you manage to solve your problem?

TomorrowDreamer, 2018-03-01:
@philips77  yes，  thank you all the same

closed 2018-03-01 by philips77