nordicsemi/Android-DFU-Library #30: Gradle error: "Configuration with name 'default' not found."

eliotstock, 2016-11-10:
Just got this when following your instructions for adding this to an existing project in Android Studio. I'm on the latest Android Studio (2.2.2) and gradle 2.14.1.

I can add this line to the dependencies bit of my app/build.gradle and it will build fine:

    compile 'no.nordicsemi.android:dfu:1.0.4'

But when I add this:

    compile project(':..:DFULibrary:dfu')

I get the error.

eliotstock, 2016-11-10:
Oh, wait a minute. I didn't properly follow your instructions, sorry. Are these steps necessary, even though I'm pulling this in from the Maven Central repo?

1. Clone the project, or just the DFULibrary folder (using sparse-checkout) to a temporary location.
2. Copy the DFULibrary folder to your projects root, for example to AndroidstudioProjects.

philips77, 2016-11-10:
Nope, they are only if you want to include the lib into your project from source (cloned as a separate project into DFULibrary folder). I should make it more clear in the docs (I guess the docs hasn't been updated since I published it on jcenter).

eliotstock, 2016-11-10:
OK, in that case I still have an issue. Any ideas?

philips77, 2016-11-10:
Done, docs updated.

philips77, 2016-11-10:
Then you also have to check the settings.gradle file and add:
```
include ':dfu'
project(':dfu').projectDir = file('../DFULibrary/dfu')
```

eliotstock, 2016-11-10:
Oh! Well that was easy. Thanks for the quick response.

closed 2016-11-10 by eliotstock