nordicsemi/Android-DFU-Library #33: Buttonless jump for Secure DFU

philips77, 2016-11-11:
Hello All DFU Users,

Currently the buttonless jump to DFU mode is not supported for Secure DFU on Android or iOS DFU Libraries. In the SDK 12.1 this feature is experimental and it may change in the future until it's stable.
As far as I know the final characteristic will have the following UUID:
8EC9**0003**-F315-4F60-9FB8-838830DAEA50
and will be inside a service with DFU service UUID (so 16-bit 0xFE59 for Secure DFU).

This characteristic will have Write and Notify properties.
To protect from DOS attack writing to it will require encryption (so bonded devices only).

Accepted write value: 
0x01 = Enter Bootloader on next reset (application specific to handle reset).
Response Codes: 
<Response Op Code><Request Op Code><UInt16 Status Code> - These values are not settled yet.

So the DFULibrary for both Android and iOS fill eventually support such buttonless service, not the one from the SDK 12.1. The ETA is somewhere in the "beginning" of 2017, hopefully.

philips77, 2016-11-11:
This is related to #31

mogstad, 2016-11-15:
@philips77 do you know if bond sharing will return in the 12.1 SDK as well?

philips77, 2016-11-15:
As far as I know sharing bond info will be supported in the final buttonless service in January+. It will be sdk 13 or more, depending how many SDKs will be until then.
12.x will not support it.

philips77, 2016-11-18:
**Important note regarding the current experimental Buttonless DFU Service.**

The service is experimental. It means it wasn't tested properly and may have bugs (and does have!). Let me write down here those I know:
1. The service does an address trick instead of sending Service Change indication on bonded devices after switching to bootloader mode. The bootloader address is incremented by 1 so it's visible for central devices as another device. The bond information are not copied to the bootloader.
2. The service does not required bonding, so anyone can send value 01 to the characteristic inside it and switch the device to bootloader mode. Even if the update would not be possible the device is now unusable unless it switches back to app mode. This may be used for DOS (Denial Of Service) attack.
3. The characteristic inside this service should have Notify and Write properties (ble_dfu.c, line 90). By default by mistake it has only Notify property. Add ```char_md.char_props.write = 1;``` to add Write property.
4. The buttonless example has a collition on code size and reserved flash pages which makes it impossible to flash it onto an nRF52. You can DFU it in, but not flash it. Info: https://devzone.nordicsemi.com/question/100609/sdk-12-bootloader-erased-after-programming/ 
5. Even the documentation for this service: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v12.0.0/ble_sdk_app_buttonless_dfu.html?cp=4_0_0_4_3_2 says it's a Heart Rate profile...

philips77, 2017-03-22:
In SDK 13 tbe buttonless service is still experimental but the API (UUIDs, commands) become final.
It has also been decided to have 2 different characteristic UUIDs: 
- one for non-bonded devices (8EC9**0003**-F315-4F60-9FB8-838830DAEA50) and 
- one for bonded (8EC9**0004**-F315-4F60-9FB8-838830DAEA50) and 

Such characteristic will be in a service with the standard 16-bit Secure DFU service UUID (**FE59**).

Be aware, that the first one, for non-bonded devices, may be used to put your device in the DFU mode by anyone that is close enough and connects to your device. It also may be used on bonded devices, but the bond information is not shared to the bootloader, therefore after switching to DFU mode it starts advertising with address +1. It is recommended to add security on this characteristic or set some timeout for the DFU mode to change it back to app mode after a timeout (or some other solutions that would minimize the risk of DoS). In the SDK 13 only this service has been implemented and it is still in experimental state (not properly tested), but at least doesn't have this compilation issues that were in SDK 12 (see posts above).
The service that will share bond info with the bootloader is still under development and may come in SDK 14 or later.
Both will be supported in DFU Library 1.3.0+ on Android and DFU Lib for iOS 3.1.0+.

philips77, 2017-04-07:
Released in version 1.3.0

closed 2017-04-07 by philips77