nordicsemi/Android-DFU-Library #36: Device does not enter DFU mode

chuks, 2016-12-01:
I've tried using this library in an Android project, but have had no luck with the device entering DFU mode. I downloaded and tried the Android Library Toolbox, but that also does not get the device into DFU mode. Digging through the code, I see that the value 0x04 is being send to the device to set it to DFU, but it looks like that isn't happening. This is using legacy DFU, not secure.

The nrf Connect Android app, on the other hand, successfully updated the firmware of the device. What would be the reason for this?

philips77, 2016-12-01:
Both nRF Toolbox and nRF Connect are using the same DFU Library, currently in version 1.1.0. Their behaviour shoudl be the same in this scope.
To switch to the DFU bootloader in Legacy dfu one needs to enable notifications and send 0x01-04 to DFU Control Point char.

chuks, 2016-12-02:
Hi, thanks for responding. What I did, was modify the part if the code that sends the 0x01 and 0x04 manually, and I can get it to DFU now . However, I an facing another problem. It seems to proceed normally until it tries to discover services after connecting. It sits there for a while, then it errors out with a disconnect ("disconnect before services discovered"). I can't get past this point.

chuks, 2016-12-02:
BTW, I'm using Legacy DfU.

philips77, 2016-12-02:
It seems like a problem on the peripheral side. Can you replicate it with nRF Connect by connecting, switching to bootloader, connecting again etc?
What error do you get instead of services?

chuks, 2016-12-02:
Interestingly, the same problem occurs with the DFU toolkit Android app. However, I am able to update successfully using the full nRF Connect app. Always failing with disconnecting before discovered services. For some reason, after the device reboots into DFU mode, the app successfully connects to the device, then tries to discover services. It seems to hang here, then I suppose it times out after a while with a disconnect, giving that error. This occurs all the time, I haven't been able to get past this, on both the toolkit app and a custom app I'm working on using the library.

philips77, 2016-12-02:
Still I think it's something on the peripheral side.
I need more info: what sdk do you forked from, do you use bonding, dfu you have Service Changed characteristic and send indication from it if bonded, what does the log in logcat and nRF Connect say? Can you replicate the error in nRF Connect by sending messages one by one? And additional info that may by important.

chuks, 2016-12-02:
SDK 8.0, and no bonding. However, I have Service Changed characteristic and also send the indications (that's how I was able to get it to DFU, for some weird reason). Basically, the steps taken were:
1 - Connect to device
2 - Turn on indications on Service changed characteristic
3 - Turn on notifications on the DFU 
4 - Send DFU signal (0x01 - 0x04). This immediately takes it to DFU mode.
5 - Restart library service
6 - library tries to reconnect to device. After a short while, connection succeeds. It's failed a few times, but I can't reproduce the connection failure. It now always reconnects.
7 - Service tries to discover services, and it gets stuck here (discovering services)
8 - After a little while, the connection breaks, and I get that error. 

See relevant logs:
`12-01 18:01:07.745 2418-3585/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: cancelOpen() - device: C0:A9:66:88:EE:29
12-01 18:01:07.747 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: waitUntilDisconnected
12-01 18:01:07.747 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: Disconnected
12-01 18:01:07.747 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] Disconnected
12-01 18:01:07.747 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: gatt.refresh() (hidden)
12-01 18:01:07.747 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] gatt.refresh() (hidden)
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: refresh() - device: C0:A9:66:88:EE:29
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Refreshing result: true
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Cleaning up...
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: gatt.close()
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] gatt.close()
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: close()
12-01 18:01:07.748 2418-3585/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: unregisterApp() - mClientIf=5
12-01 18:01:07.749 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Restarting the service
12-01 18:01:07.753 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: DFU service started
12-01 18:01:07.753 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] DFU service started
12-01 18:01:07.753 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: Opening file...
12-01 18:01:07.753 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] Opening file...
12-01 18:01:07.753 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: Opening file...using fileUri: content://com.android.providers.downloads.documents/document/23
12-01 18:01:07.753 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] Opening file...using fileUri: content://com.android.providers.downloads.documents/document/23
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: The new firmware is word-aligned.
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] The new firmware is word-aligned.
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: fileType == TYPE_AUTO and MIME_TYPE_ZIP: application/zip
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] fileType == TYPE_AUTO and MIME_TYPE_ZIP: application/zip
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: MIME_TYPE_ZIP = application/zip
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] MIME_TYPE_ZIP = application/zip
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: fileType is TYPE_APPLICATION
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] fileType is TYPE_APPLICATION
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: Image file opened (35992 bytes in total)
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] Image file opened (35992 bytes in total)
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: wait(1000)
12-01 18:01:07.788 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] wait(1000)
12-01 18:01:08.287 2418-2418/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
12-01 18:01:08.305 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: wait(1000)
12-01 18:01:08.305 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] wait(1000)
[... cut ...]
12-01 18:01:38.065 2418-3585/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: unregisterApp() - mClientIf=5
12-01 18:01:38.074 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - message: wait(600)
12-01 18:01:38.074 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendLogBroadcast - fullMessage: [DFU] wait(600)
12-01 18:01:38.675 2418-3585/no.nordicsemi.android.nrftoolbox I/DfuBaseService: sendErrorBroadcast
12-01 18:01:38.801 2418-2452/no.nordicsemi.android.nrftoolbox V/RenderScript: 0x9e142000 Launching thread(s), CPUs 4`

philips77, 2017-02-23:
Hi, sorry for coming back so late. Did you manage to solve the problem?

closed 2017-03-22 by philips77

chuks, 2017-03-22:
It still happens occasionally. I'm not always successful. Digging though the library code, I see there are a lot of locks used for asynchronous calls. I'd have preferred if queues were used instead. Still, it's a problem.

zinwalin, 2021-08-28:
> Interestingly, the same problem occurs with the DFU toolkit Android app. However, I am able to update successfully using the full nRF Connect app. Always failing with disconnecting before discovered [...]

exactly same behaviors here.