nordicsemi/Android-DFU-Library #37: [ANDROID] DFU file error 4098 while initializing DFU service / An exception occurred while calculating file size

schech-a, 2016-12-11:
Hi all,

I get an error while trying to make a DFU on my device, but the thing is that it only happens with the RELEASE APK. In debug apk, it works like a charm.

I get this error :  E/DfuBaseService: An exception occurred while calculating file size. 
It seems to be corresponding to an error code 4098.

I checked if the firmware file was correctly found in release mode by simply displaying the raw content and file size in the LOG. And it does appear in the log, meaning that the file is found,with the correct size.

This is the first project on which I get this weird behavior....

Here is how I call the DFU service :
 new DfuServiceInitiator(adress)
                .setDeviceName(device.getName())
                .setKeepBond(true)
                .setZip(R.raw.my_firmware)
                .start(context, DfuService.class);

Any idea ? Am I missing a stupid thing like a permission in the manifest, that only breaks when in release mode?

Thanks in advance for your help :)

philips77, 2016-12-12:
Hi, in your DFUService class override the isDebug() method and return true in release Britain as well.
Then build a signed release version of your app and check out the logcat. You should get an IOException printed there. It's a generic exception thrown when reading the file fails and knowing details will help.

schech-a, 2016-12-12:
Hi,
Thanks for your answer.

I can not manage to find any debug fonction to override in my DfuBaseService extend.
@Override
    protected boolean IsDebug() {return true;}

I looked up in the DfuBaseFile. I am working with a 0.6 version, maybe that is why there is no such fonction. (I cant update the DFU lib for now, but the issue appears on the latest version as well)

This is the complete error that I get : 
java.io.IOException: The ZIP file must contain an Application, a Soft Device and/or a Bootloader.
                                                       at e.a.a.a.a.a.<init>(ArchiveInputStream.java:237)
                                                       at e.a.a.a.a.a(DfuBaseService.java:2120)
                                                       at e.a.a.a.a.onHandleIntent(DfuBaseService.java:1200)
                                                       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:65)
                                                       at android.os.Handler.dispatchMessage(Handler.java:102)
                                                       at android.os.Looper.loop(Looper.java:145)
                                                       at android.os.HandlerThread.run(HandlerThread.java:61)

schech-a, 2016-12-12:
Hi,
Thanks for your answer.

This is the complete error that I get : 

java.io.IOException: The ZIP file must contain an Application, a Soft Device and/or a Bootloader.
                                                       at e.a.a.a.a.a.<init>(ArchiveInputStream.java:237)
                                                       at e.a.a.a.a.a(DfuBaseService.java:2120)
                                                       at e.a.a.a.a.onHandleIntent(DfuBaseService.java:1200)
                                                       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:65)
                                                       at android.os.Handler.dispatchMessage(Handler.java:102)
                                                       at android.os.Looper.loop(Looper.java:145)
                                                       at android.os.HandlerThread.run(HandlerThread.java:61)

philips77, 2016-12-12:
Hello,
First of all I highly recommend to updating you library dependency to the most recent version. It is fully backwards compatible so should work in your project, unless you made some modifications on your own. But that's a minor.

From the output you gave I see, that you are using a ZIP file without the manifest file, using the 'old method' that bases on file names. Then the hex/bin and dat files in the ZIP must have proper names: application.hex/bin/dat or system.hex/bin/dat. But again, was it intentional?

Also, if the above is not true, it may be that reading the ZIP file failed. Reasons I can see are: proguard, some encryption of the R.raw folder, ZIP file was opened before and not closed? I think the ```getResources().openRawResource(resId);``` may reuse the same cached stream or something like that? Could you do some debugging and see if [this](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/v0.6.3/dfu/src/main/java/no/nordicsemi/android/dfu/internal/ArchiveInputStream.java#L255) method finds something?

schech-a, 2016-12-13:
Changing the name of the files worked. I still don't understand why it was only working in debug though...

Thank you very much Phillip!

closed 2016-12-13 by schech-a

varunnarisetty, 2017-11-13:
I have the same problem. and I think I found the reason why it fails in release version but not on debug.
Its because of proguard. When proguard obfuscate the code for some reason DFU lib was not able to find the manifest file inside the zip file. so according to the Nordic code mentioned above if it don't find the manifest file it will look for at least one of these files  softdevice.bin/hex, bootloader.bin/hex, application.bin/hex. 
To fix this all we need to do is add this line to the proguard-rules.pro file
-keep class no.nordicsemi.android.dfu.** { *; }

pmahsky, 2019-04-04:
@varunnarisetty I was facing the DFU File Error issue. This proguard rule helped me out. Thanks.

suyxin, 2019-08-19:
it works for me! thank you very much!

suyxin, 2019-08-19:
> I have the same problem. and I think I found the reason why it fails in release version but not on debug.
> Its because of proguard. When proguard obfuscate the code for some reason DFU lib was not [...]

it works for me! thank you very much!

rodrigosimoesrosa, 2019-11-08:
It works for me! Thank you @varunnarisetty