nordicsemi/Android-DFU-Library #38: Device returned error after sending file (error 6)

LouisLuo2016, 2016-12-14:
12-14 13:47:42.991 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Connecting to the device...
12-14 13:47:43.033 6069-6091/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Connected to GATT server
12-14 13:47:43.040 6069-6091/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Attempting to start service discovery... succeed
12-14 13:47:43.558 6069-6090/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Services discovered
12-14 13:47:44.610 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Reading DFU version number...
12-14 13:47:44.651 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Version number read: 0.8
12-14 13:47:44.651 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Enabling notifications...
12-14 13:47:45.691 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
12-14 13:47:45.726 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending image size array to DFU Packet (0b, 0b, 142752b)
12-14 13:47:56.271 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
12-14 13:47:56.300 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending 142752 bytes of init packet
12-14 13:47:56.300 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending init packet (Value = FF-FF-FF-FF-FF-FF-00-00-01-00-64-00-AB-E9)
12-14 13:47:56.304 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
12-14 13:47:56.361 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending the number of packets before notifications (Op Code = 8, Value = 12)
12-14 13:47:56.392 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending Receive Firmware Image request (Op Code = 3)
12-14 13:47:56.430 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Uploading firmware...
12-14 13:47:57.891 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Response received. Op Code: 16 Req Op Code = 3, Status = 6
12-14 13:47:57.891 6069-6797/no.nordicsemi.android.nrftoolbox E/DfuImpl: Device returned error after sending file (error 6)
12-14 13:47:57.892 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuImpl: Sending Reset command (Op Code = 6)
12-14 13:48:10.723 6069-6110/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Disconnected from GATT server
12-14 13:48:10.728 6069-6069/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
12-14 13:48:10.750 6069-6797/no.nordicsemi.android.nrftoolbox I/DfuBaseService: Disconnecting from the device...

philips77, 2016-12-14:
Looks like your device can't handle that many packets per interval connection. Lower the prn number to 1 and try again, then try with a bit higher values. Something like 6 should also work.

LouisLuo2016, 2016-12-14:
Hi,Philips
     This vale of number of package is perference in nrf-toobox.
I don't find it used to dfu. Where is used SETTINGS_NUMBER_OF_PACKETS perference?

philips77, 2016-12-14:
This value may be given as a service Extra (preferred way) or as setting. Here it is being read: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/BaseCustomDfuImpl.java#L227

philips77, 2017-02-23:
Hi, did you manage to fix your problem?

licl1008, 2017-03-14:
Device returned error after sending init packet (error 6)  How about this problem?  What reasons may leads this problem?

philips77, 2017-03-14:
It most probably mean that the signature was not valid. Did you use the correct private key to sign it matching the public key that you had placed in the bootloader code?

philips77, 2017-03-14:
Btw, the Secure DFU bootloader from SDK 13 (released yesterday) returns Extended error code instead with failure reason.

licl1008, 2017-03-14:
@philips77  This problem didn't appear every time when I upgrade firmware. If I restart my app again, sometime the SDK can works well. I think if the key you told me is not correct, SDK will can't work always. ???

philips77, 2017-03-14:
Are you using Legacy or Secure DFU?

licl1008, 2017-03-14:
Legacy

philips77, 2017-03-14:
Hmm.. yes. Then it would always fail. The Operation Failed (error 6) may be sent when some validation on Init packet fails: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk51.v9.0.0/bledfu_example_init.html 
Are you sure you are sending the fw to the same device and it has the same configuration every time? Perhaps it fails when you upload it to already upgraded version, or something like that? You would have to get more logs from the device to check the reason. And please, write a case on our DevZone as this is not a library issue, but a DFU question and our Support team will be much more capable of answering your question :)

licl1008, 2017-03-14:
Yes, because of test,  I upgraded fw file which  has upgraded in my product.  What you mean is if I use  fw_0.2 now,  some day I want use fw_0.1 again(fw_0.1 has written to product before), it will leads this problem?  I have write this question in DevZones.  Thks a lot!!!

philips77, 2017-03-14:
As far as I know the app version specified in Init packet was not checked by default in Legacy DFU bootloader (a TODOL implement me) was left for developers. That should prevent from downgrading the fw if this is a requirement. By default only hw type and revision and the Required SoftDevice are checked against the values from some registers. But you will get much more details on DevZone.

closed 2017-07-28 by philips77