nordicsemi/Android-DFU-Library #40: Notification click not working

tomasbedrich, 2017-01-22:
Hello, thanks for your great library.

I have a small problem. When I extend your `DfuBaseService` and implement a `getNotificationTarget` method, my `NotificationActivity` is not created (not even *instantiated*) when clicking on the notification. On the other hand, the *Abort* action works fine. My `NotificationActivity` is almost a copy-paste from [your example](https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation#usage). I have traced down the problem [here](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1414), but I am an Android noob, so sorry that I cannot propose a solution.

Tested on Sony Xperia Z5 (Android 6.0.1) and Nexus 5 (Android 7.1.1).

philips77, 2017-01-22:
Did you add it to the Android manifest?

tomasbedrich, 2017-01-22:
Thank you very much @philips77! That was really a noob mistake. 😀  I was missing an `<activity>` record in the `AndroidManifest.xml`. I followed the guide where is only noted to add a service.

Maybe it is worth adding it to the guide too.

closed 2017-01-22 by tomasbedrich

philips77, 2017-01-22:
Good idea. I'll update it on the next occasion.