nordicsemi/Android-DFU-Library #42: GATT Invalid Handle

gcerto, 2017-02-15:
Hello,

We are currently using your Android lib and we are experiencing the following behaviour:
1. When we connect after starting the DFU with a "button", we don't have any issue. We connect to our device on the Mac Address (+1) and device name "DfuTarg".
2. When we use the method "button less", we always get the same following issue: "GATT INVALID HANDLE".

We are using the following method:
- With our BLE lib, we connect to the GATT peripheral.
- We list the services.
- We write an own defined characteristic to allow the DFU.
- The device is then ready to start in DFU mode.
- We disconnect from the peripheral.

Thereafter, we use you library in the following way (same code as the one use in behaviour 1):

```
public void startUpdate(String macAddress, String deviceName, String filePath)
{
    if (m_dfuServiceController == null)
    {
        final DfuServiceInitiator starter = new DfuServiceInitiator(macAddress)
                .setDeviceName(deviceName)
                .setKeepBond(false);    // keep bond??

        starter.setZip(null, filePath);
        m_dfuServiceController = starter.start(m_context, DfuService.class);
    }
}
```

We have tried several other solutions found around the web but with no luck.

We have checked the following:
- The peripheral exposes correctly the characteristic "Service Update" in the firmware.
- If we activate our own flag through a generic app (like lightblue) and we only do the call to your library, the outcome is the same.
- If we use the NRF Connect app, it works correctly.

Here are some logs:
02-15 16:12:53.784 INFO: 15-02-2017 04:12:53 : Scan started
02-15 16:12:58.812 INFO: 15-02-2017 04:12:58 : Scan interrupted
02-15 16:13:05.025 DEBUG: 15-02-2017 04:13:05 : Updater state : StartingDFU
02-15 16:13:07.389 DEBUG: 15-02-2017 04:13:07 : Updater state : EnablingDFU
02-15 16:13:09.825 DEBUG: 15-02-2017 04:13:09 : Updater state : DeviceConnecting
02-15 16:13:11.971 DEBUG: 15-02-2017 04:13:11 : Updater state : StartingDFU
02-15 16:13:13.069 DEBUG: 15-02-2017 04:13:13 : Updater state : DeviceDisconnecting
02-15 16:13:13.788 ERROR: 15-02-2017 04:13:13 : Updater Error : GATT INVALID HANDLE

Do you have any idea why we get such error?
What's the difference between what we do and what is done on the NRF Connect app? Are we missing any steps?

Thank you in advance for your help.

philips77, 2017-02-15:
Hi,
The difference is that you are switching to DFU on your own, not using the library for that. The DFU lib, which nRF Connect and nRF Toolbox are using, take care of clearing the cache after sending "jump to bootloader" command. If your device does not change device address when in bootloader mode (it does, when you switch mode using a button, that's why it works) but changes the services Android is not aware of it. It should not cache services if Service Changed char is present in the attribute list on the peripheral (for not-bonded devices) but Android always caches them (this bug will be fixed one day).

So the solutions are:
1. Use DFU lib to switch to DFU mode. If yo have the correct service (DFU Service) with "jump to BL mode" feature in the app mode, simple start DFU while being connected to it. DFU Lib will send correct command, clear cache, wait for disconnection and reconnect and continue with DFU.
2. Clear the cache after you disconnect, before you start DFU. Here's how we do it: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1318

YannSDW, 2017-02-16:
Hello Philips,

I'm a gcerto colleague. I'm personally facing the problem. Thank you for your rapid answer.
I already tried the solution of clearing cache. I tried before the BluetoothGatt disconnect(), after the disconnect() and after the BluetoothGatt close() without any success. 

I have another information : our device is PAIRED (bonded) just before the disconnection : That means the methods does nothing if I don’t modify it or force the flag every time. Then I removed the test to be sure about refreshing every time :

`mBluetoothGatt.getDevice().getBondState()`

If I call the method just after the disconnect() the invoke method cannot be executed (invoke returns false): 

```
public void close() 
{
   
   Logger.getInstance().logWarning(TAG, "close()");
   
   synchronized(m_BluetoothGattMutex){
   // Clean the GATT object
     if (mBluetoothGatt == null)
        {
            return;
        }
        mBluetoothGatt.disconnect();

        refreshDeviceCache();
   }
}
```

If I call the method just after the close() in the BluetoothGattCallback, the invoke method cannot be executed (returns false). But if I call it in the callback before the close(), the invoke success but nothing changes in the dfu process after… :

```
private final BluetoothGattCallback m_gattCallback = new BluetoothGattCallback() {
    @Override
    public void onConnectionStateChange(BluetoothGatt gatt, int status, int newState) {

	(…)
if (newState == BluetoothProfile.STATE_DISCONNECTED) {
       
   Logger.getInstance().logWarning(TAG, "m_gattCallback STATE_DISCONNECTED");
   
   // End the service
   disconnect();
   refreshDeviceCache();  // HERE success to call the method but nothing happens
    
   synchronized(m_BluetoothGattMutex){
      if(mBluetoothGatt != null){
         Logger.getInstance().logWarning(TAG, "m_bluetoothGatt close()");
      mBluetoothGatt.close();

               //refreshDeviceCache();  // HERE failed to call the method

      mBluetoothGatt = null;
      Logger.getInstance().logWarning(TAG, "m_bluetoothGatt nulled");
      }
   }

(…)

}
```
For your first point, the only thing I do with your library is simply calling the previous written method startUpdate() after my device disconnection. You wrote “simple start DFU while being connected to it ». I must not disconnect?

Any help can be appreciated. Thank you by advance

Best regards

Yann

philips77, 2017-02-16:
Hello,

> I have another information : our device is PAIRED (bonded) just before the disconnection : That means the methods does nothing if I don’t modify it or force the flag every time. Then I removed the test to be sure about refreshing every time :

I don't understand. You use your device as not bonded and then bond just before disconnection? What methods? What flag? What test?

> refreshDeviceCache();  // HERE success to call the method but nothing happens

This is exactly how we do it.
call gatt.disconnect();
// wait for onConnectionStateChange(newState = Disconnected)
in this callback call refreshDeviceCache() and then close().
There is no callback for clearing the cache, but Android should perform service discovery next time you call discoverServices() as the cache will be empty.

Keep in mind, that if your device is bonded, and you are using Legacy DFU from SDK 8+, where the bootloader supports sending Service Changed indication, you don't have to clear the cache at all. The bond information will be copied to the bootloader and the after connecting to bootloader it will send the indication and Android will perform service discovery on its own.

> For your first point, the only thing I do with your library is simply calling the previous written method startUpdate() after my device disconnection. You wrote “simple start DFU while being connected to it ». I must not disconnect?

The DFU lib handles switching to bootloader mode on its own. The requirements for Legacy DFU are as follows:
- In your app mode you have to have DFU Service with 1, 2 or 3 characteristics (DFU Packet (optional, not present in mbed impl), DFU Control Point (mandatory), and for bootloaders from SDK 7+: DFU Version)
- After enabling DFU Control Point notifications and writing 01-04 the device should restart in bootloader mode.
- The service will take care of disconnection and will reconnect to the same device address. If necessary it will also clear cache (necessary = device was not bonded || device was bonded and DFU Version was not present or returned value lower than 0.8 (?) (0x08-00) ).

You may, bot don't have to disconnect from your device before calling initiator.start(). On Android each BluetoothGatt object sees the connection independently, so the service will connect to already connected device and will see it as if it has just been connected. It will not be aware that some other app or component in your app is already connected. Sending 01-04 will enforce disconnection from target side, so all connected components will get disconnect event.

YannSDW, 2017-02-16:
Yes that was it! 

In the same time you wrote your answer, I test without disconnection and it perfectly works like that...
I test on 2 mobiles for the moment. It seems to be OK. 

I now start testing on the full Android devices we have but I’m quite sure everything will be OK!
Thank you for your time. I think you can close the thread :-)

Best regards

philips77, 2017-02-16:
Glad to hear that! Closing issue then. Good luck!

closed 2017-02-16 by philips77