nordicsemi/Android-DFU-Library #46: GATT Invalid Handle / unable to read version number on first attempt when bonded

JJansenJJansen, 2017-02-28:
When we want to perform DFU while maintaining a bond, the first attempt always fails after the device has switched to bootloader (dual bank, s132) and we receive the error "GATT INVALID HANDLE" from the DFU library.
When we retry (while the device is still in bootloader) it succeeds. However the first attempt when starting from the application always fails after the bootloader has been entered (I have verified that the device is in bootloader mode at that time).

This issue does not occur when we don't use bonding and it also doesn't occur when we use nrfConnect to perform an update (either bonded or not bonded, both succeed in that case).

The Android log shows a connection state change error and an error on reading the version number:
D/ViewRootImpl: ViewPostImeInputStage processPointer 0
D/ViewRootImpl: ViewPostImeInputStage processPointer 1
D/BluetoothGatt: refresh() - device: FB:E5:EA:21:FC:CF
D/BluetoothGatt: onClientConnParamsChanged() - Device=FB:E5:EA:21:FC:CF interval=6 status=0
D/BluetoothGatt: onClientConnParamsChanged() - Device=FB:E5:EA:21:FC:CF interval=39 status=0
I/DfuBaseService: Connecting to the device...
D/BluetoothGatt: connect() - device: FB:E5:EA:21:FC:CF, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=306917d8-60b2-4b06-a91a-6c3971e9f9e1
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FB:E5:EA:21:FC:CF
I/DfuBaseService: Connected to GATT server
I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt: discoverServices() - device: FB:E5:EA:21:FC:CF
I/DfuBaseService: Attempting to start service discovery... succeed
D/BluetoothGatt: onSearchComplete() = Device=FB:E5:EA:21:FC:CF Status=0
I/DfuBaseService: Services discovered
I/DfuBaseService: Services discovered
I/DfuImpl: Reading DFU version number...
I/DfuImpl: Version number read: 0.1
W/DfuImpl: Application with buttonless update found
I/DfuImpl: Enabling notifications...
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=6 device=FB:E5:EA:21:FC:CF
D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=8 device=FB:E5:EA:21:FC:CF
**E/DfuBaseService: Connection state change error: 19 newState: 0**
I/DfuBaseService: Cleaning up...
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=6
I/DfuImpl: Starting service that will connect to the DFU bootloader
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I/DfuBaseService: Connecting to the device...
D/BluetoothGatt: connect() - device: FB:E5:EA:21:FC:CF, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=ef88d4c9-2a71-4c43-b51c-b750b54337bd
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FB:E5:EA:21:FC:CF
I/DfuBaseService: Connected to GATT server
I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt: discoverServices() - device: FB:E5:EA:21:FC:CF
I/DfuBaseService: Attempting to start service discovery... succeed
D/BluetoothGatt: onSearchComplete() = Device=FB:E5:EA:21:FC:CF Status=0
I/DfuBaseService: Services discovered
I/DfuBaseService: Services discovered
I/DfuImpl: Reading DFU version number...
**E/DfuImpl: Characteristic read error: 1
E/DfuBaseService: Unable to read version number (error 1)**
I/DfuBaseService: Disconnecting from the device...
D/BluetoothGatt: cancelOpen() - device: FB:E5:EA:21:FC:CF
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FB:E5:EA:21:FC:CF
I/DfuBaseService: Disconnected from GATT server
I/DfuBaseService: Cleaning up...
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=6
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED

I am suspecting a disconnect in the bootloader for some reason, however nrfConnect does not seem to have any issues with this (or is simply performing a retry in the background?) . As mentioned, the issue does not occur when we don't bond with the device. Any ideas what might be causing this?

philips77, 2017-02-28:
Hi, of course :)
As far I remember proper bonding support was added in SDK 8. There the bond information is passed to the bootloader and connection is secure also in bootloader mode. The bootloader after gets connected should send Service Changed indication (newer Android versions enable it automatically while for older the dfu lib will do before jumping to BL mode). From the log I see that it must have been enabled prior to starting DFU. After connecting to the bootloader the service waits for the indication: 

I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...

Android, when it's received, should refresh cache automatically and start service discovery. Calling gatt.discoverServices() will either return then from cache, or stop and restart discovery if it didn't finish. Now you should have correct services and read should work.

In your case it looks as if the indication wasn't sent out the phone ignored it. It was supported correctly on Android since the beginning, i think, but on some phones it may be the case (although we didn't have any customer issue regarding this and it used to work on all our phones).

Please check if you send this indication. 
The service doesn't force refresh service when the device is bond as it does otherwise (refresh method has a parameter `force`). That's why it works without bonding.

If you are using s132 i assume you are using sdk 8+?

JJansenJJansen, 2017-03-01:
Hi Aleksander, thank you very much for your very quick reply :)

We are indeed using SDK 11.0

Your suggestion makes sense, I am wondering whether the characteristisch changed indication is sent/received at all, even though we have enabled it in both the application and the bootloader. I will take a look at this.

What does surprise me is that the nrfConnect app does not seem to have an issue with it at all. Please take a look at the log from nrfConnect below. I do see a call to gatt.refresh just after the application disconnects and jumps to the bootloader. Is this how rediscovery is forced in the app?:

nRF Connect, 2017-03-01
MFR99999917 (FB:E5:EA:21:FC:CF)
V	12:44:39.649	Connecting to FB:E5:EA:21:FC:CF...
D	12:44:39.649	gatt = device.connectGatt(autoConnect = false)
D	12:44:40.034	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:44:40.034	Connected to FB:E5:EA:21:FC:CF
D	12:44:40.034	wait(600ms)
D	12:44:40.046	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	12:44:40.644	Discovering services...
D	12:44:40.644	gatt.discoverServices()
D	12:44:40.779	[Callback] Services discovered with status: 0
I	12:44:40.782	Services discovered
V	12:44:40.817	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Serial Number String [R] (0x2A25)
- Hardware Revision String [R] (0x2A27)
- Firmware Revision String [R] (0x2A26)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Unknown Service (517caf00-4511-80a0-e611-8201eca2a07d)
- Unknown Characteristic [W] (517cfabc-4511-80a0-e611-8201eca2a07d)
- Unknown Characteristic [N R] (517caf01-4511-80a0-e611-8201eca2a07d)
   Client Characteristic Configuration (0x2902)
V	12:44:45.941	Starting pairing...
D	12:44:45.941	device.createBond()
D	12:44:45.981	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D	12:44:46.740	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I	12:44:46.740	Device bonded
V	12:45:18.664	[DFU] DFU service started
V	12:45:18.664	[DFU] Opening file...
I	12:45:18.743	[DFU] Image file opened (78868 bytes in total)
D	12:45:18.743	[DFU] wait(1000)
D	12:45:19.760	[DFU] wait(1000)
V	12:45:20.732	[DFU] Connecting to DFU target...
D	12:45:20.732	[DFU] gatt = device.connectGatt(autoConnect = false)
I	12:45:20.798	[DFU] Connected to FB:E5:EA:21:FC:CF
D	12:45:20.798	[DFU] wait(1600)
V	12:45:22.389	[DFU] Discovering services...
D	12:45:22.390	[DFU] gatt.discoverServices()
I	12:45:22.437	[DFU] Services discovered
D	12:45:22.520	[DFU] wait(1000)
V	12:45:23.511	[DFU] Reading DFU version number...
D	12:45:23.511	[DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I	12:45:23.574	[DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 01-00
A	12:45:23.574	[DFU] Version number read: 0.1
W	12:45:23.574	[DFU] Application with buttonless update found
V	12:45:23.574	[DFU] Jumping to the DFU Bootloader...
V	12:45:23.574	[DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D	12:45:23.574	[DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
[... 91 lines cut ...]
D	12:46:33.489	gatt.close()
V	12:46:33.492	Connecting to FB:E5:EA:21:FC:CF...
D	12:46:33.493	gatt = device.connectGatt(autoConnect = false)
D	12:46:33.871	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:46:33.871	Connected to FB:E5:EA:21:FC:CF
D	12:46:33.871	wait(600ms)
D	12:46:33.882	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	12:46:34.476	Discovering services...
D	12:46:34.476	gatt.discoverServices()
D	12:46:34.485	[Callback] Services discovered with status: 0
I	12:46:34.486	Services discovered
V	12:46:34.511	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)

Thanks again for your assistance!

philips77, 2017-03-01:
Yes, the **refresh()** called by nRF Connect is separate from the DFU lib (no [DFU] in the log there).
Actually, by default it does it only until Android 5 or 6 and is disabled for more recent versions. On your phone you may also disable it in *Settings->Connectivity->Clear device cache->Never/Not bonded*. It used to be that connecting to a device with clear cache was much more stable, now it's stable even without it and as it's a "trick" (called using reflections) I try not to use it if not explicitly required (like in DFU on not-bonded devices).

JJansenJJansen, 2017-03-03:
I found the problem. It was indeed caused by the service changed indication not being sent in the bootloader, after fixing this the DFU continues flawlessly. Because we were testing with nrfConnect at first the problem was 'masked' by the forced refresh.
Thanks for the support Aleksander!

closed 2017-03-03 by JJansenJJansen

philips77, 2017-03-03:
Glad to hear that! Good luck! :)