nordicsemi/Android-DFU-Library #48: Custom characteristics

jonkun, 2017-03-14:
We use custom characteristics on bluetooth device to upload firmware (or bootloader)

Problem:
We can't use this library from maven center repository, we must checkout it from github and change characteristics manually. 

Could you implement in additional methods in DfuServiceInitiator to set custom characteristics.
Example.:

final DfuServiceInitiator starter = new DfuServiceInitiator(mSelectedDevice.getAddress())
&nbsp;&nbsp;&nbsp;&nbsp;.setDeviceName(mSelectedDevice.getName())
&nbsp;&nbsp;&nbsp;&nbsp;.setKeepBond(keepBond)
&nbsp;&nbsp;&nbsp;&nbsp;.setCustomService("MY CUSTOM SERVICE CHARACTERISTIC UUID")
&nbsp;&nbsp;&nbsp;&nbsp;.setCustomControlPoint("MY CUSTOM CONTROL POINT CHARACTERISTIC UUID")
&nbsp;&nbsp;&nbsp;&nbsp;.setCustomDataPoint("MY CUSTOM DATA POINT CHARACTERISTIC UUID")
&nbsp;&nbsp;&nbsp;&nbsp;...

Thanks.

philips77, 2017-03-14:
Hi, yes, i was actually thinking about it too. But we have to make it available for both Android and iOS so please give us some time. I'll ask sdk guys if they have anything against (for example the legacy dfu service uuid is blacklisted by Google as potentially dangerous). That's why secure dfu has a new one and whitelisted. I'll check how about with custom ones.

philips77, 2017-03-21:
Hi, could you check the https://github.com/NordicSemiconductor/Android-DFU-Library/commit/4642ad4ae3b0c746f1d257d9b3924706738b573a commit and see if you are satisfied? There may be some issues why one would start 2+ DFU services with different custom UUIDs but that's highly unlikely.

jonkun, 2017-03-22:
I tested it using method: `.setCustomUuidsForSecureDfu(SERVICE_UUID, CONTROL_POINT_UUID, PACKET_UUID)`
Firmware upload was successful (as expected).
Thanks.

closed 2017-03-22 by jonkun