nordicsemi/Android-DFU-Library #49: Bonding information removed on phone side after combined BL+SD+APP update

JJansenJJansen, 2017-03-15:
Hi Aleksander,

I just noticed that when we perform a full update (zip package with bl, sd and app) while having a bond between phone and device, the bond appears to be removed on the phone side after the update. 
I think it is removed after the download of the BL+SD and before reconnecting to the bootloader for uploading the application.
I've tested it also in nrfConnect and the behaviour is the same (even though we set 'Keep bonding information' to true in the settings tab).
My assumption is that the bond is erased by the DFU library? Is this a necessity and/or can we prevent it from happening?

Thanks in advance!

Jarno

philips77, 2017-03-15:
Hi, yes, it's true. It used to be that flashing a new SD was removing the old app as well (first, it requires the side for dual bank, second, the old app may not be compatible with the new SD) together with all its data, including bond info. So had I left it on the phone, the following connection wouldn't be possible. Has it changed?

JJansenJJansen, 2017-03-15:
Hi Aleksander,

Thank you for the quick reply!
It's true that the application code flash is overwritten during a SD+BL update. However I am not sure whether data flash is also erased. In my case the bonding information is actually retained on the device after a full update (including erase of application). So I assume that data flash is not erased or that it is actually maintained via the no-init RAM section which the bootloader uses for sharing bond information during normal updates. But perhaps you can verify this with your colleagues? We are using SDK 11.0 S132 v2.0.0 and a dual bank bootloader.

philips77, 2017-03-15:
I'll talk with my colleagues then and come back to you soon.

philips77, 2017-03-15:
Hi, I discussed your issue and here's what I learned:
- Preparing the update the DFU bootloader does not erase DFU_APP_DATA_RESERVED number of pages of the App Data space.
   - in Legacy DFU the default value of DFU_APP_DATA_RESERVED was 0 ([link](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v11.0.0/bledfu_memory_appdata.html?cp=4_0_3_4_3_1_3_3)) but a developer could modify it to N times PAGE SIZE (0x0200) as I mentioned [here](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/v1.2.0/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L132).
   - in Secure DFU, starting from SDK 12, this value is by default set to 3 pages, as they are using FDS-system (new file storage).
- The bond information is kept in those pages (at least 2 for Legacy DFU?)
- The app data will be overwritten even with setting this value to > 0 if more space is required for the dual bank, that is when new SD+BL requires more space ([graphic](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v9.0.0%2Fbledfu_memory_banks.html&cp=4_0_5_4_3_3_1_0&anchor=bledfu_dual_bank))
- And, most important, the App Data are preserved also if SD+BL update is done, that means that there in fact is a bug in the DFU Lib 1.2.0 which I will fix ASAP. There is a setting [DfuServiceInitiator#setKeepBond(boolean)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/v1.2.0/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L100) that I would like to keep used in such case, so please make sure you will call it with parameter *true*. That applies also for Secure DFU (to make default behavior similar to Legacy DFU) even though the default setting is set to 3 pages.

So, TL;DR: expect version 1.2.1 soon.

JJansenJJansen, 2017-03-15:
Hi Aleksander,

Thanks for the extensive information!
I have full confidence that this will be fixed in the next version of the library. For now we will continue using the DFU without bonding. Thanks once again for the effort!

philips77, 2017-03-15:
Hi, could you try with the newest version from GitHub? 
I haven't tested it at all, but looks good in my opinion.

JJansenJJansen, 2017-03-15:
Hi Aleksander,
We've decided to focus on not maintaining the bond during DFU, so unfortunately I won't be able to test it properly.

philips77, 2017-03-15:
Then you could at least check if it doesn't crash ;) That's cool. I'll do more testing and development tomorrow and hopefully release it then. Thanks again for pointing us this bug.

closed 2017-03-17 by philips77