nordicsemi/Android-DFU-Library #50: onClientConnectionState() - status=8

nmzkchina, 2017-03-21:
hi，Philips
I am facing a problem in ota, the error message is 

2017-03-21 02:12:55.271  03-21 02:12:55.257 11619 11632 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=8 device=D7:EB:D9:A5:2D:B8
2017-03-21 02:12:55.273  03-21 02:12:55.257 11619 11632 E DfuBaseService: Connection state change error: 8 newState: 0
2017-03-21 02:12:55.274  03-21 02:12:55.257 11619 12087 E DfuImpl : Disconnected while sending data
2017-03-21 02:12:55.278  03-21 02:12:55.257 11619 12087 E DfuBaseService: Uploading Firmware Image failed: device disconnected
2017-03-21 02:12:55.280  03-21 02:12:55.257 11619 12087 D BluetoothGatt: close()
2017-03-21 02:12:55.282  03-21 02:12:55.257 11619 12087 D BluetoothGatt: unregisterApp() - mClientIf=8

our  dfu sdk in version 1.1.1, in s7 edge Android 6.0.1, the error occurs every time in dfu percent 40%, and  another phone MI NOTE Pro Android 7.0 occurs every time in  37%. I found error 8 in GATTError is timeout, how i fix this problem。

philips77, 2017-03-21:
You may try forcing Packet Receipt Notifications (PRNs) by starting the DFU Service with an initiator on which you called [this](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L105) method with parameter *true*. As you see, bu default for Marshmallow and Nougat is disabled (they should handle the buffer overflow problem correctly, but apparently not all phones do...). Then you may set the PRN value using a method below that one. Default value is 12, but something close to 200 should also work and it may be faster. But start with lower values.

nmzkchina, 2017-03-22:
I set PRN enable and set value 12 、50 and 150, but the error still exists in s7 edge

philips77, 2017-03-22:
We have S7 (not edge) at house and will try to reproduce it. First i need some more info:
How large your firmware that you send is?
What sdk is your project and dfu bootloader forked from?
Are you using legacy or secure dfu?
Did you try sending your update using nrf connect app? Could you post the logs here?

nmzkchina, 2017-03-22:
1、our firmware is about 90kb
2、our project dfu sdk version is 1.1.1 and bootloader is based in SDK 11, our firmware runs on nrf52832
3、we used legacy maybe,
4、we don't have this phone , our beta user send this log to us;
android code：
final DfuServiceInitiator starter = new DfuServiceInitiator(address)
                .setDeviceName(name)
                .setKeepBond(false)
                .setDisableNotification(true)
                .setPacketsReceiptNotificationsEnabled(true).setPacketsReceiptNotificationsValue(12)
                ;
        starter.setZip(null, zipPath);
        if (!isDfuServiceRunning()) {
            starter.start(context, DfuService.class);
        }

our app log:

2017-03-22 14:48:09.801  03-22 14:48:09.790 16848 16911 D BluetoothGatt: onClientConnectionState() - status=19 clientIf=8 device=D7:EB:D9:A5:2D:B8
2017-03-22 14:48:09.812  03-22 14:48:09.790 16848 16911 E DfuBaseService: Connection state change error: 19 newState: 0
2017-03-22 14:48:09.817  03-22 14:48:09.800 16848 17272 D BluetoothGatt: refresh() - device: D7:EB:D9:A5:2D:B8
2017-03-22 14:48:09.818  03-22 14:48:09.800 16848 17272 D BluetoothGatt: close()
2017-03-22 14:48:09.819  03-22 14:48:09.800 16848 17272 D BluetoothGatt: unregisterApp() - mClientIf=8
2017-03-22 14:48:10.852  03-22 14:48:10.840 16848 17272 D BluetoothGatt: connect() - device: D7:EB:D9:A5:2D:B8, auto: false
2017-03-22 14:48:10.855  03-22 14:48:10.840 16848 17272 D BluetoothGatt: registerApp()
2017-03-22 14:48:10.857  03-22 14:48:10.840 16848 17272 D BluetoothGatt: registerApp() - UUID=b4a895db-ca4e-4961-b74c-9e796e7393c0
2017-03-22 14:48:10.903  03-22 14:48:10.890 16848 17153 D BluetoothGatt: onClientRegistered() - status=0 clientIf=8
2017-03-22 14:48:11.241  03-22 14:48:11.230 16848 16911 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=D7:EB:D9:A5:2D:B8
2017-03-22 14:48:11.247  03-22 14:48:11.240 16848 16911 D BluetoothGatt: discoverServices() - device: D7:EB:D9:A5:2D:B8
2017-03-22 14:48:11.736  03-22 14:48:11.720 16848 16860 D BluetoothGatt: onClientConnParamsChanged() - Device=D7:EB:D9:A5:2D:B8 interval=24 status=0
2017-03-22 14:48:11.956  03-22 14:48:11.940 16848 16859 D BluetoothGatt: onClientConnParamsChanged() - Device=D7:EB:D9:A5:2D:B8 interval=6 status=0
2017-03-22 14:48:12.097  03-22 14:48:12.090 16848 16911 D BluetoothGatt: onSearchComplete() = Device=D7:EB:D9:A5:2D:B8 Status=0
2017-03-22 14:48:12.099  03-22 14:48:12.090 16848 17272 I DfuBaseService: Services discovered
2017-03-22 14:48:12.103  03-22 14:48:12.090 16848 16848 E DfuProgressListener: --------onDeviceConnected--------D7:EB:D9:A5:2D:B8
2017-03-22 14:48:12.157  03-22 14:48:12.150 16848 16859 D BluetoothGatt: onClientConnParamsChanged() - Device=D7:EB:D9:A5:2D:B8 interval=24 status=0
2017-03-22 14:48:12.218  03-22 14:48:12.200 16848 17271 D GetBoxSoftHardwareVersionCommand: execute: 当前线程：Timer-22-1238
2017-03-22 14:48:12.219  03-22 14:48:12.200 16848 17271 D GetBoxSoftHardwareVersionCommand: execute: 开始超时倒计时
2017-03-22 14:48:12.669  03-22 14:48:12.660 16848 16911 D BluetoothGatt: onClientConnParamsChanged() - Device=D7:EB:D9:A5:2D:B8 interval=24 status=0
2017-03-22 14:48:13.149  03-22 14:48:13.140 16848 17272 D BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
2017-03-22 14:48:15.339  03-22 14:48:15.330 16848 16848 E DfuProgressListener: --------onDfuProcessStarted--------D7:EB:D9:A5:2D:B8
2017-03-22 14:48:15.340  03-22 14:48:15.330 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :0, currentPart :1, partsTotal :1
2017-03-22 14:48:15.520  03-22 14:48:15.510 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :1, currentPart :1, partsTotal :1
2017-03-22 14:48:15.727  03-22 14:48:15.720 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :2, currentPart :1, partsTotal :1
2017-03-22 14:48:15.935  03-22 14:48:15.920 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :3, currentPart :1, partsTotal :1
2017-03-22 14:48:16.468  03-22 14:48:16.460 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :4, currentPart :1, partsTotal :1
2017-03-22 14:48:16.646  03-22 14:48:16.640 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :5, currentPart :1, partsTotal :1
2017-03-22 14:48:16.846  03-22 14:48:16.830 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :6, currentPart :1, partsTotal :1
2017-03-22 14:48:17.369  03-22 14:48:17.360 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :7, currentPart :1, partsTotal :1
2017-03-22 14:48:17.579  03-22 14:48:17.570 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :8, currentPart :1, partsTotal :1
2017-03-22 14:48:17.761  03-22 14:48:17.750 16848 16848 E DfuProgressListener: --------onProgressChanged----deviceAddress----D7:EB:D9:A5:2D:B8, percent :9, currentPart :1, partsTotal :1
[... cut ...]
2017-03-22 14:48:30.376  03-22 14:48:30.350 16848 17272 E DfuBaseService: Uploading Firmware Image failed: device disconnected
2017-03-22 14:48:30.377  03-22 14:48:30.350 16848 17272 D BluetoothGatt: close()
2017-03-22 14:48:30.379  03-22 14:48:30.350 16848 17272 D BluetoothGatt: unregisterApp() - mClientIf=8
2017-03-22 14:48:30.381  03-22 14:48:30.360 16848 16848 E DfuProgressListener: --------onDeviceDisconnected--------D7:EB:D9:A5:2D:B8
2017-03-22 14:48:30.383  03-22 14:48:30.360 16848 16848 E DfuProgressListener: --------onError--------D7:EB:D9:A5:2D:B8

nmzkchina, 2017-03-31:
hi，Philips，we test the dfu in our s7 edge, but everything is ok, maybe the hardware is something wrong, because another hardware also update error in this s7 edge but update success in other phone, so i close this question， thank you.

closed 2017-03-31 by nmzkchina