nordicsemi/Android-DFU-Library #53: build tools version

vliedel, 2017-04-20:
Hi, I'm using this wonderful library, but recently ran into some troubles due to other libraries working with build tools version 23 and crashes with version 25.
So I now cloned the dfu library, changed the build tools to version 23, and it works again.

My question: is there any reason you upgraded to build tools version 25? Or is it just to stay up to date? Any project including this library via maven will be forced to use build tools 25, with which many projects seem to have issues.

philips77, 2017-04-20:
Hmm... Just to stay up to date. I'll think about it.

philips77, 2017-07-28:
Hi, so I'm about to release an update and wondering what build tools versions should I use. I'm now working on 25.0.3 but could lower the version, I guess. How about this: https://stackoverflow.com/questions/24494077/gradle-force-build-tools-version-on-third-party-libraries

vliedel, 2017-07-28:
Cool, i'll give that a try, will let you know.

vliedel, 2017-07-28:
It builds, but I'm getting runtime errors from elsewhere now. I'll have a look if i can force the build version for just the dfu lib.

philips77, 2017-07-28:
I just released version 1.3.1 (https://github.com/NordicSemiconductor/Android-DFU-Library/releases/tag/v1.3.1) with build tools 25.0.3 hoping this above link fixes your issue.

vliedel, 2017-07-28:
It's ok, I'll just pull and change the build.gradle again. Shame there seems to be no good solution for such issues.

philips77, 2017-07-28:
Check out this: https://developer.android.com/studio/build/index.html -> Configure project-wide properties

vliedel, 2017-07-31:
Added some prints to see if the buildtool and sdk version actually changed, they did.
Also tried to only change the version of the dfu lib, but that didn't work either. Weirdly enough, it leads to getting some C++ error from react-native, so I don't want to bother you with that :P 
Thanks for the help though!

closed 2017-08-31 by philips77