nordicsemi/Android-DFU-Library #54: Error 4097 dfu file not found

AnD010, 2017-04-22:
Hi, 
I know that the message is clear...but I try to get .zip file through 2 ways
```
  Uri mFileStreamUri;
 mFileStreamUri=Uri.fromFile(new File("///android_asset/vit2_10_20170207.zip"));
 mFileStreamUri=Uri.parse("android.resource://"+mContext.getPackageName()+"/raw/vit2_10_20170207.zip");
```
and then... I do
      final DfuServiceInitiator starter = new DfuServiceInitiator(dongle.getMac())
                    .setDeviceName(dongle.getName())
                    .setKeepBond(false)
                    .setForceDfu(false)
                    .setPacketsReceiptNotificationsEnabled(false)
                    .setPacketsReceiptNotificationsValue(12)
                    .setDisableNotification(true)
                    .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
            starter.setZip(mFileStreamUri,null);
            starter.start(mContext, DfuService.class);
Always get the same error 

> DFU FILE NOT FOUND
> 4097.

What wrong with me code???

Thanks
Regards

philips77, 2017-04-22:
If your zip file is included in your app, place it in res/raw and then use R.raw.your-file, instead of uri. There is a method for this in the initializer.

AnD010, 2017-04-22:
Sorry.. my fault... I didn´t see the method for  raw/res file...

thaks

closed 2017-04-22 by AnD010