nordicsemi/Android-DFU-Library #55: DFU fails due to timing

softcharles, 2017-05-10:
Sometimes DFU fails directly due to timing. The DFUBaseService registers itself as BluetoothGattCallback for all BluetootGatt objects and does not keep track from who the onConnectionStateChange is called. 

Check for instance the log that is attached. That log shows that DFU is started and while LegacyDfuImpl is waiting for the first Connection to disconnect (waitUntilDisconnect) ACL_DISCONNECT is received which results in mConnectionState = STATE_DISCONNECTED, i.e. the thread waiting in waitUntilDisconnected is notified and running again. Then the new connection is tried and connecting and suddenly the first onConnectionStateChanged (disconnected) is received. The DfuBaseService Thinks that one belongs to the new gatt object and fails the DFU...

This one in the log 

05-10 10:03:42.362 11457-11471/com.kronaby.watch.debug I/DfuBaseService: onConnectionStateChange  newstate 2 

belongs to the new Connection attempt (second GattObject), hence the DFU should have continued...

This issue happens once in a while, but it is very easy to reproduce, just dubugging and adding some breakpoints in onConnectionStateChanged, which Changes the timing a little bit. The DFU library is way too timing dependent.

05-10 10:03:11.566 11457-13299/com.kronaby.watch.debug I/DfuImpl: Version number read: 0.1
05-10 10:03:11.567 11457-13299/com.kronaby.watch.debug W/DfuImpl: Application with buttonless update found
05-10 10:03:11.568 11457-13299/com.kronaby.watch.debug I/DfuImpl: Enabling notifications...
05-10 10:03:13.384 11457-13299/com.kronaby.watch.debug I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
05-10 10:03:33.060 11457-11457/com.kronaby.watch.debug I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED 
05-10 10:03:33.068 11457-13299/com.kronaby.watch.debug I/DfuBaseService: Refreshing result: true 
05-10 10:03:33.069 11457-13299/com.kronaby.watch.debug I/DfuBaseService: Cleaning up... 
05-10 10:03:33.070 11457-13299/com.kronaby.watch.debug I/DfuImpl: Starting service that will connect to the DFU bootloader
05-10 10:03:35.143 11457-13299/com.kronaby.watch.debug I/DfuBaseService: Connecting to the device... 
05-10 10:03:38.174 11457-11498/com.kronaby.watch.debug I/DfuBaseService: onConnectionStateChange newstate 0 
05-10 10:03:38.174 11457-11498/com.kronaby.watch.debug E/DfuBaseService: Connection state change error: 8 newState: 0
05-10 10:03:38.174 11457-13299/com.kronaby.watch.debug E/DfuBaseService: Device got disconnected before service discovery finished
05-10 10:03:42.362 11457-11471/com.kronaby.watch.debug I/DfuBaseService: onConnectionStateChange  newstate 2 
05-10 10:03:42.363 11457-11471/com.kronaby.watch.debug I/DfuBaseService: Connected to GATT server 
05-10 10:03:42.374 11457-11471/com.kronaby.watch.debug I/DfuBaseService: Attempting to start service discovery... succeed 
05-10 10:03:42.386 11457-11472/com.kronaby.watch.debug I/DfuBaseService: Services discovered 
05-10 10:03:44.430 11457-13299/com.kronaby.watch.debug I/DfuBaseService: Disconnecting from the device... 
05-10 10:03:44.447 11457-13299/com.kronaby.watch.debug I/DfuBaseService: Refreshing result: true 
05-10 10:03:44.447 11457-13299/com.kronaby.watch.debug I/DfuBaseService: Cleaning up... 
05-10 10:03:46.054 11457-11457/com.kronaby.watch.debug D/WatchDfuProvider: state: 4, message: DFU error: DFU DEVICE DISCONNECTED
05-10 10:03:46.055 11457-11457/com.kronaby.watch.debug D/DfuStatusController: updating: DFU failed

philips77, 2017-08-31:
Thanks for the bug report! I'm really sorry for the enormous delay, but this issue got lost somewhere.
I finally understand now where the "Device got disconnected before service discovery finished" #35 may come from. Those ACTION_ACL_ events are very unreliable. They are called in random order, sometimes before gatt callback, sometimes much after. You are correct that receiving this event in wrong time may do much more bad than good. It was added there long time ago as some devices failed to deliver onConnectionStateChanged with state 0 (disconnected).

Ill look into this deeply now.

philips77, 2017-08-31:
OK, few observations:

1. With handling this ACL_DISCONNECTED event I was calling `gatt.close()` on an object that didn't yet get `onConnectionStateChanged(DISCONNECTED)`. That might have caused some troubles. After cleaning up the old `gatt` object and trying to connect again to the device we get those weird *status 8, newState 0* followed by *status 0, newState 2*. A closed gatt should not get any callbacks, but it appears as there are some race conditions there, or something not correctly handled.

I'll check with removing setting state to disconnected in ACL receiver for the next release. The app should base on gatt callbacks after all. ACL receiver was a hack.

philips77, 2017-08-31:
I'll leave it open so that you could verify if the next release has fixed/improved stability.

philips77, 2017-08-31:
Please check version 1.4.0.

nikithalakshmanan, 2017-11-11:
I'm still seeing this issue with v2.5.1

philips77, 2017-11-11:
Could you paste logcat here?

philips77, 2017-11-11:
I've fixed the reason why it was failing because of the race conditions, but perhaps there is something else causing it.

nikithalakshmanan, 2017-11-13:
Hi, 

This is the logcat:

11-13 09:31:42.136 3001-3001/com.example.nikitha.run_2 D/ContentValues: onStop
11-13 09:31:42.137 3001-3001/com.example.nikitha.run_2 D/BluetoothAdapter: stopLeScan()
11-13 09:31:42.138 3001-3001/com.example.nikitha.run_2 D/BluetoothAdapter: scan not started yet
11-13 09:31:42.139 3001-3001/com.example.nikitha.run_2 D/BluetoothAdapter: stopLeScan()
11-13 09:31:42.140 3001-3001/com.example.nikitha.run_2 D/BluetoothAdapter: scan not started yet
11-13 09:31:44.191 3001-3612/com.example.nikitha.run_2 I/DfuBaseService: Connecting to the device...
11-13 09:31:44.192 3001-3612/com.example.nikitha.run_2 D/BluetoothGatt: connect() - device: EA:D0:81:80:D9:E0, auto: false
11-13 09:31:44.192 3001-3612/com.example.nikitha.run_2 D/BluetoothGatt: registerApp()
11-13 09:31:44.192 3001-3612/com.example.nikitha.run_2 D/BluetoothGatt: registerApp() - UUID=e66a3079-6b62-4657-ad09-87cf3c81b786
11-13 09:31:44.194 3001-3014/com.example.nikitha.run_2 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
11-13 09:31:46.955 3001-3001/com.example.nikitha.run_2 D/BluetoothAdapter: stopLeScan()
11-13 09:31:46.956 3001-3001/com.example.nikitha.run_2 D/BluetoothAdapter: scan not started yet
11-13 09:32:14.197 3001-3014/com.example.nikitha.run_2 D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=EA:D0:81:80:D9:E0
11-13 09:32:14.197 3001-3014/com.example.nikitha.run_2 E/DfuBaseService: Connection state change error: 133 newState: 0
11-13 09:32:14.197 3001-3612/com.example.nikitha.run_2 E/DfuBaseService: Device got disconnected before service discovery finished
11-13 09:32:14.199 3001-3612/com.example.nikitha.run_2 D/BluetoothGatt: refresh() - device: EA:D0:81:80:D9:E0
11-13 09:32:14.200 3001-3612/com.example.nikitha.run_2 I/DfuBaseService: Refreshing result: true
11-13 09:32:14.200 3001-3612/com.example.nikitha.run_2 I/DfuBaseService: Cleaning up...
11-13 09:32:14.201 3001-3612/com.example.nikitha.run_2 D/BluetoothGatt: close()
11-13 09:32:14.201 3001-3612/com.example.nikitha.run_2 D/BluetoothGatt: unregisterApp() - mClientIf=5

philips77, 2017-11-13:
This is a different issue. You get 133 after a timeout, so your device with this address is not reachable:
- can be out of range,
- can be non-connectable
- can be bonded to another phone, or this phone may think it's bonded to it (i.e. you had a previous app on this DK, bonded it to your phone and then flagged the dfu, boy phone still tries security)
- is advertising but you didn't scan for it since Bluetooth was enabled on the phone, so the connect request is sent to the same address, but with type 'public'
- something else.

nikithalakshmanan, 2017-11-13:
This was fixed! Thank you so much!

commit b7d123b4 (2017-08-31): Bug hopefully fixed: #55
closed 2017-11-14 by philips77