nordicsemi/Android-DFU-Library #57: Initial DfuTarg Connection Issue

troyjperales, 2017-05-17:
I've been able to limit the number of 133 errors for the most part except for one specific case.

When updating a "virgin" device (one that just came out of programming from production), our app using the DFU library is not able to connect to the DfuTarg. If that device has previously been connected to the smartphone to perform a DFU update, then it will perform fine.  If the device is "new" to the phone (e.g. the phone does not yet have bonding information for the DfuTarg), then it will fail.

The only way to catch the issue is to test with a device that has never been bonded to the phone before using NRF connect.  Even if the device is re-programmed, the phone will hold onto bonding information that masks the issue if the device has been bonded in the past. It likely has to do with the DfuTarg bonding.

For a virgin device - If you put the device in DFU mode and then connect to the DfuTarg using NRF connect, then the app can complete the actual update process. Our app is able to successfully connect to the DFU target to complete future updates. I believe this is because connection information allowing the phone to connect to the DfuTarg is stored on the phone and then can be reused by the app.

Has anyone else experienced this initial DfuTarg connection issue for devices that have just been programmed? Any suggestions?

softcharles, 2017-05-18:
The dfu library uses BluetoothDevice.connectGatt to connect to the device. I Think that will only work if the device has a public Bluetooth address or the device has been scanned at least once before connection attempt since bluetooth adapter have been started.

Read the chapter 3 in the BLE_on_Android_v1.0.1.pdf on this page https://devzone.nordicsemi.com/blogs/1046/what-to-keep-in-mind-when-developing-your-ble-andr/

troyjperales, 2017-05-21:
@softcharles Thanks for sharing the document. It helped tremendously!

I was able to connect to the device after scanning with a filter that included it's address. I believe a lot of 133 errors could be prevented if the scanning for the given device address was included in the library prior to connecting.

philips77, 2017-07-28:
Hi, you can start the DFU while being connected in your app, so initial scanning in the app is not possible as it may not advertise at the moment. I think DFUing a device that has never been even scanned is not a common thing... Glad that the document has helped.

closed 2017-07-28 by philips77