nordicsemi/Android-DFU-Library #58: (BL/SD)+App update fails on App update process with 133

troyjperales, 2017-05-25:
I've prevented 133 errors from occurring by scanning for the DFU target address, getting a scan result, and _then_ starting the DFU service. However, I experienced a 133 error in the middle of a (BL/SD)+App update. The library breaks up the update into two parts, first the BL/SD followed by the App. After BL/SD part finishes updating, a 133 error occurs when trying to reconnect for the App update portion. We end up having is a partial update, where BL/SD is new and App old.

A mismatched BL/SD and App could become problematic.

Garfielt, 2017-07-10:
E/DfuBaseService: Connection state change error: 133 newState: 0
E/DfuBaseService: Device got disconnected before service discovery finishe!

I got this error too with an android 7.1.1.

philips77, 2017-07-28:
How often does it happen? Before connecting to the new BL the library performs a scan. This may fail if the app didn't have Location permission, or the Location service was disabled. I don't know if that's the case (looks like it isn't). 
Unfortunately, I don't know why this 133 happens then. The only solution we have right now is to repeat the DFU (this time the target device has new bootloader and it might have remove the old app, so perhaps selecting new target may be required.

closed 2017-08-31 by philips77