nordicsemi/Android-DFU-Library #59: Keep Existing Connection?

kenwdelong, 2017-06-16:
We are experiencing a lot of DFU failures.  I'm wondering if there's any best practice advice on 
- if there is a connection already open, should we close it before we call `DfuServiceInitiator.start()`?
- is is a good or bad idea to clear the GATT cache (using the undocumented clear method http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android) before calling `DfuServiceInitiator.start()`?

Thanks for any help!!

KentaHarada, 2017-06-22:
@kenwdelong 
> if there is a connection already open, should we close it before we call DfuServiceInitiator.start()?
Yes. You should close it before call DFU Library.
Because GATT connection is processed in DFU Library. 

> is is a good or bad idea to clear the GATT cach
I think that it not a good  idea. Because refresh() is hide method. I tried it on some Android devices but
Failed to call it(Some android device don't have refresh() method.) 

@philips77 
Is my understanding correct ?

philips77, 2017-06-22:
Hi, so in theory you may have as many BluetoothGatt objects as you want (to some extent) connected to the same device. Also other apps can do as well. The system keeps the physical connection and each gatt obj is just an interface of it. The physical connection is closed when the last gatt is disconnected. Keep in mind, that during the dfu you'll get a disconnect event, so there is not much sense in keeping one not closed, unless you want to listen to its notifications... for some reason.

Refresh() method used to help connecting to a device on older platforms. But since Android ~6 it's no longer needed and connections, at least on higher quality phone, are stable. I'd say: try without, try with, check if it's worth. Also it's not easy to use it, as you call it in BluetoothGatt object, which you get on connection. There is no callback for refresh. It tigers internal service discovery after which you will not be notified when it's done, and calling discoverServices() too early will restart it. You may call it after disconnecting, before close(), but then the first connection may fail... And so on :)

kenwdelong, 2017-06-22:
Excellent, thank you everyone!

closed 2017-06-22 by kenwdelong