nordicsemi/Android-DFU-Library #60: How I add Dfu library into my project?

gn00618777, 2017-06-20:
I follow below steps:
Clone the project into DFULibrary folder (by default it will be cloned into Android-DFU-Library folder) to your projects root, for example to AndroidstudioProjects.
Add the dfu module to your project:
Add '..:DFULibrary:dfu' to the settings.gradle file:
include ':dfu'
project(':dfu').projectDir = file('../DFULibrary/dfu')
Open Project Structure -> Modules -> app -> Dependencies tab and add dfu module dependency. You may also edit the build.gradle file in your app module manually by adding the following dependency: compile project(':..:DFULibrary:dfu')

I rename  Android-DFU-Library to DFULibrary and my 
settings.gradle:

include ':app'
include ':dfu'
project(':dfu').projectDir = file('../DFULibrary/dfu')

build.gradle(Module:app):

dependencies {
    compile fileTree(include: ['*.jar'], dir: 'libs')
    androidTestCompile('com.android.support.test.espresso:espresso-core:2.2.2', {
        exclude group: 'com.android.support', module: 'support-annotations'
    })
    compile 'com.android.support:appcompat-v7:25.2.0'
    compile 'com.android.support:design:25.2.0'
    compile 'com.android.support:cardview-v7:25.2.0'
    compile 'com.android.support:recyclerview-v7:25.2.0'
    compile 'com.android.support.constraint:constraint-layout:1.0.2'
    compile 'no.nordicsemi.android.support.v18:scanner:1.0.0'
    compile 'com.github.machinarius:preferencefragment:0.1.1'
    compile 'com.github.PhilJay:MPAndroidChart:v3.0.2'
    compile project(':..:DFULibrary:dfu')
    testCompile 'junit:junit:4.12'
}
and the error is shown in build.gradle(Module:app)
Gradle sync failed: Project with path ':..:DFULibrary:dfu' could not be found in project ':app'.
					Consult IDE log for more details (Help | Show Log)

and the error is shown in settings.gradle(Project Settings)
Gradle sync failed: Project with path ':..:DFULibrary:dfu' could not be found in project ':app'.
					Consult IDE log for more details (Help | Show Log)

I need modify code with DFU library. Thank you.

philips77, 2017-06-28:
Hi, sorry for the late reply. Did you manage to fix your issue?
I think you should include:

```
compile project(':dfu')
```

instead of 
```
compile project(':..:DFULibrary:dfu')
```
in your app/build.gradle.

closed 2017-07-28 by philips77

PiyushKumar186, 2018-05-17:
Hi, I am getting this error everytime "Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve project :dfu." Though I have include the dfu library as per the instruction.

philips77, 2018-05-17:
Try opening the dfu project in a new instance of Android Studio, this will create the iml file in dfu folder.

mshandan, 2019-04-19:
When I uploaded my app to github, I found that the dfu package was not uploaded together. How can I solve this problem?

philips77, 2019-04-23:
What do you mean? The DFU lib is just a jcenter dependency. Gradle should sync it based on the `build.gradle` file.