nordicsemi/Android-DFU-Library #61: Error code 3 or 10

YoonYoungDoo, 2017-06-28:
I used buttonless dfu. And my App's service use that.
Sometime, error code 3 or 10 appear in my app. Retry again and again, dfu is success.
What is error_code 3, 10? I known INVALID_PARAM and OPERATION_FAILED. what is mean? 
please, help. thanks.

philips77, 2017-06-28:
Hi,
You'd have to provide more information. What DFU are you using, legacy or secure one? From which SDK?
When do the errors pop up?

YoonYoungDoo, 2017-06-29:
`06-28 18:19:43.005 16975-23567/? W/NamedFutures: Timeout future task has been cancelled: 10000 milliseconds
06-28 18:19:52.790 32670-1500/com.skt.sh I/DfuBaseService: Connecting to the device...
06-28 18:19:52.952 32670-32682/com.skt.sh I/DfuBaseService: Connected to GATT server
06-28 18:19:52.963 32670-32682/com.skt.sh I/DfuBaseService: Attempting to start service discovery... succeed
06-28 18:19:52.972 32670-494/com.skt.sh I/DfuBaseService: Services discovered
06-28 18:19:52.977 32670-1500/com.skt.sh W/DfuImpl: Secure DFU bootloader found
06-28 18:19:53.991 32670-1500/com.skt.sh I/DfuImpl: Enabling notifications...
06-28 18:19:54.762 16975-23567/? W/NamedFutures: Timeout future task has been cancelled: 10000 milliseconds
06-28 18:19:55.050 32670-1500/com.skt.sh I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
06-28 18:19:55.087 32670-1500/com.skt.sh I/DfuImpl: Init packet CRC is the same
06-28 18:19:55.087 32670-1500/com.skt.sh I/DfuImpl: -> Whole Init packet was sent before
06-28 18:19:55.088 32670-1500/com.skt.sh I/DfuImpl: Executing init packet (Op Code = 4)
06-28 18:19:55.119 32670-1500/com.skt.sh I/DfuImpl: Setting object to Data (Op Code = 6, Type = 1)
06-28 18:19:55.161 32670-1500/com.skt.sh I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/5)
06-28 18:19:55.200 32670-1500/com.skt.sh I/DfuImpl: Uploading firmware...
06-28 18:19:55.907 32670-1500/com.skt.sh I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
06-28 18:19:56.019 32670-1500/com.skt.sh I/DfuImpl: Checksum received (Offset = 256, CRC = D5D51CE4)
06-28 18:19:56.021 32670-1500/com.skt.sh I/DfuImpl: CRC does not match! Expected 1AB8AFD3 but found D5D51CE4. Retrying...(2/3)
06-28 18:19:56.046 32670-1500/com.skt.sh I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 7936) (1/5)
06-28 18:19:56.110 32670-1500/com.skt.sh E/DfuImpl: Creating Command object failed (error 3)
06-28 18:19:56.111 32670-1500/com.skt.sh I/DfuBaseService: Disconnecting from the device...
06-28 18:19:56.125 32670-494/com.skt.sh I/DfuBaseService: Disconnected from GATT server
06-28 18:19:56.130 32670-1500/com.skt.sh I/DfuBaseService: Refreshing result: true
06-28 18:19:56.130 32670-1500/com.skt.sh I/DfuBaseService: Cleaning up...`
`06-28 18:28:12.697 10648-12009/com.skt.sh I/DfuBaseService: Connecting to the device...
06-28 18:28:12.826 10648-10856/com.skt.sh I/DfuBaseService: Connected to GATT server
06-28 18:28:12.833 10648-10856/com.skt.sh I/DfuBaseService: Attempting to start service discovery... succeed
06-28 18:28:12.847 10648-10661/com.skt.sh I/DfuBaseService: Services discovered
06-28 18:28:12.850 10648-12009/com.skt.sh W/DfuImpl: Secure DFU bootloader found
06-28 18:28:13.859 10648-12009/com.skt.sh I/DfuImpl: Enabling notifications...
06-28 18:28:14.641 16975-7056/? W/NamedFutures: Timeout future task has been cancelled: 10000 milliseconds
06-28 18:28:14.894 10648-12009/com.skt.sh I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
06-28 18:28:14.935 10648-12009/com.skt.sh I/DfuImpl: Disabling Packet Receipt Notifications (Op Code = 2, Value = 0)
06-28 18:28:14.935 10648-12009/com.skt.sh I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
06-28 18:28:14.966 10648-12009/com.skt.sh I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 135)
06-28 18:28:14.999 10648-12009/com.skt.sh I/DfuImpl: Sending 0 bytes of init packet...
06-28 18:28:14.999 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = 12-84-01-0A-3E-08-01-12-3A-08-02-10-34-1A-02-91-01-20-00-28)
06-28 18:28:15.003 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = 00-30-00-38-EC-91-01-42-24-08-03-12-20-E0-90-B5-7D-C5-C5-79)
06-28 18:28:15.007 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = 44-E4-01-8B-E4-2F-58-D7-B6-CE-30-54-16-D5-89-CC-92-F2-85-FE)
06-28 18:28:15.011 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = 0F-6B-23-AC-6C-48-00-10-00-1A-40-D5-54-A4-E9-85-F0-6D-CF-43)
06-28 18:28:15.014 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = 9B-27-A5-27-71-07-B5-1F-62-45-C2-5C-02-44-41-36-44-D2-76-DE)
06-28 18:28:15.018 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = B0-58-94-BC-79-B3-FA-CD-0E-9A-C1-9D-D4-24-6B-CE-11-BF-B7-E3)
06-28 18:28:15.022 10648-12009/com.skt.sh I/DfuImpl: Sending init packet (Value = 54-EE-C1-AE-4D-84-7D-44-09-8E-57-EC-63-68-BF)
06-28 18:28:15.026 10648-12009/com.skt.sh I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
06-28 18:28:15.063 10648-12009/com.skt.sh I/DfuImpl: Checksum received (Offset = 135, CRC = EF375F53)
06-28 18:28:15.064 10648-12009/com.skt.sh I/DfuImpl: Executing init packet (Op Code = 4)
06-28 18:28:15.245 10648-12009/com.skt.sh I/DfuImpl: Setting object to Data (Op Code = 6, Type = 1)
06-28 18:28:15.418 10648-12009/com.skt.sh I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/5)
06-28 18:28:15.454 10648-12009/com.skt.sh I/DfuImpl: Uploading firmware...
06-28 18:28:16.086 10648-12009/com.skt.sh I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
06-28 18:28:16.254 10648-12009/com.skt.sh I/DfuImpl: Checksum received (Offset = 256, CRC = F1A79479)
06-28 18:28:16.256 10648-12009/com.skt.sh I/DfuImpl: CRC does not match! Expected 4CAC9341 but found F1A79479. Retrying...(2/3)
06-28 18:28:16.280 10648-12009/com.skt.sh I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 7936) (1/5)
06-28 18:28:16.315 10648-12009/com.skt.sh E/DfuImpl: Creating Command object failed (error 3)
06-28 18:28:16.317 10648-12009/com.skt.sh I/DfuBaseService: Disconnecting from the device...
06-28 18:28:16.330 10648-10661/com.skt.sh I/DfuBaseService: Disconnected from GATT server
06-28 18:28:16.336 10648-12009/com.skt.sh I/DfuBaseService: Refreshing result: true
06-28 18:28:16.336 10648-12009/com.skt.sh I/DfuBaseService: Cleaning up...`

This is Log.
I use, secure dfu, sdk 12.3.0 and nrf52_3.1.0. My board use nrf52832. thanks.

philips77, 2017-06-30:
Hi,

Let me copy paste your log in more readable format.

DfuBaseService: Connecting to the device...
DfuBaseService: Connected to GATT server
DfuBaseService: Attempting to start service discovery... succeed
DfuBaseService: Services discovered
DfuImpl: Secure DFU bootloader found
DfuImpl: Enabling notifications...
DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
DfuImpl: Init packet CRC is the same
DfuImpl: -> Whole Init packet was sent before
DfuImpl: Executing init packet (Op Code = 4)
DfuImpl: Setting object to Data (Op Code = 6, Type = 1)
DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/5)
DfuImpl: Uploading firmware...
DfuImpl: Sending Calculate Checksum command (Op Code = 3)
DfuImpl: Checksum received (Offset = 256, CRC = D5D51CE4)
DfuImpl: CRC does not match! Expected 1AB8AFD3 but found D5D51CE4. Retrying...(2/3)
DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 7936) (1/5)
DfuImpl: Creating Command object failed (error 3)
DfuBaseService: Disconnecting from the device...
DfuBaseService: Disconnected from GATT server
DfuBaseService: Refreshing result: true
DfuBaseService: Cleaning up...
/DfuBaseService: Connecting to the device...
/DfuBaseService: Connected to GATT server
/DfuBaseService: Attempting to start service discovery... succeed
/DfuBaseService: Services discovered
/DfuImpl: Secure DFU bootloader found
/DfuImpl: Enabling notifications...
/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
/DfuImpl: Disabling Packet Receipt Notifications (Op Code = 2, Value = 0)
/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 135)
/DfuImpl: Sending 0 bytes of init packet...
/DfuImpl: Sending init packet (Value = 12-84-01-0A-3E-08-01-12-3A-08-02-10-34-1A-02-91-01-20-00-28)
/DfuImpl: Sending init packet (Value = 00-30-00-38-EC-91-01-42-24-08-03-12-20-E0-90-B5-7D-C5-C5-79)
/DfuImpl: Sending init packet (Value = 44-E4-01-8B-E4-2F-58-D7-B6-CE-30-54-16-D5-89-CC-92-F2-85-FE)
/DfuImpl: Sending init packet (Value = 0F-6B-23-AC-6C-48-00-10-00-1A-40-D5-54-A4-E9-85-F0-6D-CF-43)
/DfuImpl: Sending init packet (Value = 9B-27-A5-27-71-07-B5-1F-62-45-C2-5C-02-44-41-36-44-D2-76-DE)
/DfuImpl: Sending init packet (Value = B0-58-94-BC-79-B3-FA-CD-0E-9A-C1-9D-D4-24-6B-CE-11-BF-B7-E3)
/DfuImpl: Sending init packet (Value = 54-EE-C1-AE-4D-84-7D-44-09-8E-57-EC-63-68-BF)
/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
/DfuImpl: Checksum received (Offset = 135, CRC = EF375F53)
/DfuImpl: Executing init packet (Op Code = 4)
/DfuImpl: Setting object to Data (Op Code = 6, Type = 1)
/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/5)
/DfuImpl: Uploading firmware...
/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
/DfuImpl: Checksum received (Offset = 256, CRC = F1A79479)
/DfuImpl: CRC does not match! Expected 4CAC9341 but found F1A79479. Retrying...(2/3)
/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 7936) (1/5)
/DfuImpl: Creating Command object failed (error 3)
/DfuBaseService: Disconnecting from the device...
/DfuBaseService: Disconnected from GATT server
/DfuBaseService: Refreshing result: true
/DfuBaseService: Cleaning up...

philips77, 2017-06-30:
So, I see one issue here, but completely not related to your issue:

Sending 0 bytes of init packet... <- here should write number of bytes, not 0

philips77, 2017-06-30:
But coming back to your issue. As you see, there is a problem with CRC calculation.
```
CRC does not match! Expected 1AB8AFD3 but found D5D51CE4...
...
CRC does not match! Expected 4CAC9341 but found F1A79479...
```
The Init packet CRC validation in the first case is OK (CRC is matching), but fails when you send the firmware. I guess it's caused by this:
```
Creating Data object (Op Code = 1, Type = 2, Size = **4096**) (1/5)
Uploading firmware...
Sending Calculate Checksum command (Op Code = 3)
Checksum received (Offset = **256**, CRC = D5D51CE4)
```
looks like 4096 bytes were to be sent, but the bootloader received only 256. Is 256 maximum object size in your implementation? Looks like I don't print this info in the log, and I should be. I'll fix it.

Is there an option for you to send me the ZIP you are using and your bootlaoder+softdevice, unless they are default? I'll have a look at the code trying to debug it without it, but would be cool to test it before releasing :)
Thanks for the bug report!

philips77, 2017-06-30:
Hi again,

Could you set the PRN to something ~12?
[setPacketsReceiptNotificationsEnabled(true)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L133)
[setPacketsReceiptNotificationsValue(DfuServiceInitiator.DEFAULT_PRN_VALUE)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L146)
and try again? 

My suspicion is that you  phone may send data too fast, or the device may have problems with saving them, and only firs 256 bytes of data object were saved and then reported. Setting PRN value will slow down it a bit, but ensure that device has time to store data.

YoonYoungDoo, 2017-07-03:
Thank you!! You saved me. I will try that. And I will write result.

I have another question. 
If i use library code in commercial apps, copyright is problem?
I contact major company and we will sell device.

philips77, 2017-07-03:
Check licences. Our DFU libraries are BSD-3 clause, so you may basically do commercial as well, edit them, etc. Similar with the bootloader and our SDK, which has a Nordic's licence that is based on BSD-3, just added that it must be used with our chips, which is obvious.

philips77, 2017-07-03:
Btw, the bug you had, despite the reason, should be handled correctly by the lib. Right now I'm assuming all 4096 bytes were sent , despite that in the Calculate Checksum request I get a response from the the device, that only 256 were received, which I ignore :) So as you see there is room for improvement which I'm tying to do now. I don't know why only 256, but it shouldn't abort it CRC for those 256 was correct.

philips77, 2017-07-03:
Hi, could you clone the newest version of this library from here and try again without using PRNs (like you had before)? Not, after receiving offset = 256 it should continue sending data to the same object, hopefully the rest, or at least next 256. Could you also post the logs from LogCat here afterward?

philips77, 2017-07-03:
Also, did you try updating your device on iPhone?

YoonYoungDoo, 2017-07-05:
I try using android dfu library in Galaxy. I added the code above. That error rarely happens.
Unfortunately, there is no log.
The schedule has begun commercialization and we plan to use the updated library for the next update.
Thanks to that, I turned over a big problem.
I will pass log, if i have a chance.

philips77, 2017-07-05:
Looking forward to see logs. I hope the dfu will succeed now even if this error occurs, just look for 'bytes were lost' in logs.

philips77, 2017-07-07:
Any success?

YoonYoungDoo, 2017-07-21:
I'm so late. sorry.
We start selling in Korea. Currently, 50,000 units have been sold. So I'm very busy..  :(

This is Log when dfu success.

`07-21 14:37:41.761 1463-1476/com.skt.sh D/Tracker_BleModule: New GATT service discovered.
07-21 14:37:41.762 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Service: 00001800-0000-1000-8000-00805f9b34fb i = 0
07-21 14:37:41.762 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Char: android.bluetooth.BluetoothGattCharacteristic@632a14b [i][j] = 0,0
07-21 14:37:41.762 1463-1476/com.skt.sh D/Tracker_BleModule: GATT uuid: 00002a00-0000-1000-8000-00805f9b34fb
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Char: android.bluetooth.BluetoothGattCharacteristic@bb5b728 [i][j] = 0,1
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT uuid: 00002a01-0000-1000-8000-00805f9b34fb
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Char: android.bluetooth.BluetoothGattCharacteristic@b1d5c41 [i][j] = 0,2
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT uuid: 00002a04-0000-1000-8000-00805f9b34fb
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Char: android.bluetooth.BluetoothGattCharacteristic@8e245e6 [i][j] = 0,3
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT uuid: 00002aa6-0000-1000-8000-00805f9b34fb
07-21 14:37:41.763 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Service: 00001801-0000-1000-8000-00805f9b34fb i = 1
07-21 14:37:41.764 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Service: 0000fe59-0000-1000-8000-00805f9b34fb i = 2
07-21 14:37:41.764 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Char: android.bluetooth.BluetoothGattCharacteristic@f880727 [i][j] = 2,0
07-21 14:37:41.764 1463-1476/com.skt.sh D/Tracker_BleModule: GATT uuid: 8ec90002-f315-4f60-9fb8-838830daea50
07-21 14:37:41.764 1463-1476/com.skt.sh D/Tracker_BleModule: GATT Char: android.bluetooth.BluetoothGattCharacteristic@e431dd4 [i][j] = 2,1
07-21 14:37:41.764 1463-1476/com.skt.sh D/Tracker_BleModule: GATT uuid: 8ec90001-f315-4f60-9fb8-838830daea50
07-21 14:37:41.812 1463-1475/com.skt.sh D/BluetoothGatt: onClientConnParamsChanged() - Device=D4:E3:CB:6D:4D:C4 interval=24 status=0
07-21 14:37:42.384 1463-1476/com.skt.sh D/BluetoothGatt: onClientConnParamsChanged() - Device=D4:E3:CB:6D:4D:C4 interval=24 status=0
07-21 14:37:42.785 1463-1463/com.skt.sh D/Tracker_TrackerService: otaDeviceAddr  - D4:E3:CB:6D:4D:C4
07-21 14:37:42.785 1463-1463/com.skt.sh D/Tracker_TrackerService: Service - MESSAGE_STATE_CHANGE: 17
07-21 14:37:42.785 1463-1463/com.skt.sh D/Tracker_BleModule: writeCharacteristic false [value] : 01
07-21 14:37:43.052 1463-1463/com.skt.sh D/Tracker_TrackerManager: tracker OTA Update start
07-21 14:37:45.098 1463-25106/com.skt.sh I/DfuBaseService: Connecting to the device...
07-21 14:37:45.109 1463-25106/com.skt.sh D/BluetoothGatt: connect() - device: D4:E3:CB:6D:4D:C4, auto: false
07-21 14:37:45.110 1463-25106/com.skt.sh D/BluetoothAdapter: isSecureModeEnabled
07-21 14:37:45.111 1463-25106/com.skt.sh D/BluetoothGatt: registerApp()
07-21 14:37:45.111 1463-25106/com.skt.sh D/BluetoothGatt: registerApp() - UUID=d2e24bfe-3d15-4893-997d-edfc4b0bc2a8
07-21 14:37:45.219 1463-1879/com.skt.sh D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
07-21 14:37:45.238 1463-1476/com.skt.sh D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=D4:E3:CB:6D:4D:C4
07-21 14:37:45.242 1463-1476/com.skt.sh I/DfuBaseService: Connected to GATT server
07-21 14:37:45.245 1463-1476/com.skt.sh D/BluetoothGatt: discoverServices() - device: D4:E3:CB:6D:4D:C4
07-21 14:37:45.247 1463-1476/com.skt.sh I/DfuBaseService: Attempting to start service discovery... succeed
07-21 14:37:45.250 1463-1475/com.skt.sh D/BluetoothGatt: onSearchComplete() = Device=D4:E3:CB:6D:4D:C4 Status=0
07-21 14:37:45.250 1463-1475/com.skt.sh I/DfuBaseService: Services discovered
07-21 14:37:45.252 1463-25106/com.skt.sh W/DfuImpl: Secure DFU bootloader found
07-21 14:37:45.276 1463-1463/com.skt.sh D/Tracker_TrackerOtaReceiver: Ota state Receive : PROGRESS_STARTING
07-21 14:37:45.277 1463-1463/com.skt.sh D/Tracker_TrackerOtaReceiver: Ota state Receive : PROGRESS_STARTING
07-21 14:37:46.259 1463-25106/com.skt.sh I/DfuImpl: Enabling notifications...
07-21 14:37:46.260 1463-25106/com.skt.sh D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
07-21 14:37:47.312 1463-25106/com.skt.sh I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
07-21 14:37:47.364 1463-25106/com.skt.sh I/DfuImpl: Disabling Packet Receipt Notifications (Op Code = 2, Value = 0)
07-21 14:37:47.365 1463-25106/com.skt.sh I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
07-21 14:37:47.423 1463-25106/com.skt.sh I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 135)
07-21 14:37:47.482 1463-25106/com.skt.sh I/DfuImpl: Sending 135 bytes of init packet...
07-21 14:37:47.483 1463-25106/com.skt.sh I/DfuImpl: Sending init packet (Value = 12-84-01-0A-3E-08-01-12-3A-08-0A-10-34-1A-02-91-01-20-00-28)
07-21 14:37:47.487 1463-25106/com.skt.sh I/DfuImpl: Sending init packet (Value = 00-30-00-38-80-92-01-42-24-08-03-12-20-8A-F9-87-43-04-01-60)
07-21 14:37:47.491 1463-25106/com.skt.sh I/DfuImpl: Sending init packet (Value = C6-2D-F5-7B-5B-55-56-DD-A7-BC-82-B0-2F-28-F9-40-ED-9B-E9-7C)
[... cut ...]
07-21 14:38:06.152 1463-25106/com.skt.sh D/BluetoothGatt: refresh() - device: D4:E3:CB:6D:4D:C4
07-21 14:38:06.157 1463-25106/com.skt.sh I/DfuBaseService: Refreshing result: true
07-21 14:38:06.157 1463-25106/com.skt.sh I/DfuBaseService: Cleaning up...
07-21 14:38:06.158 1463-25106/com.skt.sh D/BluetoothGatt: close()
07-21 14:38:06.159 1463-25106/com.skt.sh D/BluetoothGatt: unregisterApp() - mClientIf=10
07-21 14:38:06.187 1463-1463/com.skt.sh I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
`

YoonYoungDoo, 2017-07-21:
It is expected to sell 100,000 units a month. So dfu was very important.
Thank you very much.

philips77, 2017-07-21:
Looks like it's working fine with PRN = 12, yes?

YoonYoungDoo, 2017-07-21:
Yes. I have this code.

`final DfuServiceInitiator starter = new DfuServiceInitiator(bluetoothGatt.get(0).getDevice().getAddress())
                        .setDeviceName(bluetoothGatt.get(0).getDevice().getName())
                        .setKeepBond(false)
                        .setPacketsReceiptNotificationsEnabled(true)
                        .setPacketsReceiptNotificationsValue(DfuServiceInitiator.DEFAULT_PRN_VALUE)
                        .setZip(null,a);
                starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);`

and DfuServiceInitiator.DEFAULT_PRN_VALUE = 12
this is default value in library.

philips77, 2017-07-21:
Did you try without PRN but with my latest changes? (Library loader from source code, not from jcenter) Just to check if my changes helped.

YoonYoungDoo, 2017-07-24:
Sorry. I don't use newest version. I did not have enough time. So I add this code only.
`setPacketsReceiptNotificationsEnabled (true) 
setPacketsReceiptNotificationsValue (DfuServiceInitiator.DEFAULT_PRN_VALUE)`
And This is good. I'm sorry I can not help you.

philips77, 2017-07-28:
Nevertheless I'll release the update with latest fixes soon.

closed 2017-07-28 by philips77