nordicsemi/Android-DFU-Library #66: 8197: Remote CRC error on Android 7, but not on 6...

tompi, 2017-08-17:
Hi, thank you for the good work on this library!

I'm having a problem on Android 7, with the same code that works fine consistently on Android 6.
Are there any known issues with unsecure dfu upload on Android 7?

Could very well be I'm doing something wrong which triggers an error just on Android 7, but I tried to follow the method of the nrf connector app.

Here is the log with the process that ends with error 8197, hope you can spot my mistake!:

[android7-error-log.txt](https://github.com/NordicSemiconductor/Android-DFU-Library/files/1230608/android7-error-log.txt)

philips77, 2017-08-18:
Hi,
The error 5 is Invalid Object. I have no idea why you get CRC error at the very end, but let's ignore it for now.
In line 482 of your log you get the status 5 in response to Create Data Object no 4.
This error can be sent either because the timing was not good (SD doing a whole lot of other things) or more likely there was something erroneous about the address. When Create Object cmd is received the bootloader erases the next page/s. Here's the code from SDK 14 (you are using 12.x, but should be similar):

```C
if (nrf_dfu_flash_erase((m_firmware_start_addr + s_dfu_settings.progress.firmware_image_offset),
                                    CEIL_DIV(p_req->object_size, CODE_PAGE_SIZE), NULL) != NRF_SUCCESS)
{
   NRF_LOG_ERROR("Erase operation failed");
   return NRF_DFU_RES_CODE_INVALID_OBJECT;
} 
```

I would extend the NRF_LOG_ERROR to report the address (but then again. If logging is on, the issue may not be present). That's line 906 or something in the 14.0 release.
[‎18.‎08.‎2017 14:02] Kvamtrø, Frank Audun: 
I would extend the NRF_LOG_ERROR to report the address (but then again. If logging is on, the issue may not be present
line 906 or something in the 14.0 release
It is also possible to change the number of retries. Right now NRF_FSTORAGE_MAX_RETRIES is set to 8. This is the reason i think it is about the address rather than the SoftDevice call. Unaligned address because of some corruption? The address pointing to something that cannot be erased (because of block protection)? However, as it works with Android 6, it is possible that 7 does something quicker and some timing issue arises. Does it happen often? Like 100% of times when testing with Android 7? Did you try different phone with 7?

philips77, 2017-08-31:
Any update on this?

tompi, 2017-08-31:
Hi, thank you for replying, I don't really have access to the device-side of things, so I have to involve our embedded guy. He's trying to upgrade to SDK 14 first, and we are crossing our fingers that problem will disappear then. Will try your suggestion if the upgrade fails to resolve the issue.

You can close this issue if you like, and I can create a new one if sdk 14 does not help?

philips77, 2017-09-04:
Then closing.

closed 2017-09-04 by philips77