nordicsemi/Android-DFU-Library #67: Disconnecting notification dosen't remove on some devices

kinber95, 2017-08-22:
Hey ,

the background notification state "DISCONNECTING" from Android-DFU, dosen't disappear although the device disconnected and the process finish correctly. 

I get this bug on my Moto G5 and BQ X5. On my Nexus 5X its working without any problems.

Any solution for this problem ? 

Cheers

philips77, 2017-08-31:
Hi, sorry for late reply.
The notification is removed (or changed to "DFU completed") when the service receives ```onConnectionStateChanged``` event with state *DISCONNECTED*. It's more an Android issue as this event should come at the end. I'm not sure if I can let's say clear notification few seconds after calling ```gatt.disconnect()``` in a timeout as from the lib perspective the connection is still open.
Could you post here the log (part when the dfu finises) from nRF Connect? Let's confirm that it's because of lack of the event, not something else.

colinmadere, 2017-09-27:
Could this be the culprit if the next notification comes too quickly?  It looks like it skips the notification update if the previous one was less than 250ms before, even if it's a final ones that auto-cancel.

Seems you want to add a check and not skip if its a final-state notification that auto-cancel at the end.  Do you accept outside PRs?
 
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/v1.4.2/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1398

philips77, 2017-09-27:
Yes, we accept PRs. You may be right here. Final one should always be shown.

closed 2018-10-24 by philips77