nordicsemi/Android-DFU-Library #71: DFU update failed on Samsung S8 (android 7.0.0) with Suunto Movesense

skrabacz-michal, 2017-09-18:
On a bunch of different devices, everything works normal and fine (like Samsung S7Edge, A5, S6, Nexus5, MotoXPlay). I run tests several times with same result.

```
31m  FirmwareUpdateService[0m [30;104m D [0m (FirmwareUpdateService.java:17) in onCreate => Start firmware update service [user=pudzian@dka.io]
[34mpdateFirmwareSubscriber[0m [30;104m D [0m (FirmwareUpdatePresenter$UpdateFirmwareSubscriber.java:197) in onNext => Update firmware done Firmware(installed=false) (is view bind=true) [user=pudzian@dka.io]
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] DFU service started [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Opening file... [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Image file opened (275492 bytes in total) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] wait(1000) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] wait(1000) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Connecting to DFU target... [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] gatt = device.connectGatt(autoConnect = false) [user=pudzian@dka.io]
[36m          BluetoothGatt[0m [30;104m D [0m connect() - device: D7:74:2A:0C:51:E0, auto: false
[33m       BluetoothAdapter[0m [30;104m D [0m isSecureModeEnabled
[36m          BluetoothGatt[0m [30;104m D [0m registerApp()
                        [30;104m D [0m registerApp() - UUID=d8da9eae-07d0-432c-a2b2-e3fb28cc9b7e
                        [30;104m D [0m onClientRegistered() - status=0 clientIf=7
                        [30;104m D [0m onClientConnectionState() - status=0 clientIf=7 device=D7:74:2A:0C:51:E0
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Connected to D7:74:2A:0C:51:E0 [user=pudzian@dka.io]
[36m          BluetoothGatt[0m [30;104m D [0m discoverServices() - device: D7:74:2A:0C:51:E0
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Discovering services... [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] gatt.discoverServices() [user=pudzian@dka.io]
[36m          BluetoothGatt[0m [30;104m D [0m onClientConnParamsChanged() - Device=D7:74:2A:0C:51:E0 interval=13 status=0
                        [30;104m D [0m onClientConnParamsChanged() - Device=D7:74:2A:0C:51:E0 interval=6 status=0
                        [30;104m D [0m onSearchComplete() = Device=D7:74:2A:0C:51:E0 Status=0
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Services discovered [user=pudzian@dka.io]
[32mogressListenerAdapter$1[0m [30;104m D [0m (FirmwareUpdatePresenter$dfuProgressListenerAdapter$1.java:51) in onDeviceConnected => Dfu progress ... (is view bind=true) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdatePresenter$dfuProgressListenerAdapter$1.java:58) in onDfuProcessStarting => Dfu progress ... (is view bind=true) [user=pudzian@dka.io]
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] wait(1000) [user=pudzian@dka.io]
[36m          BluetoothGatt[0m [30;104m D [0m onClientConnParamsChanged() - Device=D7:74:2A:0C:51:E0 interval=13 status=0
                        [30;104m D [0m onClientConnParamsChanged() - Device=D7:74:2A:0C:51:E0 interval=13 status=0
                        [30;104m D [0m configureMTU() - device: D7:74:2A:0C:51:E0 mtu: 517
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Requesting new MTU... [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] gatt.requestMtu(517) [user=pudzian@dka.io]
[36m          BluetoothGatt[0m [30;104m D [0m onConfigureMTU() - Device=D7:74:2A:0C:51:E0 mtu=23 status=0
                        [30;104m D [0m setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
[35mdateActivity$onCreate$3[0m [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] MTU changed to: 23 [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Enabling notifications for 8ec90001-f315-4f60-9fb8-838830daea50 [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] gatt.setCharacteristicNotification(8ec90001-f315-4f60-9fb8-838830daea50, true) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Data written to descr.8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-00 [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Notifications enabled for 8ec90001-f315-4f60-9fb8-838830daea50 [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Notifications enabled [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] wait(1000) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50 [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50) [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 06-01 [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-06-01-00-01-00-00-87-00-00-00-B3-ED-83-09 [user=pudzian@dka.io]
[... cut ...]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Remote DFU error: REMOTE DFU OPERATION NOT PERMITTED [user=pudzian@dka.io]
                        [30;104m D [0m (FirmwareUpdateActivity$onCreate$3.java:39) in onLogEvent => Update firmware on device D7:74:2A:0C:51:E0:[DFU] Disconnecting... [user=pudzian@dka.io]
[32mogressListenerAdapter$1[0m [30;104m D [0m (FirmwareUpdatePresenter$dfuProgressListenerAdapter$1.java:76) in onDeviceDisconnecting => Dfu progress ... (is view bind=true) [user=pudzian@dka.io]
```

philips77, 2017-09-18:
Hi, do you get the same result when you try to update the device starting from beginning, not resuming the upload? 
On the log you sent I see that you try to resume upload, the init packet is the same, the first page of data is the same and the library tries to execute it. For unknown reason it fails. But what if you erase the first page and start again? As the library does not allow to do this yet, you may replace the app.bin file in your zip (or just change first byte in it) and try again. This time the CRC will be different so the lib, instead of sending 0x04 (execute), will send 06-02 (create data object) which will erase the old one. You may then abort the upload (it would fail eventually as the signature check wouldn't pass), revert the first byte and try again.

skrabacz-michal, 2017-09-26:
Hi,
ok - i'll try it and back with the result. 
How can I check/verify if my package is correct? Right now I getting error ``` Update firmware on device D7:74:2A:0C:51:E0:[DFU] Remote DFU error: REMOTE DFU INVALID OBJECT ``` ?

philips77, 2017-09-26:
As you get this error it means it's not correct :)
Do you get it just after sending init packet or whole fw? In the first case make sure it's signed with correct key, the same used in the bootloader. In the second case - you must have alerted the bin file.

ojousima, 2017-10-27:
I've had reports of DFU failing on Samsung S8 and OnePlus 5 (Android 7 on both). The reports come from end users so I do not have detailed logs. The users are able to run DFU to same nRF52 device with same DFU package on Samsung S5 and ZTE Blade Android 7 on both.

Additionally S8 user reports that DFU service failures started after recent Android update. I can contact the users for more detailed reports if necessary, please give exact instructions on how to report if more details are needed.

yjwfn, 2018-03-23:
I facing this problem .  I fork this project, and then disable resume feature by modify source code. It is working fine.

Beremor, 2018-06-07:
@yjwfn Could you please share your solution ?

yjwfn, 2018-06-08:
@Beremor I found `performDfu` method in `SecureDfuImpl` first, and then you can see two methods 
 `sendInitPacket(gatt)` and `sendFirmware(gatt)` within it.  The methods will correct CRC before send  package, you can make it always not equals to CRC of you package. This is not very nice solution,  hope DFULibrary can add some options to disable it.

philips77, 2018-10-24:
I'll add an option to do fresh DFU in 1.8.0.

closed 2018-10-24 by philips77

philips77, 2018-10-29:
Released in 1.8.0

alixwar, 2018-10-31:
I have seen this in 1.8.0:

```
2018-10-31 09:51:19.1 D/FirmwareUpgradeProgressListener: [I=4, T=M] onProgressChanged(deviceAddress=[C2:30:BA:E4:4F:90], percent=[57], speed=[1.7865353], avgSpeed=[2.262649], currentPart=[1], partsTotal=[1])
2018-10-31 09:51:19.7 D/FirmwareUpgradeProgressListener: [I=4, T=M] onProgressChanged(deviceAddress=[C2:30:BA:E4:4F:90], percent=[58], speed=[3.6065574], avgSpeed=[2.2773843], currentPart=[1], partsTotal=[1])
2018-10-31 09:51:19.8 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:20.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-F0-01-00-97-F1-4D-E0
2018-10-31 09:51:20.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-F0-01-00-97-F1-4D-E0
2018-10-31 09:51:20.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Checksum received (Offset = 126976, CRC = E04DF197)
2018-10-31 09:51:20.4 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:20.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
2018-10-31 09:51:20.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
2018-10-31 09:51:20.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data object executed
2018-10-31 09:51:20.4 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:20.6 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
2018-10-31 09:51:20.6 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
2018-10-31 09:51:20.6 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data object (32/54) created
2018-10-31 09:51:20.6 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Uploading firmware...
2018-10-31 09:51:20.6 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Sending firmware to characteristic 8ec90002-f315-4f60-9fb8-838830daea50...
2018-10-31 09:51:21.0 D/FirmwareUpgradeProgressListener: [I=4, T=M] onProgressChanged(deviceAddress=[C2:30:BA:E4:4F:90], percent=[59], speed=[1.6973479], avgSpeed=[2.2643194], currentPart=[1], partsTotal=[1])
2018-10-31 09:51:21.3 D/FirmwareUpgradeProgressListener: [I=4, T=M] onProgressChanged(deviceAddress=[C2:30:BA:E4:4F:90], percent=[60], speed=[6.728395], avgSpeed=[2.2895877], currentPart=[1], partsTotal=[1])
2018-10-31 09:51:21.4 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:22.3 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 03
2018-10-31 09:51:22.3 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-00-02-00-71-A8-0B-27
2018-10-31 09:51:22.3 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Checksum received (Offset = 131072, CRC = 270BA871)
2018-10-31 09:51:22.3 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 04
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data object executed
2018-10-31 09:51:22.4 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data object (33/54) created
2018-10-31 09:51:22.4 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Uploading firmware...
2018-10-31 09:51:22.4 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Sending firmware to characteristic 8ec90002-f315-4f60-9fb8-838830daea50...
2018-10-31 09:51:23.1 D/FirmwareUpgradeProgressListener: [I=4, T=M] onProgressChanged(deviceAddress=[C2:30:BA:E4:4F:90], percent=[61], speed=[1.2166204], avgSpeed=[2.2567873], currentPart=[1], partsTotal=[1])
2018-10-31 09:51:23.7 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
2018-10-31 09:51:26.4 E/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Error (0x85): GATT ERROR
2018-10-31 09:51:27.1 E/FirmwareUpgradeProgressListener: [I=4, T=M] onError(deviceAddress=[C2:30:BA:E4:4F:90], error=[133], errorType=[2], message=[GATT ERROR])
2018-10-31 09:51:27.1 W/FirmwareUpgradeDialogFragment: [I=4, T=M] Retrying... (try #2)
2018-10-31 09:51:27.1 W/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] [Broadcast] Abort action received
2018-10-31 09:51:28.1 D/FirmwareUpgradeDialogFragment: [I=4, T=M] initFirmwareUpgrade()
[... cut ...]
2018-10-31 09:51:31.6 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-05
2018-10-31 09:51:31.6 E/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Remote DFU error: INVALID OBJECT
2018-10-31 09:51:31.6 D/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Disconnecting...
2018-10-31 09:51:31.7 I/DfuService: [T=M] [C2:30:BA:E4:4F:90] [DFU] Disconnected
2018-10-31 09:51:32.3 E/FirmwareUpgradeProgressListener: [I=4, T=M] onError(deviceAddress=[C2:30:BA:E4:4F:90], error=[517], errorType=[3], message=[INVALID OBJECT])
```

philips77, 2018-10-31:
Could you move it to a new issue?

alixwar, 2018-10-31:
Sure

PR #141 (2018-10-24): Option to disable resume in Secure DFU
This also fixes #71