nordicsemi/Android-DFU-Library #74: GATT ERROR

Baniares, 2017-10-12:
I'm facing this problem in an LG phone and I'm also facing problems with some samsung galaxy s8 any ideas on what is going on?

(I don't have access to a samsung galaxy s8 at the moment but soon I will and I'll be posting stack trace for those too)

```
I: Connecting to the device...
D: connect() - device: CC:67:14:BB:7A:C1, auto: false
D: registerApp()
D: registerApp() - UUID=13c0e951-8504-4841-bd05-8f779c3cafca
D: onClientRegistered() - status=0 clientIf=6
W: update focus... 
D: onClientConnectionState() - status=0 clientIf=6 device=CC:67:14:BB:7A:C1
I: Connected to GATT server
D: discoverServices() - device: CC:67:14:BB:7A:C1
I: Attempting to start service discovery... succeed
I: Action received: android.bluetooth.device.action.ACL_CONNECTED
D: onSearchComplete() = Device=CC:67:14:BB:7A:C1 Status=0
I: Services discovered
D: Connected to monitor
D: Starting DFU process
I: Reading DFU version number...
I: Version number read: 0.6 -> Bootloader from SDK 8.0 or newer. Bond sharing supported
W: Legacy DFU bootloader found
I: Enabling notifications...
D: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
E: Descriptor write error: 133
E: Unable to set notifications state (error 133)
I: Disconnecting from the device...
D: cancelOpen() - device: CC:67:14:BB:7A:C1
D: onClientConnectionState() - status=8 clientIf=6 device=CC:67:14:BB:7A:C1
E: Connection state change error: 8 newState: 0
D: refresh() - device: CC:67:14:BB:7A:C1
D: Disconnecting from monitor
I: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I: Refreshing result: true
I: Cleaning up...
D: close()
D: unregisterApp() - mClientIf=6
D: Disconnected from monitor
D: GATT ERROR
```

philips77, 2017-10-12:
Try restarting the Bluetooth adapter on your phone, or the whole phone. It may be some non-released resources in the Bluetooth stack. But I would say it is something wrong with your peripheral. Looks like it doesn't handle the descriptor write request properly. Error 133 is thrown when the phone didn't get an expected reply or got something unexpected (a device behaved incorrectly).
It's not an issue in this DFU library.

veonua, 2017-12-03:
have the same error due to service list cache, we need to force refresh service cache (it's hidden API)

philips77, 2018-10-24:
Let me close the issue, as it's a year old now. If you still are facing the problem, feel free to create a new one.

closed 2018-10-24 by philips77

veonua, 2018-10-24:
> Let me close the issue, as it's a year old now.
Have you done something? 

Android 7+ caches BT service list, but Nordic changes service set when DFU starts. 
so when application tries to connect to DFU service it fails.

philips77, 2018-10-24:
That's why we use address +1 in bootloader mode (for bonded devices, when you use Buttonless service with bonds, it relies on Service Changed indication instead), to avoid cache problems.

philips77, 2018-10-24:
Sorry, I wasn't clear enough. There was no any change in the DFU library. It is the bootloader that is using incremented address. It has been like that since the beginning, or at least some very old SDK. Bootloader uses address +1 to avoid caching issues (although it is not required in my opinion, as there is this hidden *refresh()* method, which could be used if we know that services are to be changed). In Secure DFU with bonds, and Legacy DFU when bonding is used, starting from SDK 8 (afair) the same address is used and cache is refreshed based on Service Changed indication.

But as you are facing this problem (and refreshing the cache does help), that means that you either are using the same address in app and bootloader mode, or are connecting to the secondary address with some other services. But, to be honest, the 133 on enabling notifications doesn't look like caching issue for me. It would rather send INVALID HANDLE error instead, I think.

veonua, 2018-10-24:
to fix this issue I had to add 

refreshDeviceCache() to

https://github.com/NordicSemiconductor/Android-DFU-Library/blob/4840d895338bd7634698eb1147cbe82e7b8a328f/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L780