nordicsemi/Android-DFU-Library #76: ERROR: DFU DEVICE DISCONNECTED

Esperanz0, 2017-10-17:
Hey I have problem with DFU update on Android.

I tested it on 5 different phones and OS versions. 
Can You help me?

Steps:
1. Connect with BLE device.
2. Run DFU Mode.
3. Increment mac address by one. ( needed by DFU )
3. Start Nordic DFU Service on device with incremented address. 

```
D/DfuService: isDebug() 
 I/DfuBaseService: Connecting to the device...
 D/BluetoothGatt: connect() - device: D5:1D:A7:76:6F:AF, auto: false
 D/BluetoothGatt: registerApp()
 D/BluetoothGatt: registerApp() - UUID=2b538b77-64be-4e12-9cbc-4a6017c1658e
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=D5:1D:A7:76:6F:AF
E/DfuBaseService: Connection state change error: 133 newState: 0
 E/DfuBaseService: Device got disconnected before service discovery finished
 D/BluetoothGatt: refresh() - device: D5:1D:A7:76:6F:AF
 I/DfuBaseService: Refreshing result: true
 I/DfuBaseService: Cleaning up...
 D/BluetoothGatt: close()
 D/BluetoothGatt: unregisterApp() - mClientIf=5
 D/DfuService: getNotificationTarget() 
D/DfuPresenter: DfuProgress onDeviceDisconnected
E/DfuPresenter: DfuProgress onError: DFU DEVICE DISCONNECTED address: D5:1D:A7:76:6F:AF error: 4096 errorType: 0
```

philips77, 2017-10-18:
Not enough information. Do you start the service twice? The lib should be able to restart device into bootloader mode automatically if you use standard Nordic buttonless service and bootloader. You don't have to do it manually. Just start it when you are connected to your device, after that you may disconnect in your app as the lib will take over the connection.
Error 133 is common in older devices, what devices are you using?

Esperanz0, 2017-10-18:
Yes after error I'm trying again. 

Devices:
Sony Xperia Z3
Sony Xperia Z5
Sony Xperia Z5 Premium
Nexus 5x
Nexus 6p
Google Pixel
Samsung Galaxy s5
Samsung Galaxy s7

Lib can put device into DFU mode? 
Ok I didn't kown. 

I tried with .setForceDfu(false) and .setForceDfu(true) but devices didn't change mode into DFU.

philips77, 2017-10-18:
Which SDK is your project based on?
In almost all of them there is a buttonless service which should be configured in your app mode. The DFU lib will determine has it connected to the bootloader or app with buttonless feature based on the discovery result. It will then automatically switch to bootloader mode and continue with update. You start it once when in app mode and wait for completed/error events.

Esperanz0, 2017-10-19:
So what I can do from Android side to confirm this is device issue ?

SDK is 12.2 

nrf Toolbox DFU should also put device into DFU mode if everything is fine?
I'm getting error DFU SERVICE NOT FOUND when trying to use nrf Toolbox with bt device. 

That's why I always manually changed Mode into DFU.

philips77, 2017-10-19:
Yes, nRF Toolbox and nRF Connect should automatically switch to DFU mode. Please, try with nRF Connect and post logs fro the app here so I could check what went wrong.

You may get DFU SERVICE NOT FOUND if Android has cached services and there was no DFU buttonless service there. If you use different UUIDs for that purpose, you can also set them using the DfuServiceInitiator.

Esperanz0, 2018-02-05:
I didn't saw this problem again for a long time. 

If I remember correctly there was a problem with android behavior and mac address.

Android can't connect to public mac address if device is not scanned before. 

So I added short BLE scanning between enabling DFU MODE and Start Dfu Service.

closed 2018-02-05 by Esperanz0

philips77, 2018-02-05:
Small clarification:
Can't connect to private mac address. By default it assume public one. How would you know random address before scanning? Random addresses may change often.

Esperanz0, 2018-02-06:
In our devices mac address is not random so I know it from start.