nordicsemi/Android-DFU-Library #79: Unable to read version number (error 133)

fradali, 2017-10-27:
Hello,
I'm trying to connect in DFU mode my device, I've checked some old issues about (error 133), but it didn't help me at all.
So what I'm trying to do is:

Connect to device
Execute a command to switch to DFU mode
Disconnect the device to search for the device(in DFU mode) with a new Mac Address (+1)
Start the update of Firmware using the device connected.

I tried to add a delay to start the DFU but it didn't work , I tried also to start DFU for the second Time and it didn't work.

This problem is happening with huawei P8/P9 lite

Please find below the Logfile
[Log.txt](https://github.com/NordicSemiconductor/Android-DFU-Library/files/1421759/Log.txt)

philips77, 2017-10-28:
Hi, the library should do what you wrote automatically. It switches to bootloader mode, reconnects and performs dfu when started in app mode. If you get 133, after all my experience with writing this lib, is that you can't really avoid it by modifying the code. What could be done was done, I think. It's either the BLE strach on those phones that causes the troubles, or your device. 133 comes when Android gets something unexpected or doesn't get something expected, or timings are incorrect. Perhaps setting a different connection interval, or something similar could fix. But trying to improve what this lib is doing won't help, in my opinion.

philips77, 2017-10-28:
Do you use the lastest version? It's 1.5.1 now.

fradali, 2017-10-30:
Yes I'm using the latest version 1.5.1 , this happens only with huawei , it's weird. I think the problem is coming from the device

philips77, 2017-10-30:
Could you try DFU using nRF Connect and copy logs from the app here?

fradali, 2017-10-30:
Yes sure 
[Log 2017-10-30 10_49_55.txt](https://github.com/NordicPlayground/Android-DFU-Library/files/1426715/Log.2017-10-30.10_49_55.txt)

modislaszlox2, 2018-01-03:
I'm having constant 133 error also with the latest version. Before updating the library to 1.6.1 I was using 1.5.0 and that one is working properly. Something got broken somewhere between those versions. I will continue using the old version I just wanted to notify the devs about possible issue.

philips77, 2018-10-24:
Hi, did you manage to fix your issue? On the last log it seems that the DFU completed successfully.

closed 2019-01-29 by philips77