nordicsemi/Android-DFU-Library #80: Getting FW version error even though only specifying scope

vikeri, 2017-10-31:
I followed the tip here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/154#issuecomment-338974651

I still get the FW error: `FW version failure` though and the logs say `DFU target does not support (SD/BL)+App update, splitting into 2 parts` so I'm not sure the `setScope` method is working properly for me.

philips77, 2017-10-31:
Hi, could you write what sdk is your bootloader from, what do you have in the zip file, what scope you set and copy paste all logs you get from LogCat?

vikeri, 2017-10-31:
- Our bootloader is from SDK 13
- The package contains: BL+SD+APP
- The scope is set to `DfuServiceInitiator.SCOPE_APPLICATION`

The logs:
```
10-31 12:22:39.302 18519 22775 I DfuImpl : Sending Enter Bootloader (Op Code = 1)
10-31 12:22:39.372 18519 22775 I DfuImpl : Response received (Op Code = 1, Status = 1)
10-31 12:22:59.472 18519 18531 E DfuBaseService: Connection state change error: 8 newState: 0
10-31 12:22:59.472 18519 22775 I DfuBaseService: Refreshing result: true
10-31 12:22:59.472 18519 22775 I DfuBaseService: Cleaning up...
10-31 12:22:59.482 18519 22775 I DfuImpl : Restarting to bootloader mode
10-31 12:22:59.522 18519 18519 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
10-31 12:22:59.632 18519 22775 I DfuImpl : Scanning for new address finished with: E5:3A:38:E5:92:88
10-31 12:23:01.742 18519 22775 I DfuBaseService: Connecting to the device...
10-31 12:23:04.262 18519 18519 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
10-31 12:23:04.282 18519 18531 I DfuBaseService: Connected to GATT server
10-31 12:23:04.282 18519 18531 I DfuBaseService: Attempting to start service discovery... succeed
10-31 12:23:05.352 18519 18643 I DfuBaseService: Services discovered
10-31 12:23:05.352 18519 22775 W DfuImpl : DFU target does not support (SD/BL)+App update, splitting into 2 parts
10-31 12:23:05.352 18519 22775 W DfuImpl : Secure DFU bootloader found
10-31 12:23:06.372 18519 22775 I DfuImpl : Requesting MTU = 517
10-31 12:23:06.452 18519 18530 W DfuImpl : MTU changed to: 23
10-31 12:23:06.452 18519 22775 I DfuImpl : Enabling notifications...
10-31 12:23:07.532 18519 22775 I DfuImpl : Setting object to Command (Op Code = 6, Type = 1)
10-31 12:23:07.592 18519 22775 I DfuImpl : Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
10-31 12:23:07.592 18519 22775 I DfuImpl : Sending the number of packets before notifications (Op Code = 2, Value = 0)
10-31 12:23:07.642 18519 22775 I DfuImpl : Creating Init packet object (Op Code = 1, Type = 1, Size = 140)
10-31 12:23:07.722 18519 22775 I DfuImpl : Sending 140 bytes of init packet...
10-31 12:23:07.722 18519 22775 I DfuImpl : Sending init packet (Value = 12-89-01-0A-43-08-01-12-3F-08-02-10-34-1A-05-98-01-00-9E-01)
10-31 12:23:07.732 18519 22775 I DfuImpl : Sending init packet (Value = 20-03-28-B8-B0-07-30-B8-9A-02-38-00-42-24-08-03-12-20-D6-99)
10-31 12:23:07.742 18519 22775 I DfuImpl : Sending init packet (Value = BB-FC-BF-62-28-AB-B4-E9-05-99-50-4D-35-4A-C8-38-7A-45-7A-20)
10-31 12:23:07.742 18519 22775 I DfuImpl : Sending init packet (Value = 19-B0-FD-A2-07-B4-6E-D1-8E-80-48-00-10-00-1A-40-33-D1-B6-09)
10-31 12:23:07.752 18519 22775 I DfuImpl : Sending init packet (Value = 9E-26-29-FF-B7-15-06-60-4E-0E-E0-D6-50-95-69-D2-52-C1-42-01)
10-31 12:23:07.752 18519 22775 I DfuImpl : Sending init packet (Value = 08-59-C4-71-2A-A2-2F-F5-9F-50-52-46-51-CD-EA-D2-83-4D-FF-77)
10-31 12:23:07.762 18519 22775 I DfuImpl : Sending init packet (Value = D2-D2-ED-C8-C7-66-52-9E-2D-74-9A-CC-9C-AD-AF-8A-0D-14-1C-DD)
10-31 12:23:07.762 18519 22775 I DfuImpl : Sending Calculate Checksum command (Op Code = 3)
10-31 12:23:07.862 18519 22775 I DfuImpl : Checksum received (Offset = 140, CRC = 2340117B)
10-31 12:23:07.862 18519 22775 I DfuImpl : Executing init packet (Op Code = 4)
10-31 12:23:07.922 18519 22775 E DfuImpl : Executing object failed (error 11) (extended error 5): EXTENDED ERROR
10-31 12:23:07.932 18519 22775 E DfuImpl : Extended Error details: FW version failure
10-31 12:23:07.972 18519 22775 I DfuBaseService: Disconnecting from the device...
10-31 12:23:07.982 18519 18643 I DfuBaseService: Disconnected from GATT server
10-31 12:23:07.992 18519 22775 I DfuBaseService: Refreshing result: true
10-31 12:23:07.992 18519 22775 I DfuBaseService: Cleaning up...
10-31 12:23:08.012 18519 18519 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
10-31 12:24:13.582 18519 23364 I DfuBaseService: Connecting to the device...
10-31 12:24:15.862 18519 18519 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
10-31 12:24:15.882 18519 18675 I DfuBaseService: Connected to GATT server
10-31 12:24:15.882 18519 18675 I DfuBaseService: Attempting to start service discovery... succeed
10-31 12:24:16.842 18519 18675 I DfuBaseService: Services discovered
10-31 12:24:16.842 18519 23364 W DfuImpl : DFU target does not support (SD/BL)+App update, splitting into 2 parts
10-31 12:24:16.842 18519 23364 I DfuImpl : Buttonless service without bond sharing found -> SDK 13 or newer
10-31 12:24:17.862 18519 23364 I DfuImpl : Enabling indications...
10-31 12:24:19.022 18519 23364 I DfuImpl : Sending Enter Bootloader (Op Code = 1)
10-31 12:24:19.082 18519 23364 I DfuImpl : Response received (Op Code = 1, Status = 1)
10-31 12:24:39.182 18519 18643 E DfuBaseService: Connection state change error: 8 newState: 0
10-31 12:24:39.202 18519 23364 I DfuBaseService: Refreshing result: true
10-31 12:24:39.202 18519 23364 I DfuBaseService: Cleaning up...
10-31 12:24:39.222 18519 23364 I DfuImpl : Restarting to bootloader mode
10-31 12:24:39.232 18519 18519 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
10-31 12:24:39.382 18519 23364 I DfuImpl : Scanning for new address finished with: E5:3A:38:E5:92:88
10-31 12:24:41.502 18519 23364 I DfuBaseService: Connecting to the device...
[... cut ...]
10-31 12:24:47.602 18519 23364 E DfuImpl : Extended Error details: FW version failure
10-31 12:24:47.612 18519 23364 I DfuBaseService: Disconnecting from the device...
10-31 12:24:47.622 18519 18675 I DfuBaseService: Disconnected from GATT server
10-31 12:24:47.622 18519 23364 I DfuBaseService: Refreshing result: true
10-31 12:24:47.622 18519 23364 I DfuBaseService: Cleaning up...
10-31 12:24:47.642 18519 18519 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

philips77, 2017-10-31:
It looks like a bug in the library. I'll take a looks at it tomorrow. Thank you for reporting it.

philips77, 2017-11-03:
Hi,

So there are couple of issues. First of all, how did you manage to set the scope when the DfuServiceInitiator has a bug here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/2c994b36cdb7cd3b78ae90bbd293b7cf3c9f19af/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L203
Of course it should throw if mime-type is NOT equal to ZIP. And the `setScope(...)` method should be called after `setZip(...)`, as `setZip(...)` resets the initiator state.
Secondly, I don't think you set the file type as `DfuBaseService.TYPE_APPLICATION`, otherwise you wouldn't get `W DfuImpl : DFU target does not support (SD/BL)+App update, splitting into 2 parts` in the log. 
I'll release 1.5.2 soon.

philips77, 2017-11-06:
Hi, did you try with version 1.5.2?

vikeri, 2017-11-07:
Hi! Haven't gotten around to try it yet. We went with another strategy to roll it out. So it might be a while before I can test the latest version. I'll close it for now and reopen it if it still didn't work.

closed 2017-11-07 by vikeri