nordicsemi/Android-DFU-Library #81: ERROR:4096 MESSAGE: DFU DEVICE DISCONNECTED

rengshun, 2017-11-01:
I have problem with DFU update on some devices (like Huawei honorV9).

the logs:
11-01 10:12:47.419 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Connecting to the device...
11-01 10:12:47.419 12473-13044/cn.gengee.insaits2 I/BluetoothDevice: connectGatt
11-01 10:12:47.521 12473-12681/cn.gengee.insaits2 I/DfuBaseService: Connected to GATT server
11-01 10:12:47.523 12473-12681/cn.gengee.insaits2 I/DfuBaseService: Attempting to start service discovery... succeed
11-01 10:12:47.565 12473-12485/cn.gengee.insaits2 I/DfuBaseService: Services discovered
11-01 10:12:47.566 12473-12473/cn.gengee.insaits2 I/SensorUpgradeDialog: onDeviceConnected
11-01 10:12:47.566 12473-12473/cn.gengee.insaits2 I/SensorUpgradeDialog: onDfuProcessStarting
11-01 10:12:48.566 12473-13044/cn.gengee.insaits2 I/DfuImpl: Reading DFU version number...
11-01 10:12:48.603 12473-12486/cn.gengee.insaits2 W/BluetoothGatt: onCharacteristicRead() - Device=FA:58:60:37:28:E2 handle=61 Status=0
11-01 10:12:48.603 12473-13044/cn.gengee.insaits2 I/DfuImpl: Version number read: 0.1 -> Application with Legacy buttonless update from SDK 7.0 or newer
11-01 10:12:48.603 12473-13044/cn.gengee.insaits2 W/DfuImpl: Application with legacy buttonless update found
11-01 10:12:48.604 12473-13044/cn.gengee.insaits2 I/DfuImpl: Enabling notifications...
11-01 10:12:49.634 12473-13044/cn.gengee.insaits2 I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
11-01 10:12:49.634 12473-12473/cn.gengee.insaits2 I/SensorUpgradeDialog: onEnablingDfuMode
11-01 10:12:50.484 12473-12485/cn.gengee.insaits2 I/DfuBaseService: Disconnected from GATT server
11-01 10:12:50.486 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Refreshing result: true
11-01 10:12:50.486 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Cleaning up...
11-01 10:12:50.489 12473-13044/cn.gengee.insaits2 I/DfuImpl: Starting service that will connect to the DFU bootloader
11-01 10:12:50.591 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
11-01 10:12:51.141 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
11-01 10:12:52.511 12473-12473/cn.gengee.insaits2 I/SensorUpgradeDialog: onDeviceConnecting
11-01 10:12:52.512 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Connecting to the device...
11-01 10:12:52.512 12473-13044/cn.gengee.insaits2 I/BluetoothDevice: connectGatt
11-01 10:12:52.641 12473-13012/cn.gengee.insaits2 I/DfuBaseService: Connected to GATT server
11-01 10:12:52.642 12473-13012/cn.gengee.insaits2 I/DfuBaseService: Attempting to start service discovery... succeed
11-01 10:12:55.583 12473-12485/cn.gengee.insaits2 I/DfuBaseService: Disconnected from GATT server
11-01 10:12:55.584 12473-13044/cn.gengee.insaits2 E/DfuBaseService: Device got disconnected before service discovery finished
11-01 10:12:55.585 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Refreshing result: true
11-01 10:12:55.585 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Restarting the service
11-01 10:12:55.677 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
11-01 10:12:55.794 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
11-01 10:12:57.597 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Connecting to the device...
11-01 10:12:57.598 12473-13044/cn.gengee.insaits2 I/BluetoothDevice: connectGatt
11-01 10:12:57.726 12473-12681/cn.gengee.insaits2 I/DfuBaseService: Connected to GATT server
11-01 10:12:57.727 12473-12681/cn.gengee.insaits2 I/DfuBaseService: Attempting to start service discovery... succeed
11-01 10:13:00.245 12473-12486/cn.gengee.insaits2 I/DfuBaseService: Disconnected from GATT server
11-01 10:13:00.251 12473-13044/cn.gengee.insaits2 E/DfuBaseService: Device got disconnected before service discovery finished
11-01 10:13:00.252 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Refreshing result: true
11-01 10:13:00.252 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Restarting the service
11-01 10:13:00.351 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
11-01 10:13:00.545 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
11-01 10:13:02.273 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Connecting to the device...
11-01 10:13:02.273 12473-13044/cn.gengee.insaits2 I/BluetoothDevice: connectGatt
11-01 10:13:02.398 12473-13012/cn.gengee.insaits2 I/DfuBaseService: Connected to GATT server
11-01 10:13:02.399 12473-13012/cn.gengee.insaits2 I/DfuBaseService: Attempting to start service discovery... succeed
11-01 10:13:04.987 12473-12485/cn.gengee.insaits2 I/DfuBaseService: Disconnected from GATT server
11-01 10:13:04.987 12473-13044/cn.gengee.insaits2 E/DfuBaseService: Device got disconnected before service discovery finished
11-01 10:13:05.089 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Refreshing result: true
11-01 10:13:05.089 12473-13044/cn.gengee.insaits2 I/DfuBaseService: Cleaning up...
11-01 10:13:05.091 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
11-01 10:13:05.596 12473-12473/cn.gengee.insaits2 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED

usamaaftab80, 2017-11-01:
I recently faced the same problem while running my Android app on Nexus 6P. Device randomly (not consistently) disconnects from the peripheral with the error code 4096. After disconnection it put my device in DfuTarg broadcast mode.

If i understand the problem correctly, it seems to be an issue of SDK. Can you verify which SDK are you running on the peripheral?

In my case, i was running SDK 12.0 which showed this bug. What i did was to manually put the device in DfuTarg mode (by writing 0x01 on the DFU characteristic) and then run Dfu Service with device name "DfuTarg". Let me know if it helps?

ngallazzi, 2017-11-27:
Hello, I have the same problem on a Huawei phone (android 7.0), meanwhile on another phone (General Mobile Gm6 android 7.0) everything works fine.. Have you figured out something from your tests? Thanks

guerjon, 2018-05-10:
Same problem here with the google pixel Android 8.1

igor-gorjanc, 2018-10-01:
Same problem with Android 8.0, Mesh DFU.

philips77, 2018-10-24:
Hi, I'm about to release version 1.8.0 today. Could you check if the issue still happens there?

philips77, 2018-10-29:
1.8.0 released.

pacebout, 2018-11-13:
It's still doing the same issue on Huawei phones... having the same error log:
`Device got disconnected before service discovery finished`
I have no further details then that at the moment.

bittergourdbd, 2018-11-29:
Hi, I have the same problem on ASUS_X018D (Android 7.0). Please help me? Thanks.

beegee-tokyo, 2019-03-23:
Any news on this?
Using 1.8.1 and having the same problem **`Device got disconnected before service discovery finished`** on a Huawei MediaTab M5 (Android 8.0), a Samssung Galaxy J6 Core (Android 8.1), and a Samsung Galaxy Note 10.1 (Android 7.1.2)

nir-ilumisolutions, 2019-12-13:
I am facing the same issue on Google Pixel 9.0

**onFailure: error: java.lang.RuntimeException: Update process failed!!! ERROR: 4096 - 0 - DFU DEVICE DISCONNECTED**

Any solution to this?

harishankar-jayasakthi, 2020-06-03:
Hello, I have the same problem with LLD-AL20 : ANDROID 9 

dfu : onError  error CODE: 4096, type :0, message :DFU DEVICE DISCONNECTED 
LATEST dfu:1.10.3

Any solution for this issue?