nordicsemi/Android-DFU-Library #83: Android DFU upgrade function package use confusion after error, upgrade function is invalid

adsfpj, 2017-11-06:
Android DFU upgrade function package use confusion after error, upgrade function is invalid

#dfu
-dontwarn no.nordicsemi.android.dfu.**
-keep class no.nordicsemi.android.dfu.** { *; }

-keepattributes Signature
-keep class sun.misc.Unsafe { *; }
-keep class com.google.gson.stream.** { *; }
-keep class com.google.gson.examples.android.model.** { *; }
-keep class com.google.gson.** { *;}
-keep class no.nordicsemi.android.dfu.manifest.** {*;}

-keep class com.daolai.www.** {
    public void set*(***);
    public *** get*();
    public *** is*();
}

philips77, 2017-11-06:
Hi, it looks like your manifest file is invalid. The name of the bin file specified in the manifest does not match the application bin file in the ZIP, or the bin file is missing. The error comes from here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/internal/ArchiveInputStream.java#L136

The correct ZIP packet should have 3 or 5 files inside:
[image]
If your distribution file contains only the application firmware, the `sd_bl.*` should not be there.

Here's an example of a correct manifest file matching those files for legacy DFU:
```json
{
    "manifest": {
        "application": {
            "bin_file": "ble_app_hrm_dfu_s132_v2_0_0_sdk_11_0.bin",
            "dat_file": "ble_app_hrm_dfu_s132_v2_0_0_sdk_11_0.dat",
            "init_packet_data": {
                "application_version": 4294967295,
                "device_revision": 65535,
                "device_type": 65535,
                "firmware_crc16": 37812,
                "softdevice_req": [
                    65534
                ]
            }
        },
        "dfu_version": 0.5,
        "softdevice_bootloader": {
            "bin_file": "sd_bl.bin",
            "bl_size": 13324,
            "dat_file": "sd_bl.dat",
            "init_packet_data": {
                "application_version": 4294967295,
                "device_revision": 65535,
                "device_type": 65535,
                "firmware_crc16": 29480,
                "softdevice_req": [
                    65534
                ]
            },
            "sd_size": 106484
        }
    }
}
```

and for Secure DFU:
```json
{
    "manifest": {
        "application": {
            "bin_file": "ble_app_hrm_dfu_s132_v2_0_0_sdk_14_1.bin",
            "dat_file": "ble_app_hrm_dfu_s132_v2_0_0_sdk_14_1.dat"
        },
        "softdevice_bootloader": {
            "bin_file": "sd_bl.bin",
            "dat_file": "sd_bl.dat",
            "info_read_only_metadata": {
                "bl_size": 13324,
                "sd_size": 106484
            }
        }
    }
}
```
Of course you may name the app and sd/bl files however you want, just make sure the manifest has the same names.

adsfpj, 2017-11-07:
It seems not the problem, then why I confused when the package is all right, the use of confusion packaging will have this problem

philips77, 2017-11-10:
Does it work without proguard?

adsfpj, 2017-11-11:
without proguard it work

philips77, 2017-11-17:
Hi, did you manage to solve this issue?
How about adding:
```gradle
-keep class no.nordicsemi.android.dfu.** { *; }
```
to the *proguard-rules.pro*?

adsfpj, 2017-11-21:
No.
￼￼
[image]
￼

philips77, 2017-11-22:
Looks like the `@SerializeName` annotation has been removed by proguard.
How about adding:
```gradle
-keep com.google.gson.annotations.SerializeName
-keepclassmembers class no.nordicsemi.android.dfu.internal.manifest.** { *; }
```
or
```gradle
-keepattributes *Annotation*
```
I'm reading this: https://stackoverflow.com/questions/15476315/how-keep-my-class-from-obfuscate-by-proguard. Also, here: http://movingbitseasy.blogspot.no/2015/01/proguard-android-obfuscation.html they write that:

> One possibility is to use `@SerializedName` annotation do set a static name for an attribute, which is not affected from obfuscation.

For some reason your proguard removes it.

philips77, 2018-10-24:
Hi, did you manage to fix the issue? It hangs here almost an year.

adsfpj, 2018-10-25:
It was solved a year ago.

-dontwarn no.nordicsemi.android.**
-keep class no.nordicsemi.android.** { *; }
-keep class com.google.gson.** { *;}
-keepclassmembers class no.nordicsemi.android.dfu.internal.manifest.** { *; }
-keepclassmembers class * implements java.io.Serializable {
    static final long serialVersionUID;
    private static final java.io.ObjectStreamField[] serialPersistentFields;
    private void writeObject(java.io.ObjectOutputStream);
    private void readObject(java.io.ObjectInputStream);
    java.lang.Object writeReplace();
    java.lang.Object readResolve();
}

adsfpj, 2018-10-25:
It has been solved.

-dontwarn no.nordicsemi.android.**
-keep class no.nordicsemi.android.** { *; }
-keep class com.google.gson.** { *;}
-keepclassmembers class no.nordicsemi.android.dfu.internal.manifest.** { *; }
-keepclassmembers class * implements java.io.Serializable {
    static final long serialVersionUID;
    private static final java.io.ObjectStreamField[] serialPersistentFields;
    private void writeObject(java.io.ObjectOutputStream);
    private void readObject(java.io.ObjectInputStream);
    java.lang.Object writeReplace();
    java.lang.Object readResolve();
}

closed 2018-10-25 by adsfpj

philips77, 2018-10-25:
Great, thanks for letting us know!