nordicsemi/Android-DFU-Library #84: DFU stuck at 90/91%

ABouretz, 2017-11-06:
Device configuration
SDK 5.2.0
SoftDevice 6.0.0

Uploading Application .bin file

Trying DFU with last version of nRF Connect, the DFU is stuck at 91% (DFU library 1.5.2). It doesn't work with v1.3.0 but works with the old 1.0.4 version.

philips77, 2017-11-06:
Hi, I just did a test using this setup: https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/master/documentation/Automated%20tests/DFU%20Tests/nRF51/SDK%205.2%20-%20not%20used and the newest nRF Connect and I was able to update my nRF51 DK using PRNs set to 10.

ABouretz, 2017-11-06:
Thanks for your quick answer, enabling PRNs and setting them to 10 resolved the issue. I didn't find any documentation about that on this repo, is there a detailed documentation somewhere ? For information, I was testing on a Samsung Galaxy S7 (Android 7.0).

closed 2018-02-02 by philips77

tusharSensovium, 2018-04-30:
I also face same issue in Redmi 3s prime(xiaomi). some times dfu stopped at 20% - 25% . And same code works on others mobiles like samsung, sony.