nordicsemi/Android-DFU-Library #86: Error code 5 (Invalid object)

Terik23, 2017-11-10:
We use on device SDK 12.2

1. We send a cmdto the device for enter to DFU mode.

2. After the cmd was sent, we do the following in `BluetoothGattCallback.onCharacteristicWrite`:

```
new Thread(new Runnable() {
    @Override
    public void run() {
        try {
            Thread.sleep(1500);
            Intent intent = new Intent().setAction(ACTION_START_DFU);
            context.sendBroadcast(intent);
        } catch (InterruptedException e) {
            e.printStackTrace();
        }
    }
}).run();
```
3. Receiver code:

```
String action = intent.getAction();
if (action.equals(ACTION_START_DFU)) {
    BluetoothDevice dfuTarg = getDfuTarg();
    final DfuServiceInitiator starter = new DfuServiceInitiator(dfuTarg.getAddress())
            .setDeviceName(dfuTarg.getName())
            .setKeepBond(false)
            .setForceDfu(false)
            .setPacketsReceiptNotificationsEnabled(Build.VERSION.SDK_INT < Build.VERSION_CODES.M)
            .setPacketsReceiptNotificationsValue(1)
            .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
            .setDisableNotification(true);
    starter.setZip(firmwareUri, firmwareUri.getPath());
    starter.start(context, DfuService.class);
}
```

It was tested on the following smartphones:
- Xiaomi Mi5s (Android 6.0.1)
- Samsung Galaxy S6 (Android 7.0)
- Samsung Galaxy S8 (Android 7.0)

On S6, sometimes after the percent == 100 error 5 (Invalid object) occurs. On Mi5S and S8, there was no such error.
Sometimes on any of the smartphones there is an error 133 and it repeats again every time you try to flash the device until the firmware is made using the NRF Toolbox (in this case the device is manually switched to DFU mode and we find through the NRF Toolbox a DFUTarg device with a MacAdress + 1).

philips77, 2017-11-10:
I saw this error 5 once. Don't know what may cause it, I'll ask our SDK team on Monday.

philips77, 2017-11-17:
I got some more information about error 5 after executing the last object:

> Erase operation timed out. SoftDevice is the one that is going to allow the write
If it is super busy with other things it might synchronously report back that it can't allow the erase to happend. Note that fstorage has retried getting this scheduled X number of times and it is getting busy every time (from the softdevice). Connection interval will have an impact on this. You can retry the execute N number of times in the phone.

So I'll implement additional Execute in case of error 5 was received after sending all objects. Should work in the next release (1.6.0).

philips77, 2017-11-17:
One other situation when error 5 (INVALID_OBJECT) would be thrown after the last execute is when you tried to update only the SoftDevice (without bootloader) and the API has changed (the SD major version is different). Then the bootloader would not work anymore so such update is not possible. But I guess this is not your case, so just FYI.

philips77, 2017-11-17:
Maybe one more info. Retrying will not solve error 5 issue, but will increase a chance that Execute will succeed. If still didn't work, start DFU again (this time it will be very fast as the firmware is already on the device and CRCs are the same, so it should just compare Init packet and execute Init packet and the Data object.

closed 2017-11-17 by philips77