nordicsemi/Android-DFU-Library #87: setDisableNotification(true) not working anymore

fabriceimperial, 2017-11-10:
Hi,
it seems that the value of `.setDisableNotification(true)` from the _DfuServiceInitiator_ is not taken into account and therefore the `null` value returned by the `getNotificationTarget()` in the _DfuService_ class extending the _DfuBaseService_ crash the application.

```
FATAL EXCEPTION: IntentService[DfuBaseService]
Process: com.example.myapplication, PID: 31396
java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.lang.Class.getName()' on a null object reference
    at android.content.ComponentName.<init>(ComponentName.java:129)
    at android.content.Intent.<init>(Intent.java:4449)
    at no.nordicsemi.android.dfu.DfuBaseService.startForeground(DfuBaseService.java:1500)
    at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:925)
    at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:66)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:148)
    at android.os.HandlerThread.run(HandlerThread.java:61)
```

It works well with the dfu 1.3.X and 1.4.X

**System:**
```
Android 6.0.1
compileSdkVersion 26
buildToolsVersion '26.0.2'
minSdkVersion 18
targetSdkVersion 25
dfu 1.5.X
```

philips77, 2017-11-10:
Thanks for the report. I'll look into it today. Did you do any debugging why?

fabriceimperial, 2017-11-10:
Not really, I quickly looked at the usage of `DfuBaseService.EXTRA_DISABLE_NOTIFICATION` and it looks fine but I didn't do any further investigation to check that the corresponding boolean variable was used correctly to prevent the notification from appearing on build lower than Oreo.

philips77, 2017-11-16:
I found the issue. Migrating the DFU library to support Oreo I also changed the service to be the foreground service, so the system doesn't kill it if DFU goes in background (https://developer.android.com/about/versions/oreo/background.html). However, I was assuming the the `getNotificationTarget()` will always return not *null*. 

I see the following solution. I will make setting the service as a foreground service optional (with some docs that it is recommended to have it in foreground in Oreo or newer). In that case, the `getNotificationTarget()` should not return *null*. The progress notifications will still be disable..able(?) like they are now.

philips77, 2017-11-16:
Fix will be available in the next relase.

fabriceimperial, 2017-11-16:
Great! Thank you _philips77_

closed 2017-11-16 by fabriceimperial

Sammekl, 2017-11-30:
I'm on the latest version (1.6.0) and this issue still exists.

philips77, 2017-11-30:
Yes, but that's a "foreground service" notification. This is because the DFU service is by default started as a foreground service to avoid it being killed on Oreo+ before it finishes. You may disable foreground with setForegrond(false) on the initiator, but if you want the dfu to also work when your app is closed (i.e. user clicks a notification with msg that new fw is available), we recommend leaving it. Or you may enabled foreground only on Oreo+.

Yurdaaaaa, 2019-04-01:
What if i want to show notification with loading but i dont want to implement getNotificationTarget()? I can rewrite part of the service but it would be nice to just set do nothing on click.

philips77, 2019-04-01:
You may override `updateForegroundNotification` method in DfuService and remove the content intent, I suppose.

Yurdaaaaa, 2019-04-02:
Well, there is more methods using getNotificationTarget() so it will take a while to do it. But yes in this approach sure u can customize it for your needs.