nordicsemi/Android-DFU-Library #90: Unable to send a DFU update to a device if the update was started while another DFU update is in progress for a different device

justin-guan, 2017-12-18:
The library appears to not be able to send a DFU update to a device if the update was started while another DFU update is in progress for a different device. 

For example, if I have two devices (A and B), I can start an update on A. While A is in the middle of updating (`onProgressChanged()` is being called), I start an update on B. A will complete without error, and then B will do all of the intialization steps for a DFU update and hit 0% in `onProgressChanged()` (all of this work for B is done after A completes). B will then hang at 0% until a disconnection error occurs, and the update fails.

I was able to reproduce this behavior in the nRF Connect app too by starting two DFU updates to two different devices at the same time.

philips77, 2017-12-18:
This has never been tested and I agree that changes of it working are minimal. The service may be started many times at the same time, but all instances would share some data, for example the gatt callbacks or progress info object, from top on my head. I'd have to look into it.

philips77, 2018-02-02:
I guess it could work if you have 2+ identical service classes that extend DfuBaseService, so that for the system you start 2 different services, did you try this?

justin-guan, 2018-02-02:
This is not something that I tried. It never occurred to me to have 2+ service classes extending `DfuBaseService` to start the DFU update multiple times, but even if this did work, then I feel that this would be difficult to scale this to a variable amount of devices running DFU updates at the same time.
Unfortunately I no longer have the hardware necessary to try this. This ended up not being a critical feature I needed.

closed 2018-10-24 by philips77