nordicsemi/Android-DFU-Library #91: Unable to Run in android Studio

VinayTShetty, 2018-01-08:
After extracting the ZIP file i am unable to See the RUN button in android Studio.
[image]
[image]
[image]

philips77, 2018-01-08:
Of course. This is a library, not an app.

closed 2018-01-08 by philips77

VinayTShetty, 2018-01-09:
@philips77 
Can u please guide me on how to implement this library to the application.

philips77, 2018-01-09:
nRF Toolbox is using it. See the app/build.gradle file there, it imports the dfu library and then use it in dfu package. Please, read the documentation on DFU lib GitHub page on how to use it.