nordicsemi/Android-DFU-Library #92: ErrorCode: 517 errorType3,message"INVALID OBJECT"

ab2976818, 2018-01-24:
I/DfuBaseService: DFU service created. Version: 1.6.1
  I/DfuBaseService: Connecting to the device...
  I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
  I/DfuBaseService: Connected to GATT server
  I/DfuBaseService: Attempting to start service discovery... succeed
  I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
  I/DfuBaseService: Services discovered
  W/DfuBaseService: Target device disconnected with status: 8
  I/DfuBaseService: Refreshing result: true
  I/DfuBaseService: Cleaning up...
  I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
  I/DfuBaseService: Connecting to the device...
  I/DfuBaseService: Connected to GATT server
  I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
  I/DfuBaseService: Attempting to start service discovery... succeed
  I/DfuBaseService: Services discovered
  I/DfuBaseService: Disconnecting from the device...
  I/DfuBaseService: Disconnected from GATT server
  I/DfuBaseService: Refreshing result: true
  I/DfuBaseService: Cleaning up...
  I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
  I/DfuBaseService: DFU service destroyed

philips77, 2018-01-24:
"Invalid object" is an error when your init packet does not match the device (for example required SoftDevice version), or the signature is invalid. The documentation: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.2.0/lib_dfu_transport.html?cp=4_0_0_3_5_2 says: 

```
Data object does not match the firmware and hardware requirements, the signature is wrong, 
or parsing the command failed. 
```

The log you pasted is not related to the title.

philips77, 2018-02-02:
Did you manage to solve this issue?

closed 2018-10-24 by philips77