nordicsemi/Android-DFU-Library #95: DfuImpl: Creating Command object failed (error 5): INVALID OBJECT

Esperanz0, 2018-02-05:
I'm getting this error message when trying to update firmware.

I saw Your comment: 

> "Invalid object" is an error when your init packet does not match the device (for example required SoftDevice version), or the signature is invalid. The documentation: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.2.0/lib_dfu_transport.html?cp=4_0_0_3_5_2 says:
> 
> Data object does not match the firmware and hardware requirements, the signature is wrong, 
> or parsing the command failed. 
> The log you pasted is not related to the title.

But the problem is only with Samsung phones. 

Checked on:
Nexus 6p - Works fine
Sony Xperia z5 - Works fine
Sony Xperia z3 - Works fine
Huawei p8 - Works fine
Huawei p9 - Works fine
Samsung S7 - INVALID OBJECT
Samsung S8 - INVALID OBJECT

After this error I'm trying to run update agian but then getting: 
Executing object failed (error 8): OPERATION NOT PERMITTED

Logs: 

[Movesense_Samsungs7_dfu_errorr_logs.txt](https://github.com/NordicSemiconductor/Android-DFU-Library/files/1695484/Movesense_Samsungs7_dfu_errorr_logs.txt)

**So if error message telling something is wrong with firmware file why only Samsung can't proceed update?**

sengewald, 2018-07-23:
Hi,
have the same problem here.
From the onProgressChanged callback it seems like there is always a progress of 5% until the same error message comes up again.
E/DfuImpl: Creating Command object failed (error 5): INVALID OBJECT
When restarting the process the progress is not reset to 0% but keeps restarting from the last position.

If using Library version 1.7.0 there is also another problem:
Unfortunately even after i reached 100% after a couple of trys the FW was not updated but i got the error message DFU Device disconnected
E/DfuBaseService: Unable to write Op Code: device disconnected

This happens as well on my Blackberry Priv with Android 6.
When using the old 1.6.1 Library the update finish on Blackberry but still fails on Samsung with the same errors
 
Also the device does not immediately come back from _dfu mode after the error.
It always takes about half a minute until i can see the device name without the "_dfu" attached.

Its a Galaxy S7 with Android Oreo.
Works on Blackberry Priv with Android 6

Summary:
Library 1.7.0: Does not work at all, after reaching 100% DFU Device disconnected error.
Library 1.6.1: Works well on Blackberry and Android 6, crashes every couple of % on Samsung with Android 8 and needs restart about 10times until upgrade finish.

sengewald, 2018-07-23:
Problem solved after adding this line of code:

starter.setPacketsReceiptNotificationsEnabled(true);
(starter is  DfuServiceInitiator)

Not sure what this command is doing, but it works fine now with lib 1.6.1 and 1.7.0 on Samsung with Android 8 and Blackberry with Android 6.

closed 2018-10-24 by philips77